espressif/esp-idf #9367: [ESP32-S3] Interrupt IN endpoints get stuck in device mode

Huckies, 2022-07-16:
## Environment

- Development Kit: ESP32-S3-DevKit-M
- Module or chip used: ESP32-S3-mini-1
- IDF version (run ``git describe --tags`` to find it): v4.4.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System: macOS
- Using an IDE?: VS Code 
- Power Supply: USB

## Problem Description

In device mode, some interrupt IN endpoints can get stuck if multiple endpoints are fed simultaneously.

### Expected Behavior

Multiple HID endpoints works independently, they can be loaded by separate tasks and data is fetched according to the host.

### Actual Behavior

If multiple  (each endpoint is confirmed "ready" using ```tud_hid_n_ready()```) endpoints are loaded in a single task, some of them can get "unready" forever.

## Information that might help

The OTG peripheral is initialized by calling
```
    usb_phy_config_t phy_conf = {
        .controller = USB_PHY_CTRL_OTG,
        .otg_mode = USB_OTG_MODE_DEVICE,
        .target = USB_PHY_TARGET_INT,
    };
    usb_phy_handle_t phy_hdl;
    usb_new_phy(&phy_conf, &phy_hdl);
```

I've also refer to the TinyUSB example that set the task invoking  ```tud_task()``` with a higher priority than the one loading endpoints. Due to the ESP32-S3 TRM describes the OTG peripheral in one stroke and inexpertness on usb registers, I'm unable to check the ``dcd_esp32sx.c``. 

I leave a log below, seems that there is some "crosstalk" between endpoints.

It's a definite issue, please reply if more information is needed. I'm looking forward to the solution

[log.txt](https://github.com/espressif/esp-idf/files/9125265/log.txt)
.

tore-espressif, 2022-07-21:
Hello @Huckies than you for the bug report.

I couldn't reproduce this issue, so I need some more information in order to help you debug this. Could you please provide the following?:

1. USB traffic capture from [Wireshark](https://www.wireshark.org/#download) (it contains more information than the log you provided)
2. Full device descriptor of your device (you can find it with [USB Device Tree Viewer](https://www.uwe-sieber.de/usbtreeview_e.html) )
3. Log from TinyUSB: run `idf.py menconfig` and set `Component config->TinyUSB Stack->TinyUSB log level` to 3. Then run `idf.py flash monitor`
4. A function that feeds the endpoint with new data (calls `tud_hid_n_report()`)

Thanks!
And just an FYI: with `tud_hid_n_ready()` you check readiness of an _interface_ rather than an endpoint.

tore-espressif, 2023-05-19:
Closing due to lack of feedback. Please feel free to reopen

closed 2023-05-19 by tore-espressif

electromuis, 2024-07-15:
Had a similar issue with my custom game controller. Arduino library creates an IN and OUT endpoint by default. Seems a lot better when patched for just an IN endpoint.
[USBHIDnew.txt](https://github.com/user-attachments/files/16239283/USBHIDnew.txt)
[USBHIDold.txt](https://github.com/user-attachments/files/16239284/USBHIDold.txt)

In my case it got into this condition:

if (in_status & BIT(15)) {
    USB0.in_ep_reg[n].diepint = BIT(15);
    ESP_EARLY_LOGE(TAG, "Unknown Condition");//todo:
    bus_reset();
  }

In https://github.dev/arendst/esp32-arduino-lib-builder/blob/283e317b71d2094ef80b6c3778dc983d271ddeaf/components/arduino_tinyusb/src/dcd_esp32s2.c#L46