espressif/esp-idf #9368: idf.py monitor goes into endless loop

davidzuhn, 2022-07-16:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: ESP32-S3-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v1.0
- Module or chip used:  ESP32-S3-WROOM
- IDF version (run ``git describe --tags`` to find it):  v5.0-dev-4037-g9b8c558e63
- Build System:      CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32s3-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:    mac OS
- Using an IDE?:  No
- Power Supply:     USB
## Problem Description

idf.py monitor entered an endless loop trying to handle error conditions

It's still printing this output, even though the device has been disconnected for 10 minutes, so it's not new real data coming from the target.

### Expected Behavior

idf.py monitor should only output based on new data coming from the target

### Actual Behavior

The app was sitting overnight, idle.  I had openocd running using the built-in JTAG adapter.   The application was started via gdb-no-python.   In another window, I was running 'idf.py monitor --no-reset'.

The log started to show:
```

abort() was called at PC 0x40377f67 on core 1
0x40377f67: lock_acquire_generic at /Users/zoo/esp32/esp-idf-latest-20220712/components/newlib/locks.c:130

Setting breakpoint at 0x403763da and returning...
0x403763da: panic_abort at /Users/zoo/esp32/esp-idf-latest-20220712/components/esp_system/panic.c:409

abort() was called at PC 0x40377f67 on core 1
0x40377f67: lock_acquire_generic at /Users/zoo/esp32/esp-idf-latest-20220712/components/newlib/locks.c:130

Setting breakpoint at 0x403763da and returning...
0x403763da: panic_abort at /Users/zoo/esp32/esp-idf-latest-20220712/components/esp_system/panic.c:409

```

and it's doing so now, even though I completely disconnected the target and there is definitely no target attached.

### Steps to reproduce

I do not have a recipe.  I started my program, ran a couple of status commands from the internal console, and then walked away.   The last command was run at 107784 (based on the ESP_LOG output), a short burst of activity at 39447414, and then sometime (unknown - perhaps immediately, perhaps a bit later) later, this error loop began.

## Debug Logs

```
abort() was called at PC 0x40377f67 on core 1
0x40377f67: lock_acquire_generic at /Users/zoo/esp32/esp-idf-latest-20220712/components/newlib/locks.c:130

Setting breakpoint at 0x403763da and returning...
0x403763da: panic_abort at /Users/zoo/esp32/esp-idf-latest-20220712/components/esp_system/panic.c:409

```
 in an endless loop.
## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [N/A] coredump (This provides stacks of tasks.)
- [X] [bugreport.zip](https://github.com/espressif/esp-idf/files/9126029/bugreport.zip)

davidzuhn, 2022-07-18:
@dobairoland -- I did note that these repeated error messages continued long (10+ minutes) after I completely disconnected the target board.    I'm pretty sure this is not _just_ a chip / software issue.   I believe something in the idf.py monitor code was continuing to process and output the error messages in a loop.  That's why I believed this warranted a bug report.  My code crashing from time to time isn't that unusual -- the unusual part is continuing to see the messages when the device isn't even connected anymore.

dobairoland, 2022-07-19:
Hi @davidzuhn. It is weird. I haven't seen such behavior before.

Did this happen only once? Were you connected through serial port or the native USB interface? Did you disconnect the debugging (JTAG) pins as well?

I have to think about this. Until now, I saw only exception backtraces when the serial connection was severed.

davidzuhn, 2022-07-20:
I've only seen it once, and I'm pretty sure it was using the usb serial/jtag interface (that's where I've been doing most of my work lately, despite the bugs/concerns that I'm running into).

dobairoland, 2022-07-20:
Hi @davidzuhn. Thanks for the information. Unfortunately, I cannot reproduce it. I've tried it with an ESP32-S3 devkitC connected through Serial/JTAG USB. It was logging a lot and I was trying to disconnect and reconnect in the meantime. When I disconnected, I immediately received the following and was not printing anything until I reconnected.
```
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect.......
```

Looking at the code I don't understand how this could happen. It is possible that in the OS- or pyserial-level buffer the data remained (wasn't discarded by mistake and later could not be because the port was already disconnected) and was resent over and over.

I'm sorry but I'm closing this since it cannot be reproduced. Since it happened only once then it shouldn't be critical to you.

Please let us know if this happens again - and possible provide more information about the circumstances (e.g. did the monitor print anything before looping?) - and we will re-visit this.

closed 2022-07-20 by dobairoland