espressif/esp-idf #9369: [ES32-S3]  USB Serial/JTAG as primary log output - Console REPL does not work

chipweinberger, 2022-07-16:
**ESP32-S3 Dev Kit C w/ 2MB PSRAM**
**ESP-IDF v4.4.1**

I Imagine this is related to https://github.com/espressif/esp-idf/issues/9364, https://github.com/espressif/esp-idf/issues/8879, https://github.com/espressif/esp-idf/issues/8738, but those are about USB CDC Console, not Serial/JTAG Console, which uses a different underlying USB implementation. 

**Question:**
- Is there a plan to support Console REPL with USB Serial/JTAG?

**Problem:**

After setting `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`, the program no longer compiles. 

[image]

**Compile Error:**

`/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(pd_console_api.cpp.obj):(.literal._Z19pd_console_api_initP10pd_error_t+0x24): undefined reference to esp_vfs_dev_cdcacm_set_rx_line_endings`

**Other Attempt:**

I've also tried using `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y` with the uart initialization thinking _perhaps_ USB Serial/JTAG gets internally routed to the default UART pins, i.e. TX (43) and RX (44) gpios. 

```
esp_vfs_dev_uart_port_set_rx_line_endings(...)
esp_vfs_dev_uart_port_set_tx_line_endings(...)
uart_set_pin(0, 43, 44, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
uart_driver_install(0, 256, 0, 0, NULL, 0);
uart_param_config(0, &uart_config);
esp_vfs_dev_uart_use_driver(...)
esp_console_init(&console_config);
etc...
```

But somewhat expected, it doesn't work and causes the console to go haywire with lots of `console> 4;1R
Console: Unrecognized command: 4;1R[24;1R[24;14R[24;1R[24;1R[24;14R` style lines being logged, and then a WDT crash triggers.`

```
0x40382721: esp_system_abort at /esp-idf/components/esp_system/esp_system.c:128
0x4038cc0d: abort at /esp-idf/components/newlib/abort.c:46
0x420aa030: task_wdt_isr at /esp-idf/components/esp_system/task_wdt.c:176 (discriminator 3)
0x4037871a: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111
0x420087c3: usb_serial_jtag_fsync at /esp-idf/components/vfs/vfs_usb_serial_jtag.c:264
0x42008258: console_fsync at /esp-idf/components/vfs/vfs_console.c:107
0x420076e8: esp_vfs_fsync at /esp-idf/components/vfs/vfs.c:565 (discriminator 4)
0x420b26b7: flushWrite at /esp-idf/components/console/linenoise/linenoise.c:220
0x420b3b87: linenoiseRaw at /esp-idf/components/console/linenoise/linenoise.c:1084
0x420b3bd5: linenoise at /esp-idf/components/console/linenoise/linenoise.c:1134
0x4200f657: pd_console_api_enter_loop() at /pd_console_api.cpp:412 (discriminator 3)
0x4200da27: app_main at /pd_main.cpp:881 (discriminator 11)
0x42134957: main_task at /esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)
0x40387a0e: vPortTaskWrapper at /esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 3cd4856bc8447e8f
```

davidzuhn, 2022-07-17:
I have been using the USB serial/jtag controller as a console for a bit now on ESP32-S3, albeit not without some problems.   The best place to start is the system/console/basic example (definitely NOT the advanced_usb_cdc code).   Set ESP_CONSOLE_USB_SERIAL_JTAG=y in the sdkconfig, and then build & run.

This is only in quite recent revisions of esp-idf:latest.    You can find the proper initialization code in the main/console_example_main.c file, in a set of #ifdef code blocks for UART vs. the serial/jtag controller.   Then find the device appropriate for $ESPPORT (on my Mac, the first one plugged in will be /dev/cu.usbmodem101.  

I find the implementation to be buggy yet -- I've got an issue reported where lines can be lost in the output.   As I run the console/basic example at the moment, the prompt output is regularly being lost but the input is being taken (but not echoed) and executed.   It's functional, but definitely not fully so yet (imo).

igrr, 2022-07-17:
FWIW, the fixes to the console/basic example for USB_SERIAL_JTAG support (excluding the skipped characters issue which @davidzuhn has reported) have been backported to release/v4.4 branch in https://github.com/espressif/esp-idf/commit/6f57018c9d5956774a8f0d071e5e60ba0d6f94fa. This commit will be part of v4.4.2 release.

chipweinberger, 2022-07-18:
Thanks for for the responses. 

I'll look forward to 4.4.2.

Just btw, the great support & community for the ESP32 family is the main reason I chose this ecosystem for my product. So thanks for the always prompt replies.

chipweinberger, 2022-10-12:
I can confirm Usb Serial Jtag Console works in 4.4.2

You must set the primary console to USB Serial Jtag in menuconfig, then compile the basic console example.

closed 2022-10-12 by chipweinberger