espressif/esp-idf #9371: Firmware freezes on http_send

kotyara12, 2022-07-17:
Good afternoon!

I can't figure out what the problem is.

In my firmware, there are several services to which data is transmitted using a GET request: ThingSpeak, OpenMonitoring, NarodMon. Sending data is carried out by a task specially created for this purpose. The data is transmitted using httpS, a request, that is, using mbedTLS (I suspect that it is she who is the one who slows down).

Full code here: https://github.com/kotyara12/reDataSend/blob/master/src/reDataSend.cpp

Periodically, the device freezes, while "offline" tasks continue to work, but network tasks - mqtt, telegram, ota, http_send hang tightly and do not respond. PING is perfectly normal at this time and other devices continue to work.

In serial at this time such crap:

```
[0;31mE (13310633) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m

[0;31mE (13310633) task_wdt:  - IDLE (CPU 1)[0m

[0;31mE (13310633) task_wdt: Tasks currently running:[0m

[0;31mE (13310633) task_wdt: CPU 0: IDLE[0m

[0;31mE (13310633) task_wdt: CPU 1: **http_send**[0m

[0;31mE (13310633) task_wdt: Print CPU 0 (current core) backtrace[0m

Backtrace:0x40167E56:0x3FFB09B00x40083D31:0x3FFB09E0 0x40181E73:0x3FFCE7E0 0x400ED199:0x3FFCE800 0x40090FEF:0x3FFCE820 

[0;31mE (13310633) task_wdt: Print CPU 1 backtrace[0m

Backtrace:0x4008A085:0x3FFB0FC00x40083D31:0x3FFB0FE0 0x4008B3D3:0x3FFC0780 0x401153A5:0x3FFC0790 0x401154DD:0x3FFC07C0 0x40115A91:0x3FFC07E0 0x401213E1:0x3FFC0810 0x40121665:0x3FFC0840 0x4011F27B:0x3FFC08D0 0x4012055E:0x3FFC0900 0x401205ED:0x3FFC0930 0x4011D855:0x3FFC0960 0x4011D4FA:0x3FFC09A0 0x4012BD05:0x3FFC09D0 0x4012A68C:0x3FFC0A90 0x401297D5:0x3FFC0AB0 0x401297F8:0x3FFC0AD0 0x4012981E:0x3FFC0AF0 0x400E7B81:0x3FFC0B10 0x400E7555:0x3FFC0B30 0x400E79A9:0x3FFC0BB0 0x401083E8:0x3FFC0BE0 0x4017FF19:0x3FFC0C10 0x400E8A82:0x3FFC0C30 0x400E99CB:0x3FFC0C50 0x400D9BD7:0x3FFC0C70 0x400D9D23:0x3FFC0D20 
```

I tried to greatly increase the time for task_wdt, but this did not affect the situation in any way.

Current mbedTLS settings (tried different options, but the result is the same):

```
#
# mbedTLS
#
# CONFIG_MBEDTLS_INTERNAL_MEM_ALLOC is not set
# CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC is not set
CONFIG_MBEDTLS_DEFAULT_MEM_ALLOC=y
# CONFIG_MBEDTLS_CUSTOM_MEM_ALLOC is not set
CONFIG_MBEDTLS_ASYMMETRIC_CONTENT_LEN=y
CONFIG_MBEDTLS_SSL_IN_CONTENT_LEN=16384
CONFIG_MBEDTLS_SSL_OUT_CONTENT_LEN=4096

CONFIG_MBEDTLS_DYNAMIC_BUFFER=y
CONFIG_MBEDTLS_DYNAMIC_FREE_PEER_CERT=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CONFIG_DATA=y
CONFIG_MBEDTLS_DYNAMIC_FREE_CA_CERT=y

# CONFIG_MBEDTLS_DEBUG is not set

#
# mbedTLS v2.28.x related
#
# CONFIG_MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH is not set
# CONFIG_MBEDTLS_X509_TRUSTED_CERT_CALLBACK is not set
# CONFIG_MBEDTLS_SSL_CONTEXT_SERIALIZATION is not set
# end of mbedTLS v2.28.x related

#
# Certificate Bundle
#
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE is not set
# end of Certificate Bundle

# CONFIG_MBEDTLS_ECP_RESTARTABLE is not set
# CONFIG_MBEDTLS_CMAC_C is not set
CONFIG_MBEDTLS_HARDWARE_AES=y
CONFIG_MBEDTLS_HARDWARE_MPI=y
CONFIG_MBEDTLS_HARDWARE_SHA=y
CONFIG_MBEDTLS_ROM_MD5=y
# CONFIG_MBEDTLS_ATCA_HW_ECDSA_SIGN is not set
# CONFIG_MBEDTLS_ATCA_HW_ECDSA_VERIFY is not set
CONFIG_MBEDTLS_HAVE_TIME=y
# CONFIG_MBEDTLS_HAVE_TIME_DATE is not set
CONFIG_MBEDTLS_ECDSA_DETERMINISTIC=y
CONFIG_MBEDTLS_SHA512_C=y
CONFIG_MBEDTLS_TLS_SERVER_AND_CLIENT=y
# CONFIG_MBEDTLS_TLS_SERVER_ONLY is not set
# CONFIG_MBEDTLS_TLS_CLIENT_ONLY is not set
# CONFIG_MBEDTLS_TLS_DISABLED is not set
CONFIG_MBEDTLS_TLS_SERVER=y
CONFIG_MBEDTLS_TLS_CLIENT=y
CONFIG_MBEDTLS_TLS_ENABLED=y
```

ESP-IDF Version 4.4.1

KaeLL, 2022-07-17:
[Decode](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#automatic-address-decoding) the backtrace, please.

kotyara12, 2022-07-17:
Oh sure. Sorry I forgot to add...

```
c:\Users\kotyara12\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\PlatformIO\security_mo53\ota\firmware\firmware.elf 0x4008A085:0x3FFB0FC00x40083D31:0x3FFB0FE0 0x4008B3D3:0x3FFC0780 0x401153A5:0x3FFC0790 0x401154DD:0x3FFC07C0 0x40115A91:0x3FFC07E0 0x401213E1:0x3FFC0810 0x40121665:0x3FFC0840 0x4011F27B:0x3FFC08D0 0x4012055E:0x3FFC0900 0x401205ED:0x3FFC0930 0x4011D855:0x3FFC0960 0x4011D4FA:0x3FFC09A0 0x4012BD05:0x3FFC09D0 0x4012A68C:0x3FFC0A90 0x401297D5:0x3FFC0AB0 0x401297F8:0x3FFC0AD0 0x4012981E:0x3FFC0AF0 0x400E7B81:0x3FFC0B10 0x400E7555:0x3FFC0B30 0x400E79A9:0x3FFC0BB0 0x401083E8:0x3FFC0BE0 0x4017FF19:0x3FFC0C10 0x400E8A82:0x3FFC0C30 0x400E99CB:0x3FFC0C50 0x400D9BD7:0x3FFC0C70 0x400D9D23:0x3FFC0D20

0x4008a085: esp_crosscore_isr at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_system/crosscore_int.c:92
0x4008b3d3: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:176
0x401153a5: mbedtls_mpi_sub_abs at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/bignum.c:1235
0x401154dd: mbedtls_mpi_sub_mpi at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/bignum.c:1312
0x40115a91: mbedtls_mpi_mod_mpi at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/bignum.c:1815
 (inlined by) mbedtls_mpi_mod_mpi at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/bignum.c:1799
0x401213e1: calculate_rinv at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\port/esp_bignum.c:130 (discriminator 2)
0x40121665: esp_mpi_exp_mod at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\port/esp_bignum.c:318
 (inlined by) mbedtls_mpi_exp_mod at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\port/esp_bignum.c:373
0x4011f27b: mbedtls_rsa_public at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/rsa.c:747
0x4012055e: mbedtls_rsa_rsassa_pkcs1_v15_verify at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/rsa.c:2229
0x401205ed: mbedtls_rsa_pkcs1_verify at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/rsa.c:2287
0x4011d855: rsa_verify_wrap at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/pk_wrap.c:93
0x4011d4fa: mbedtls_pk_verify_restartable at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/pk.c:319
0x4012bd05: ssl_parse_server_key_exchange at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_cli.c:3402
 (inlined by) mbedtls_ssl_handshake_client_step at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_cli.c:4459
0x4012a68c: __wrap_mbedtls_ssl_handshake_client_step at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\port\dynamic/esp_ssl_cli.c:186 (discriminator 15)
0x401297d5: mbedtls_ssl_handshake_step at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:5836
0x401297f8: mbedtls_ssl_handshake at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:5873
0x4012981e: mbedtls_ssl_handshake at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:5855 (discriminator 1)
0x400e7b81: esp_mbedtls_handshake at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp-tls/esp_tls_mbedtls.c:172
0x400e7555: esp_tls_handshake at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp-tls/esp_tls.c:79 (discriminator 15)
 (inlined by) esp_tls_low_level_conn at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp-tls/esp_tls.c:426 (discriminator 15)
0x400e79a9: esp_tls_conn_new_sync at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp-tls/esp_tls.c:488
0x401083e8: ssl_connect at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\tcp_transport/transport_ssl.c:111
0x4017ff19: esp_transport_connect at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\tcp_transport/transport.c:173
0x400e8a82: esp_http_client_connect at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_http_client/esp_http_client.c:1245
 (inlined by) esp_http_client_connect at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_http_client/esp_http_client.c:1217
0x400e99cb: esp_http_client_perform at C:\Users\kotyara12\.platformio\packages\framework-espidf\components\esp_http_client/esp_http_client.c:1097
0x400d9bd7: dsCallDataApi(dataChannel_t*) at C:/PlatformIO/libs/clouds/reDataSend/src/reDataSend.cpp:321
 (inlined by) dsCallDataApi(dataChannel_t*) at C:/PlatformIO/libs/clouds/reDataSend/src/reDataSend.cpp:302
0x400d9d23: dsTaskExec(void*) at C:/PlatformIO/libs/clouds/reDataSend/src/reDataSend.cpp:483
```

kotyara12, 2022-07-18:
I think I found the source of the problem...

The fact is that on devices with confirmed freezes, memory debugging (heap traicing standalone) is enabled. Yesterday I turned it off on one of them, I check it. As long as there are no freezes...

But in general, this is not very good if the tracing seems to be there, but it cannot be used

kotyara12, 2022-07-19:
I have a guess....
It seems to me that freezes occur more often on those devices that send data to different external services with different TLS certificates.
If the device sends data to only one service, then it works stably.
But for now, this is just speculation and information for consideration.

laukik-hase, 2022-07-22:
Hello, @kotyara12!

Thank you for the issue report. Could you please provide the following info -

- Target chip or module (ESP32, ESP32-S2, ESP32-S3, etc.)
- ESP-IDF version (run `git describe --tags`) / commit ID
- Example to reproduce the issue (or a minimal code snippet)
- Debug logs

Could you try disabling `CONFIG_MBEDTLS_HARDWARE_MPI` and check whether the firmware crashes?

kotyara12, 2022-07-22:
Good day!

- ESP32. The problem is observed on Espressif ESP32-WROVER-IE 16MB (ESP32-D0WD-V3) and occasionally on ESP32-WROOM-32D/U
- PlatformIO Espressif 32 5.0.0 & ESP-IDF 4.4.1
- https://github.com/kotyara12/reDataSend/blob/master/src/reDataSend.cpp#L319-L339
- Bug report above, in the first post

kotyara12, 2022-07-22:
Some statistics. It might help.

1. Main module. ESP32-WROVER-IE 16MB. Sends data to OpenMon and ThingSpeak. Hangs up about once a day.

2. Bathroom, ventilation control. ESP32-WROOM-32U. Sends data to OpenMon and ThingSpeak. Hangs up about once a week.

3. Weather station. ESP32-WROOM-32D. Sends data to NarodMon, OpenMon and ThingSpeak. So far there haven't been any hangups.

The sdkconfig settings for 2 and 3 are identical, for 1 they are slightly different because the flash and ram sizes are different.

laukik-hase, 2022-07-22:
Hello, @kotyara12!

Could you please -
- Attach the entire example `sdkconfig`
- Try disabling `CONFIG_MBEDTLS_HARDWARE_MPI` and check whether the firmware crashes

kotyara12, 2022-07-22:
- No problem - https://disk.yandex.ru/d/a-42xNKEc5Awqw
- Thanks, I'll try

AxelLin, 2022-08-02:
I also hit the same issue with CONFIG_MBEDTLS_HARDWARE_MPI=y on v5.0-dev-4549-gb66be87f88.
Use a endless while toop to send POST request every 30 seconds should be able to reproduce the issue.

E (202191) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (202191) task_wdt:  - IDLE (CPU 1)
E (202191) task_wdt: Tasks currently running:
E (202191) task_wdt: CPU 0: IDLE
E (202191) task_wdt: CPU 1: http_cli
E (202191) task_wdt: Aborting.

abort() was called at PC 0x400fec89 on core 0
0x400fec89: task_wdt_isr at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:385 (discriminator 3)

Backtrace: 0x40081e8a:0x3ffb1e70 0x40089acd:0x3ffb1e90 0x400906ea:0x3ffb1eb0 0x400fec89:0x3ffb1f20 0x40082a11:0x3ffb1f40 0x40087ef3:0x3ffbcb10 0x400d52e6:0x3ffbcb30 0x4008ada8:0x3ffbcb50
0x40081e8a: panic_abort at /home/axel/esp/esp-idf/components/esp_system/panic.c:408

0x40089acd: esp_system_abort at /home/axel/esp/esp-idf/components/esp_system/esp_system.c:135

0x400906ea: abort at /home/axel/esp/esp-idf/components/newlib/abort.c:38

0x400fec89: task_wdt_isr at /home/axel/esp/esp-idf/components/esp_system/task_wdt.c:385 (discriminator 3)

0x40082a11: _xt_lowint1 at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1117

0x40087ef3: xt_utils_wait_for_intr at /home/axel/esp/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/axel/esp/esp-idf/components/esp_hw_support/cpu.c:101

0x400d52e6: esp_vApplicationIdleHook at /home/axel/esp/esp-idf/components/esp_system/freertos_hooks.c:59

0x4008ada8: prvIdleTask at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4157

AxelLin, 2022-08-02:
Below is the latest mbedtls message I got, it then seems freezes.

I (6427822) mbedtls: ssl_tls.c:5312 => handshake
I (6427822) mbedtls: ssl_msg.c:1789 => flush output
I (6427822) mbedtls: ssl_msg.c:1801 <= flush output
I (6427828) mbedtls: ssl_cli.c:4192 client state: 0
I (6427834) mbedtls: ssl_msg.c:1789 => flush output
I (6427839) mbedtls: ssl_msg.c:1801 <= flush output
I (6427845) mbedtls: ssl_cli.c:4192 client state: 1
I (6427851) mbedtls: ssl_cli.c:933 => write client hello
I (6427860) mbedtls: ssl_msg.c:2218 => write handshake message
I (6427864) mbedtls: ssl_msg.c:2369 => write record
I (6427871) mbedtls: ssl_msg.c:1789 => flush output
I (6427875) mbedtls: ssl_msg.c:1807 message length: 241, out_left: 241
I (6427884) mbedtls: ssl_msg.c:1814 ssl->f_send() returned 241 (-0xffffff0f)
I (6427890) mbedtls: ssl_msg.c:1842 <= flush output
I (6427896) mbedtls: ssl_msg.c:2514 <= write record
I (6427902) mbedtls: ssl_msg.c:2346 <= write handshake message
I (6427908) mbedtls: ssl_cli.c:1346 <= write client hello
I (6427915) mbedtls: ssl_msg.c:1789 => flush output
I (6427920) mbedtls: ssl_msg.c:1801 <= flush output
I (6427926) mbedtls: ssl_msg.c:1573 => fetch input
I (6427932) mbedtls: ssl_msg.c:1728 in_left: 0, nb_want: 5
I (6428943) mbedtls: ssl_msg.c:1753 in_left: 0, nb_want: 5
I (6428944) mbedtls: ssl_msg.c:1756 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (6428948) mbedtls: ssl_msg.c:1776 <= fetch input
I (6428953) mbedtls: ssl_cli.c:4192 client state: 2
I (6428959) mbedtls: ssl_cli.c:1919 => parse server hello
I (6428965) mbedtls: ssl_msg.c:3546 => read record
I (6428971) mbedtls: ssl_msg.c:1573 => fetch input
I (6428976) mbedtls: ssl_msg.c:1728 in_left: 5, nb_want: 5
I (6428983) mbedtls: ssl_msg.c:1776 <= fetch input
I (6428989) mbedtls: ssl_msg.c:1573 => fetch input
I (6428994) mbedtls: ssl_msg.c:1728 in_left: 5, nb_want: 102
I (6429001) mbedtls: ssl_msg.c:1753 in_left: 5, nb_want: 102
I (6429007) mbedtls: ssl_msg.c:1756 ssl->f_recv(_timeout)() returned 97 (-0xffffff9f)
I (6429016) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429022) mbedtls: ssl_msg.c:3620 <= read record
I (6429028) mbedtls: ssl_cli.c:2216 server hello, total extension length: 21
I (6429035) mbedtls: ssl_cli.c:2428 <= parse server hello
I (6429041) mbedtls: ssl_msg.c:1789 => flush output
I (6429047) mbedtls: ssl_msg.c:1801 <= flush output
I (6429053) mbedtls: ssl_msg.c:1573 => fetch input
I (6429058) mbedtls: ssl_msg.c:1728 in_left: 0, nb_want: 5
I (6429065) mbedtls: ssl_msg.c:1753 in_left: 0, nb_want: 5
I (6429071) mbedtls: ssl_msg.c:1756 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (6429080) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429085) mbedtls: ssl_cli.c:4192 client state: 3
I (6429091) mbedtls: ssl_tls.c:2291 => parse certificate
I (6429097) mbedtls: ssl_msg.c:3546 => read record
I (6429103) mbedtls: ssl_msg.c:1573 => fetch input
I (6429108) mbedtls: ssl_msg.c:1728 in_left: 5, nb_want: 5
I (6429115) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429120) mbedtls: ssl_msg.c:1573 => fetch input
I (6429126) mbedtls: ssl_msg.c:1728 in_left: 5, nb_want: 4003
I (6429134) mbedtls: ssl_msg.c:1753 in_left: 5, nb_want: 4003
I (6429139) mbedtls: ssl_msg.c:1756 ssl->f_recv(_timeout)() returned 3998 (-0xfffff062)
I (6429148) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429176) mbedtls: ssl_msg.c:3620 <= read record
I (6429184) mbedtls: ssl_tls.c:2402 <= parse certificate
I (6429185) mbedtls: ssl_msg.c:1789 => flush output
I (6429185) mbedtls: ssl_msg.c:1801 <= flush output
I (6429191) mbedtls: ssl_msg.c:1573 => fetch input
I (6429197) mbedtls: ssl_msg.c:1728 in_left: 0, nb_want: 5
I (6429204) mbedtls: ssl_msg.c:1753 in_left: 0, nb_want: 5
I (6429210) mbedtls: ssl_msg.c:1756 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (6429218) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429224) mbedtls: ssl_cli.c:4192 client state: 4
I (6429230) mbedtls: ssl_cli.c:2879 => parse server key exchange
I (6429236) mbedtls: ssl_msg.c:3546 => read record
I (6429242) mbedtls: ssl_msg.c:1573 => fetch input
I (6429248) mbedtls: ssl_msg.c:1728 in_left: 5, nb_want: 5
I (6429254) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429260) mbedtls: ssl_msg.c:1573 => fetch input
I (6429265) mbedtls: ssl_msg.c:1728 in_left: 5, nb_want: 305
I (6429272) mbedtls: ssl_msg.c:1753 in_left: 5, nb_want: 305
I (6429278) mbedtls: ssl_msg.c:1756 ssl->f_recv(_timeout)() returned 300 (-0xfffffed4)
I (6429287) mbedtls: ssl_msg.c:1776 <= fetch input
I (6429295) mbedtls: ssl_msg.c:3620 <= read record
I (6429301) mbedtls: ssl_cli.c:2498 ECDH curve: x25519
I (6429305) mbedtls: ssl_cli.c:2806 Server used SignatureAlgorithm 1
I (6429312) mbedtls: ssl_cli.c:2808 Server used HashAlgorithm 4

kotyara12, 2022-08-23:
Hello, @laukik-hase!

Disabling '''CONFIG_MBEDTLS_HARDWARE_MPI''' solved the problem, devices work stably.

Sorry for the long reply - I was away....

AxelLin, 2022-08-23:
> Hello, @laukik-hase!
> 
> Disabling `CONFIG_MBEDTLS_HARDWARE_MPI` solved the problem, devices work stably.

The same result with my test.
It's a regression with CONFIG_MBEDTLS_HARDWARE_MPI=y  and needs fix.
(It was working fine when test CONFIG_MBEDTLS_HARDWARE_MPI=y  with older esp-idf versions.)

laukik-hase, 2022-08-23:
Hello, @AxelLin and @kotyara12!

Thank you for the info. I have been investigating this issue (and similar ones reported in #8710 and #9256) and tried various experiments to reproduce it on my side but to no avail. 

- Starting from the `esp_http_client` example as a base, I have tried sync/async HTTPS connections firing `GET`/`POST`/`PUT` requests in a `while(1)` loop for about 4-5 hours. 
- To verify the working of the MPI module, I created a minimal example for multiplying two very large numbers and verifying their result and RSA certificate operations in an endless `while` loop.
- I tried out several server endpoints with ciphersuite restrictions  (RSA-based ones -> `TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384`).

> It's a regression with CONFIG_MBEDTLS_HARDWARE_MPI=y and needs fix.
(It was working fine when test CONFIG_MBEDTLS_HARDWARE_MPI=y with older esp-idf versions.)

This issue has been reported on the `master` branch as well as the `release/v4.4` branch.
- Could you check if your application is working as expected on the `release/v4.3` branch? 
- Could you also provide a minimal example that reproduces the issue with the corresponding `sdkconfig`?

CC: @mahavirj @ESP-Marius

AxelLin, 2022-08-23:
@laukik-hase @mahavirj @ESP-Marius 

I can reproduce it by sending post requests to webhook.site .
Maybe you can update your code with a endless loop to repeat sending post request.  (Don't enable keepalive by setting "Connection: close" header)
BTW, I didn't hit the issue on v4.4, the issue only happened on master tree in my tests.

AxelLin, 2022-09-08:
Hi @laukik-hase @mahavirj @ESP-Marius 
Just to confirm if you can reproduce the issue now?

laukik-hase, 2022-09-08:
Hello, @AxelLin!

Sorry, but I am still not able to reproduce it on the `master` branch using the configuration and endpoint you mentioned above. PFA my logs for a one-hour long run - [esp32_github_mpi.zip](https://github.com/espressif/esp-idf/files/9526626/esp32_github_mpi.zip)

AxelLin, 2022-09-08:
Hi @laukik-hase 

I just check your log and compare with my test device:
You tested on v4.4.1 with ESP32-D0WD-V3 (revision 3)
I tested on v5.1-dev-644-g867745a05c with ESP32-D0WD (revision 1) which just hit the issue.

I'm not allow to post my code in work.
Is it possible sharing your testing code so I will check if I can reproduce it on my device?
Maybe I can reproduce it with some modification and then update to you.

AxelLin, 2022-09-14:
@laukik-hase 

I try to modify esp_http_client example code to reproduce this issue.
But the issue cannot be reproduced with the example code.
However, I can still reproduce this issue on my real project.
The symptom is slightly different with v5.1-dev-736-gd6d8324ad9 now,
it does not trigger WDT but it stuck at esp_http_client_open().
So this is a more complex issue.

laukik-hase, 2022-09-14:
Hello, @AxelLin!

Thank you for the comment.

- Following is the example code from my testing. I have been running my test setup for more than 24 hours now and about 34000+ POST requests were made but no task watchdog was triggered. 
- PFA the logs from my test setup: [2022-13-09-095416_esp32_http_mpi.zip](https://github.com/espressif/esp-idf/files/9562749/2022-13-09-095416_esp32_http_mpi.zip). I'll update you once again after the setup has run for some more time.
- If you cannot provide the code, could you please give some minimal example which reproduces it? Are there any other tasks running and what are their priorities?

<details>
  <summary> Code Snippet </summary>

```
static void https_async_post_req(void)
{
    char post_resp_buf[MAX_HTTP_OUTPUT_BUFFER] = {0};
    esp_err_t err = ESP_FAIL;

    esp_http_client_config_t config_post = {
        .url = "https://httpbin.org/post",
        .event_handler = _http_event_handler,
        .crt_bundle_attach = esp_crt_bundle_attach,
        .method = HTTP_METHOD_POST,
        .user_data = post_resp_buf,
        .is_async = true,
        .timeout_ms = 5000,
    };
    esp_http_client_handle_t client_post = esp_http_client_init(&config_post);

    const char *post_data = "Hello from ESP32!";

    esp_http_client_set_post_field(client_post, post_data, strlen(post_data));
    esp_http_client_set_header(client_post, "Content-Type", "application/text");
    esp_http_client_set_header(client_post, "Connection", "close");

    while(1) {
        err = esp_http_client_perform(client_post);
        if (err == ESP_ERR_HTTP_EAGAIN) {
            continue;
        } else if (err == ESP_OK) {
            ESP_LOGI(TAG, "HTTPS Status = %d, content_length = %lld",
                    esp_http_client_get_status_code(client_post),
                    esp_http_client_get_content_length(client_post));
        } else {
            ESP_LOGE(TAG, "Error perform http request %s", esp_err_to_name(err));
        }
        break;
    }

    ESP_LOGI(TAG, "Done with POST request!");
    esp_http_client_cleanup(client_post);
}

static void http_test_task(void *pvParameters)
{
    int request_count = 0;
    int64_t t0 = 0, t1 = 0;

    while(1) {
        t0 = esp_timer_get_time();
        https_async_post_req();
        t1 = esp_timer_get_time();

        ESP_LOGW(TAG, "[POST] Request #%d done in %lld us!", ++request_count, t1 - t0);
    }
}
```

</details>

AxelLin, 2022-09-14:
Hi @laukik-hase 

Attached the code to reproduce the issue.
1. Copy esp_http_client_example.c.txt to examples/protocols/esp_http_client/main/esp_http_client_example.c 
   Edit below line and change url, you can create a new one on wehook.site for test.
   esp_http_client_set_url(client_post, "https://webhook.site/b31714cb-ae4d-4b24-82c1-b215c6e83a71");

2. Copy sdkconfig.txt to examples/protocols/esp_http_client/sdkconfig
   You need to change below settings to join your Access Point.
     CONFIG_EXAMPLE_WIFI_SSID="TEST_AP"
     CONFIG_EXAMPLE_WIFI_PASSWORD="1234223432"

The test code will keep sending post request, and it will reboot automatically after sending 50 requests.
It's not 100% reproducable, so I make the device reboot after sending 50 requests.
Which means if you observe the console output STOP, it means the issue happened.
(As I mentioned in my previous comment, it no longer triger WDT with v5.1-dev-736-gd6d8324ad9, it just STOP sending request.)

It may take some time to see the result.
Please check if you can reproduce the issue.
[esp_http_client_example.c.txt](https://github.com/espressif/esp-idf/files/9564529/esp_http_client_example.c.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9564530/sdkconfig.txt)

AxelLin, 2022-09-20:
@laukik-hase Can you reproduce the issue?

laukik-hase, 2022-09-20:
Hello, @AxelLin!

I can reproduce the issue now and am looking into it. The comment https://github.com/espressif/esp-idf/issues/8710#issuecomment-1249178698 seems to be a reasonable hypothesis.

klew, 2022-10-18:
Hi

I also encountered similar problem as reported here. I use esp_tls_conn_new_sync method to establish secured connection with my server and remaining part of callstack is almost the same as in first post here.
In my case connection hang randomly after 1-3 h of continues connect/disconnect scenario and it happened when server was using following key:

> New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
> Server public key is 2048 bit

When I tried similar setup, however with 4096 bit key, everything was working fine. Problem occured on few different servers with 2048 key and it never occuret on few other servers which were using 4096 bit key - however this may not be the only difference.

I use: Chip is ESP32-U4WDH (revision v3.0)
On: ESP-IDF: v5.1-dev-16-g30e8f19f5a-dirty

Workaround with disabled CONFIG_MBEDTLS_HARDWARE_MPI worked fine.
Additionally disabling this hardware acceleration speed up establishing connection by 100-300 ms (depending on RSA key size - ~300 ms gain on 2048 bit key and ~100 ms gain on 4096 bit key).

mahavirj, 2022-10-19:
@kotyara12 @klew 

Please try out fix suggested in https://github.com/espressif/esp-idf/issues/8710#issuecomment-1283534267

klew, 2022-10-19:
@mahavirj thanks, I'll try those corrections and I'll let you know about results, but I'll need few days to check it.

I guess that this correction won't speed up calculations. What's the point of using hardware accelerated calculations, when it is actually slower compared to disabled "acceleration"?

mahavirj, 2022-10-20:
@klew 

Please check out test code that we have for some of the MPI operations at [test_mbedtls_mpi.c](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/test_apps/main/test_mbedtls_mpi.c) and [test_rsa.c](https://github.com/espressif/esp-idf/blob/master/components/mbedtls/test_apps/main/test_rsa.c). Please share code if you are observing different behavior with respect to supported operations (like modular multiplication, modular exp) in the hardware peripheral and its performance, I will take a look.

Additionally, using hardware accelerator will have an added advantage of using lesser memory (stack) in most of the cases.

closed 2022-10-27 by espressif-bot