espressif/esp-idf #9373: SNTP and Time zone

KwangryeolPark, 2022-07-17:
Hello, I am testing SNTP functions and setenv, tzset function to adjust ESP32 time.

I leave in Korean, which time zone is UTC+9. 

Where is my code 
```
...
    setenv("TZ", "UTC+9", 1);
    tzset();

    //  Init ntp
    sntp_set_sync_mode(SNTP_SYNC_MODE_IMMED);
    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_setservername(0, "pool.ntp.org");
    sntp_setservername(1, "time.google.com");
    sntp_init();

    time_t now;
    struct tm *timeInfo;
    char strftime_buf[64] = {0};

    while (true)
    {
        time(&now); //  Get current time
        switch (sntp_get_sync_status())
        {
        case SNTP_SYNC_STATUS_RESET:
            ESP_LOGW(TAG, "Reset status");
            break;
        case SNTP_SYNC_STATUS_COMPLETED:
            ESP_LOGI(TAG, "Completed");
            break;
        case SNTP_SYNC_STATUS_IN_PROGRESS:
            ESP_LOGW(TAG, "In progress");
            break;
        }

        timeInfo = localtime(&now);
        strftime(strftime_buf, sizeof(strftime_buf), "%Y-%m-%d %H:%M:%S", timeInfo);
        ESP_LOGI(TAG, "The current time: %s", strftime_buf);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
...
```

After sync, it show 2022-07-17 06:33:10, but the real time is 2022-07-18 00:33:10.

What I realize is when I set time zone to UTC-9, it shows correct time. However the Korea time zone is UTC+9, so I think the sign is wrong.

KwangryeolPark, 2022-07-17:
I mean if I want to get UTC+9 time zone, I have to set TZ to UTC-9...

igrr, 2022-07-17:
Hi @KwangryeolPark,

The TZ variable formats supported by Newlib C library are those specified in POSIX.1 standard. The meaning of the string is different from what you expect. Quoting,

> The first format is used when there is no Daylight Saving Time (or summer time) in the local time zone:
>    `std offset`
> The `std` string specifies the name of the time zone. It must be three or more characters long and must not contain a leading colon, embedded digits, commas, nor plus and minus signs. There is no space character separating the time zone name from the offset, so these restrictions are necessary to parse the specification correctly.
> The offset specifies the time value you must **add** to the local time to get a Coordinated Universal Time value. It has syntax like [+|-]hh[:mm[:ss]]. This is positive if the local time zone is west of the Prime Meridian and negative if it is east. The hour must be between 0 and 24, and the minute and seconds between 0 and 59.
> For example, here is how we would specify Eastern Standard Time, but without any Daylight Saving Time alternative: EST+5

(From https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html)

So for Korea timezone the string should be `KST-9`.

closed 2023-09-27 by KwangryeolPark