espressif/esp-idf #9376: Bluetooth Classic Legacy Pairing WITHOUT Pin Code

qombajn, 2022-07-17:
Hi everyone!
I want to use Esp32 as a2dp-sink. It's working in Legacy Pairing mode becouse i've unchecked Secure Simple Pairing and it works perfectly. Problem is that i will be sending pincode via SPI and if pincode will be all "0"s it should be no request for pincode in android window. I don't know how to disable this request. I can have this functionality in Secure Simple Pairing but I don't know how to do it in Legacy Pairing. My idea is to break waiting for request if pincode is "00000". But i need little help.

Wth-Esp, 2022-07-18:
Hi, @qombajn 
PinCode is the mandatory requirement in the legacy pairing. Eventhough, you set the PIN Code into `000000`, the 6 digits are also the meaningful pin code to let Bluetooth start the legacy pairing process.

I'm afraid your request is not allowed by the Bluetooth Spec.

[image]

qombajn, 2022-07-18:
And is it impossible to change Bluetooth Spec? I mean we have to check if Pincode is "0000" for example and skip the request and just connect.

closed 2022-07-20 by Alvin1Zhang