espressif/esp-idf #9377: i2c_set_pin no function on ESP32-C3?

Kampi, 2022-07-18:
## Environment

- Development Kit: ESP-C3-32S
- Module or chip used: ESP32-C3
- IDF version (run ``git describe --tags`` to find it): 4.4.1
- Build System:  CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0+2021r2-patch3

## Problem Description / Actual Behavior

I need to switch the pins for I2C because I have two I2C slaves with the same address on my PCB, but I can not change the slave address. So I use `i2c_set_pin` before every transaction to choose the right bus. 

```
I2CM_Init();
...
I2CM_SwitchPins(_Instance, PIN_I2C_SDA_1, PIN_I2C_SCL_1);
... Do stuff
I2CM_SwitchPins(_Instance, PIN_I2C_SDA_2, PIN_I2C_SCL_2);
... Do stuff

But it looks like the old pins aren´t disconnected from the bus, because I have I2C communication on the first bus when I switch to the second bus.
I have worked around the problem by setting the GPIOs manually as output:

```
int I2CM_SwitchPins(i2c_port_t Instance, int SDA, int SCL)
{
    ESP_LOGI(TAG, "Switch I2C pins to: SDA: %u, SCL: %u", SDA, SCL);

    // Weird bug in I2C driver?
    // Need to disable the old SDA and SCL pins before chaning it.
    _IO_Conf.pin_bit_mask = (1ULL << _SCL) | (1ULL << _SDA);
    gpio_config(&_IO_Conf);

    if(i2c_set_pin(Instance, SDA, SCL, _I2CM_Config.sda_pullup_en, _I2CM_Config.scl_pullup_en, _I2CM_Config.mode))
    {
        return -1;
    }

    return 0;
}
```

### Expected Behavior

No need to change the pins to output by hand.

o-marshmallow, 2022-07-21:
Hi @Kampi ,

`i2c_set_pin` was not meant to be called after installing the driver. A cleaner way to switch pins at runtime would be to define your `I2CM_SwitchPins` as:
```
int I2CM_SwitchPins(i2c_port_t Instance, int SDA, int SCL)
{
    ESP_LOGI(TAG, "Switch I2C pins to: SDA: %u, SCL: %u", SDA, SCL);

    gpio_reset_pin(_SCL);
    gpio_reset_pin(_SDA);

    if(i2c_set_pin(Instance, SDA, SCL, _I2CM_Config.sda_pullup_en, _I2CM_Config.scl_pullup_en, _I2CM_Config.mode))
    {
        return -1;
    }

    return 0;
}
```

On a more general note. There are other ways to deal with two components having the same I2C address. For example, it would be possible to install a bus multiplexer (PCA9540), which would "divide" the I2C bus into two sub-buses.
Another way would be to power on the chip that you want to communicate to with a GPIO (and power off the other one), so that only one of them is on at the same time.

Kampi, 2022-07-22:
Hi @o-marshmallow,

I have thought about these ideas before and the bus multiplexer wasn´t an option, because the device had to be small and cheap, so it´s not an option.
Powering only one device was an alternative idea, especially because the sensors can be powered through a GPIO (TI has documented that in her datasheet). The amount of GPIO usage would be the same, so I pick randomly one of these two choices and I choose the two different buses :) 

I will try your reset approach and also my workaround is working, so no real problem. I just thought that I have discovered a bug in the C3 driver. Thanks anyway :)

closed 2022-07-22 by Kampi