espressif/esp-idf #9378: RTC_WDT, IWDT, TWDT

franz-ms-muc, 2022-07-18:
**Is your feature request related to a problem? Please describe.**

Hello, 

i see the 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html 
for IWDT, TWDT
Interrupt Watchdog Timer (IWDT)
Task Watchdog Timer (TWDT)

and i see the 

https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/include/rtc_wdt.h 
for RTC_WDT

**Describe the solution you'd like**

A clear and concise description of the relation of IWDT, TWDT to RTC_WDT

do we have 3 Watchdogs ? 

or just 2 and one is a higher Level API of the other.

Dazza0, 2022-07-28:
@franz-ms-muc Agreed. The WDTs is a common area of confusion for users. I'll leave this issue open until the WDT docs are modified. In the mean time, I'll leave some notes here regarding the watchdog timers:

## Watchdog Timer Hardware

It is important to separate out the underlying hardware timers from the SW watchdog features themselves. On the ESP32, there are the three following hardware timers used to implement watchdog related features.

- MWDT0 (part of Timer Group 0)
- MWDT1 (part of Timer Group 1)
- RWDT (part of the RTC module)

All three watchdog timers are essentially the same in design (and almost identical register interface), including features such as different timeout stages, and various timeout actions (such as interrupt, CPU reset, system reset etc). The only difference is that the MWDTs are in the digital clock domain (driven by the APB clock), while the RWDT is in the RTC clock domain (IIRC it's driven by the RTC Fast clock).

Why do we need an RWDT? Because in light/deep sleep, the digital clock domain is gated/powered-down. But there can be scenarios where we still require a watchdog timer while sleeping, thus the RWDT is in the RTC clock domain for this purpose.

See the [Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) for more details about the hardware watchdog timers.

## IDF Software Watchdog Features

A watchdog is essentially just a timer to ensure that a particular "something" (i.e., event/condition/scenario) is met/occurs within a given period, else a timeout error will be triggered. IDF watchdog timers watch two particular "things"

- Ensure that interrupts are never disabled/prevented from running for too long (the Interrupt Watchdog Timer, or IWDT)
- Ensure that tasks don't occupy the CPU for too long (the Task Watchdog Timer, or TWDT)

### IWDT

The IWDT checks if OS tick interrupt is able to run within a given period. If the OS tick interrupt doesn't run for an extended period of time, this is indicative of...

- The interrupts were either disabled for too long thus preventing the OS tick interrupt from running
- There was another ISR that took too long to execute, thus preventing the OS tick interrupt form running

The IWDT uses the MWDT1 as its underlying hardware timer

### TWDT

The TWDT checks that a the FreeRTOS Idle tasks (which have the lowest priority) are able to run within a given period. If the Idle tasks are unable to run for an extended period of time, this is indicative of:

- A higher priority task being stuck in an infinite loop and never blocking
- A higher priority task taking too long to execute

The TWDT uses the MWDT0 as its underlying hardware timer

###  RWDT

As for the RWDT hardware timer, this is just used as a general purpose watchdog timer in times where the IWDT/TWDT are disabled, in order guard against particular sections of code getting stuck. For example,

- The RWDT is used during system startup as the IWDT or TWDT will not have been configured at that stage.
- The RWDT is used to guard against the sleeping process getting stuck (as the IWDT/TWDT will be disabled during sleep).

The RWDT isn't used by our users very often. It's more of an internal API, but there is still an API exposed publicly incase users need an extra watchdog for their own purposes.

brunohpg, 2022-12-05:
Hi,

I have a very strange scenario.  I have IWDT and TWDT enabled by default and my system hangs after a few days. The RWDT is disabled. This started at version > 4.0.

Maybe the system is restarting due to panic. However, this would not go into download mode and would not stop. In fact, I've already validated that it's not in download mode and there was no power-on or brown out.

I already turned off coredump and put the panic handler in IRAM, nothing solved. I really don't know what the problem is, it seems like the CPU just stops.

To maintain operation I want to **start RWDT in user software** since I can't update the bootloader remotely.

Apparently it worked, but I would like to validate if it is correct. I want it to restart the whole system after 60 seconds (to take time for longer operations).

My code is:

```
    rtc_wdt_protect_off();  // Disable RTC WDT write protection
    rtc_wdt_disable();
    rtc_wdt_set_length_of_reset_signal(RTC_WDT_SYS_RESET_SIG, RTC_WDT_LENGTH_3_2us);
    rtc_wdt_set_stage(RTC_WDT_STAGE0, RTC_WDT_STAGE_ACTION_RESET_SYSTEM);
    rtc_wdt_set_time(RTC_WDT_STAGE0, 60000);
    rtc_wdt_enable();      // Start the RTC WDT timer
    rtc_wdt_protect_on();  // Enable RTC WDT write protection
```

And the main task constantly call:

```
rtc_wdt_feed();
```

An interesting thing: in my research I saw that the ESP32 only goes into download mode when it is restarted by power-on, brown out and RTC Watchdog. However, I forced RTC Watchdog to restart the system but it didn't go into download mode, even with GPIO0 low. I'm using ESP32 v3.

brunohpg, 2022-12-14:
@Dazza0 check this issue https://github.com/espressif/esp-idf/issues/10110.

In my tests the CPU does not reset with the RTC WDT in RTC_WDT_STAGE_ACTION_RESET_SYSTEM mode. Also, there is no printout of the stack trace or registers during the crash.

So I changed it to RTC_WDT_STAGE_ACTION_RESET_RTC. Now the WTD is working, however, I believe there is no message when starting. RTC restarts together.

I couldn't find the source. However, I found it very strange not to restart with RTC_WDT_STAGE_ACTION_RESET_SYSTEM.

Dazza0, 2022-12-14:
@brunohpg 

The full stack trace and register printout is done in the panic handler. But on a system reset triggered by a watchdog, the hardware will automatically save the last PC of each CPU, this then gets printed on the next boot up. Look for something like the following in the post reset boot:

```
W (84) boot.esp32: WDT reset info: PRO CPU PC=0x4008440f
W (90) boot.esp32: WDT reset info: APP CPU PC=0x40084407
```

brunohpg, 2022-12-23:
@Dazza0 check here: https://github.com/espressif/esp-idf/issues/10110#issuecomment-1363885862

As I said, I think RTC_WDT_STAGE_ACTION_RESET_RTC do not print PC.
It'll be printed only with RTC_WDT_STAGE_ACTION_RESET_SYSTEM. But with RTC_WDT_STAGE_ACTION_RESET_SYSTEM, the system do not reset!

KonstantinKondrashov, 2024-06-24:
Closes https://github.com/espressif/esp-idf/commit/8d558a0e