espressif/esp-idf #9382: spi_flash_get_chip_size () is gone with v5.0-dev-4037

nopnop2002, 2022-07-19:
## Environment
```
$ idf.py --version
ESP-IDF v5.0-dev-4037-g9b8c558e63
```

## Problem Description
I updated from ESP-IDF v5.0-dev-3202 to ESP-IDF v5.0-dev-4037

spi_flash_get_chip_size () is gone

### Code to reproduce this issue

```cpp
#include "stdio.h"

#include "esp_spi_flash.h"
//#include "spi_flash_mmap.h" // V5

void app_main()
{
    spi_flash_get_chip_size();
}
```

## Debug Logs

```
In file included from ../main/main.c:8:
/home/nop/esp-idf/components/spi_flash/include/esp_spi_flash.h:6:2: warning: #warning esp_spi_flash.h is deprecated, please use spi_flash_mmap.h instead [-Wcpp]
    6 | #warning esp_spi_flash.h is deprecated, please use spi_flash_mmap.h instead
      |  ^~~~~~~
../main/main.c: In function 'app_main':
../main/main.c:13:9: error: implicit declaration of function 'spi_flash_get_chip_size' [-Werror=implicit-function-declaration]
   13 |         spi_flash_get_chip_size();
      |         ^~~~~~~~~~~~~~~~~~~~~~~
../main/main.c:14:9: error: implicit declaration of function 'esp_flash_get_chip_size' [-Werror=implicit-function-declaration]
   14 |         esp_flash_get_chip_size();
      |         ^~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1/1] cd /home/nop/rtos/test/build/bootloader/esp-idf/...00 /home/nop/rtos/test/build/bootloader/bootloader.binBootloader binary size 0x6690 bytes. 0x970 bytes (8%) free.
ninja: build stopped: subcommand failed.
```

AxelLin, 2022-07-19:
It's removed by a690a87829a73019c8e7011ec7b16ff37dbdf02d.
Use esp_flash_get_size instead:
```
-    size_t size_flash_chip = spi_flash_get_chip_size();
+    uint32_t size_flash_chip;
+    esp_flash_get_size(NULL, &size_flash_chip);
```

nopnop2002, 2022-07-19:
@AxelLin 

Thank you.
It work.

```
#include "stdio.h"

//#include "esp_spi_flash.h"
#include "esp_flash.h" // V5

void app_main()
{
    //spi_flash_get_chip_size();
    uint32_t size_flash_chip;
    esp_flash_get_size(NULL, &size_flash_chip);
    printf("size_flash_chip=%d\n", size_flash_chip);
}
```

closed 2022-07-19 by nopnop2002