espressif/esp-idf #9383: EspNOW retry count/limit

mcccol, 2022-07-19:
Problem: EspNOW is useful for fast transmission, but each unicast packet transmission is followed by blocking for an ACK (specifically, a CTS packet.) Timeout waiting for this ACK causes a retransmission. This can occur up to 5 times, which introduces an unpredictable delay on transmission. This can cause a problem in realtime systems, as has been reported and discussed in the forum.

Context: A realtime system which would benefit from EspNOW will have its own methods to deal with lost packets, and a communication protocol used in such systems should have its own methods to detect and react to packet loss.

In such systems, something closer to a datagram would be very useful. EspNOW is very close to being able to do this, but uncontrollable ACK/retry prevents it.

Solution 1: Please add a new function call which has a parameter limiting retransmission attempts, even to 0.

Alternative 2: since EspNOW packets in transmission are placed on a FreeRTOS Queue, provide API access to the Queue

Alternative 3: once un-Queued, EspNOW packets are passed to a lower layer. Can this layer be exposed. 

There are work-arounds for this problem, but they are hacks, and are not widely known or understood.

Thanks.

MaxwellAlan, 2022-07-19:
Hi @mcccol 
Thanks for this feature request. It's not an easy change for espnow retry flow, we will evalute this.

mcccol, 2022-07-19:
Hi @MaxwellAlan.

It is fairly clear that a Queue is created for EspNOW pending transmissions. Assuming that the packet content is enqueued, it would be a matter of adding one more byte, being retry count. Instead of using the literal constant 5, one would pull the retry count max from the queue at the same time as the message, and iterate on retry only that many times.

chenlijun99, 2022-07-25:
Hi @mcccol. I'm also interested in this.
> There are work-arounds for this problem, but they are hacks, and are not widely known or understood.

Could you give some links to such workarounds? Thanks.

mcccol, 2022-07-26:
Hi @chenlijun99,

There are no links - I have experimented and verified the techniques, but not published them - I am hoping that Espressif will make the modifications. One hack is to set the broadcast group of each recipient to its MAC address, then use broadcast instead of unicast, after changing transmitter's broadcast address to recipient's MAC. Another, slightly cleaner, is to use raw transmit to generate the packet. In
 doing so one can avoid the wait-for-CTS code in the espnow transmission. One is then, of course, free to proceed immediately after transmission

chenlijun99, 2022-07-26:
@mcccol Thanks for the tricks. Is it correct that in both cases we can't have encrypted communication?

With the first approach, from https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_now.html#security
> Encrypting multicast vendor-specific action frame is not supported.

With the second approach, you have to build the WiFi frame on your own. Sure, one could build an encrypted WiFi frame, but I don't think that is very straightforward.
Furthermore, bypassing wait-for-CTS seems risky to me. There is more risk of collisions when there are hidden nodes. Though, it may make sense in an application where all the WiFi nodes are nearby and can see each other and low-latency is the paramount property to have.

mcccol, 2022-07-26:
> @mcccol Thanks for the tricks. Is it correct that in both cases we can't have encrypted communication?

I doubt that you could use the Espressif encrypted espnow mechanism, I don't know enough to judge. However, if both tx and rx had a shared secret, you could certainly encrypt a packet's content and decrypt it upon reception

mcccol, 2022-07-26:
> @mcccol Thanks for the tricks. Is it correct that in both cases we can't have encrypted communication?
>
> With the first approach, from [...]

By multicast, do they mean broadcast?
 
> With the second approach, you have to build the WiFi frame on your own. Sure, one could build an encrypted WiFi frame, but I don't think that is very straightforward.

There are plenty of crypto libs, I haven't paid much attention tomwhat esp-idf provides.

 > Furthermore, bypassing wait-for-CTS seems risky to me. There is more risk of collisions when there are hidden nodes.

CTS isn't being used for that, in this case. The CTS is functioning as an ACK. The packet has already been transmitted *and* received before the receiver sends a CTS. CTS is just useful because it's the shortest possible packet which can fuunction as an ACK.

There are a lot of applications which can use datagrams. An unacknowledged espnow-like packet would be a datagram. There are whole classes of applications which will suffer if a transmitter retransmits an old packet, there are datagram applications (and higher level protocols) which require a transmitter discard a packet once sent, whether received or not.

> Though, it may make sense in an application where all the WiFi nodes are nearby and can see each other and low-latency is the paramount property to have.

It makes sense whenever a datagram is required, rather than a send-and-wait protocol. Realtime systems are like that. Streaming audio protocols, telephony protocols, often require datagrams.

chenlijun99, 2022-07-26:
> By multicast, do they mean broadcast?
I have not found anything, but I guess so.

Anyway, thanks for the thorough reply.

mcccol, 2022-07-28:
@chenlijun99 very cool, that looks like the complete implementation, unless I am mistaken. Very useful. Thank you!

Alvin1Zhang, 2022-08-16:
Thanks for reporting, feel free to reopen.

closed 2022-08-16 by Alvin1Zhang

bhuvanchandra, 2023-10-31:
On a related note:

When interacting directly with the [ESP-NOW APIs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/network/esp_now.html#esp-now), rather than using the [ESP-NOW wrapper](https://github.com/espressif/esp-now/tree/master), I am curious about how the unicast acknowledgment (ack) is exposed through the use of the 'send' function and the 'register_send_cb' callback. The '[send_cb](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/network/esp_now.html#_CPPv417esp_now_send_cb_t)' function appears to only provide a status enum that includes error and sent states. Does this mean that the message has been acknowledged when the status is set to 'sent'?

> (BTW, the fact that they both are called ESP-NOW is IMHO rather confusing).

Indeed, it is quite confusing to use an ESP-NOW wrapper that provides no hints or notes about utilizing the ESP-NOW binary libraries in the background.

ivartaim, 2024-09-01:
Hi! has there been any updates on disabling retries in an unicast packet? i have been developing a real- time system where it is crucial that thera are no big slowdowns because of retries. I have been using broadcast mode and custom pairing procedure, but it gets too complicated when the nodes need to start talking back without collisions. it could be solved by simply disabling retries in espnow unicast messages.

zhangyanjiaoesp, 2024-09-02:
@ivartaim we will not disable the retries in the unicast packet.