espressif/esp-idf #9384: scan_evt timeout and system restart

zongshuwei, 2022-07-19:
using ble and wifi together in idf 4.41，it work well. 
but it's about 4 hours .

        [17-45-40.766]scan_evt timeout
        [17-45-40.815]scan_evt timeout
        [17-45-40.865]scan_evt timeout
        [17-45-40.916]scan_evt timeout
        [17-45-40.975]scan_evt timeout
        [17-45-41.025]scan_evt timeout
        [17-45-41.075]scan_evt timeout
        [17-45-41.116]scan_evt timeout
        [17-45-42.315]I (18) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
        [17-45-42.315]I (18) boot: compile time 19:08:40
        [17-45-42.315]I (18) boot: chip revision: 3
        [17-45-42.315]I (20) boot_comm: chip revision: 3, min. bootloader chip revision: 0
        [17-45-42.315]I (27) qio_mode: Enabling default flash chip QIO
        [17-45-42.315]I (32) boot.esp32: SPI Speed      : 80MHz
        [17-45-42.315]I (37) boot.esp32: SPI Mode       : QIO
        [17-45-42.315]I (41) boot.esp32: SPI Flash Size : 4MB
        [17-45-42.315]W (46) boot.esp32: PRO CPU has been reset by WDT.
        [17-45-42.315]W (51) boot.esp32: WDT reset info: PRO CPU PC=0x40081aac
        [17-45-42.315]W (57) boot.esp32: WDT reset info: APP CPU PC=0x40081abc

how i can fix it, thank you.

zongshuwei, 2022-07-19:
there is part of the log.
[
[20220718-154628-1.log](https://github.com/espressif/esp-idf/files/9137157/20220718-154628-1.log)
](url)
   
1. ble disconnect 

        [16-10-53.391] tcp will send len 458 
        [16-10-53.391]I (9676613) wifi_manager: tcp send len 458
        [16-10-53.391]I (9676615) tcp-transfer cache: tcp_write_output  get len 458
        [16-10-53.401]I (9676621) tcp-transfer cache: only_index output_index 214613
        [16-10-53.401]I (9676628) wifi_manager: send_packet_index=26123, send sucess 
        [16-10-53.491]W (9676730) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x13
        **[16-10-53.502]I (9676731) ble host manager: ESP_GATTC_DISCONNECT_EVT, reason = 19**
        [16-10-53.651]I (9676882) wifi_manager: opcode 169 size 2048 len 270  
        [16-10-53.691]I (9676919) tcp-transfer cache: Line=426, out_index=214613,in_index=214891 
        [16-10-53.821]I (9677034) tcp-transfer cache: only_data output_index 214613
  
2.  system restart

      [17-45-40.916]scan_evt timeout
      [17-45-40.975]scan_evt timeout
      [17-45-41.025]scan_evt timeout
      [17-45-41.075]scan_evt timeout
      [17-45-41.116]scan_evt timeout
      [17-45-42.315]I (18) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
      [17-45-42.315]I (18) boot: compile time 19:08:40

3.  sdk config
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9137185/sdkconfig.txt)

thank you very much.

esp-cjh, 2022-07-25:
I found this line in log:
`[17-43-02.666]I (15205926) ble host manager:  scan over time`
Could you please tell me the meaning of this line of print?

zongshuwei, 2022-07-25:
the log is here,

static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param)
{
    uint8_t adv_name_len = 0;
    switch (event) {

		case ESP_GAP_SEARCH_INQ_CMPL_EVT:
		ESP_LOGI(TAG, " scan over time");
                ......

esp-cjh, 2022-07-27:
Hi, shuwei. I will provide a bt lib to debug. Could you please help me reproduce the problem by using the lib and provide log? Unzip the file and replace the libbtdm_app.a in esp-idf/components/bt/controller/lib_esp32/esp32/. Thanks!

[esp32.zip](https://github.com/espressif/esp-idf/files/9196135/esp32.zip)

zongshuwei, 2022-08-03:
Hi cjh，Integrated with this library, I used a device to reproduce the problem, and failed to reproduce it several times; I will continue to reproduce this problem with multiple devices.

esp-cjh, 2022-08-03:
OK, thank you very much!

Ritesh236, 2023-09-07:
> Hi, shuwei. I will provide a bt lib to debug. Could you please help me reproduce the problem by using the lib and provide log? Unzip the file and replace the libbtdm_app.a in [...]

Hello,

We are also facing issues randomly like after 15 to 20 days like evt_scan timeout prints are coming continuously. We are using ESP32 IDF 5.1 Release and Wi-Fi and BLE together.

So, Would you please help me like what can be the cause for the same? 

Let me know if need anything else from my end.

Regards,
Ritesh Prajapati

SB-HiroakiUchiyama, 2023-10-30:
@esp-cjh 
Hello.
I am also encountering the scan_evt timeout.
I have reproduced the issue using esp32.zip shown above against ESP-IDF v4.3.2 using ESP32-WROOM-32 and logged.
* [log.txt](https://github.com/espressif/esp-idf/files/13203343/log.txt)
```log
2023-10-19 02:33:43, diag 1080e300
2023-10-19 02:33:43, diag 1080e300
2023-10-19 02:33:43, ble master soft rst
2023-10-19 02:33:43, scan_evt timeout (16257 16073)
  (repeated...)
2023-10-19 03:01:29, diag 1080e300
2023-10-19 03:01:29, diag 1080e300
2023-10-19 03:01:29, ble master soft rst
2023-10-19 03:01:29, scan_evt timeout (16290 49416)
2023-10-19 03:01:29, diag 1080e300
2023-10-19 03:01:29, scan_evt timeout (16290 49416)
  (Repetition stops here.)
```
My application is running in BLE Central mode and constantly scanning for advertisements. The logs show that the scan is repeatedly stopped and started, but that the problem reproduces even if the scan is changed to just start. The problem also reproduces in ESP-IDF v4.3.6, but I have not been able to log it because applying esp32.zip causes an exception on startup.

I am hoping to solve this problem as it causes the advertisements to stop scanning and cannot be restored. Please let me know if there is any other information you need as I would like to help in any way possible.

Thank you.

Ritesh236, 2023-10-31:
> @esp-cjh Hello. I am also encountering the scan_evt timeout. I have reproduced the issue using esp32.zip shown above against ESP-IDF v4.3.2 using ESP32-WROOM-32 and logged.
> 
> * [log.txt](https://github.com/espressif/esp-idf/files/13203343/log.txt)
> 
> ```
> 2023-10-19 02:33:43, diag 1080e300
> 2023-10-19 02:33:43, diag 1080e300
> 2023-10-19 02:33:43, ble master soft rst
> 2023-10-19 02:33:43, scan_evt timeout (16257 16073)
>   (repeated...)
> 2023-10-19 03:01:29, diag 1080e300
> 2023-10-19 03:01:29, diag 1080e300
> 2023-10-19 03:01:29, ble master soft rst
> 2023-10-19 03:01:29, scan_evt timeout (16290 49416)
> 2023-10-19 03:01:29, diag 1080e300
> 2023-10-19 03:01:29, scan_evt timeout (16290 49416)
>   (Repetition stops here.)
> ```
> 
> My application is running in BLE Central mode and constantly scanning for advertisements. The logs show that the scan is repeatedly stopped and started, but that the problem reproduces even if the scan is changed to just start. The problem also reproduces in ESP-IDF v4.3.6, but I have not been able to log it because applying esp32.zip causes an exception on startup.
> 
> I am hoping to solve this problem as it causes the advertisements to stop scanning and cannot be restored. Please let me know if there is any other information you need as I would like to help in any way possible.
> 
> Thank you.

Hello,

Would you please raise issues to Espressif Team directly as they will share few debug libraries with some debug statement using which you can get informative logs if issue is reproduced easily into your case because into my case issue is not reproduced easily?

Hope that would be helpful to address out issue.

Let me know if need any help from my end.

Regards,
Ritesh Prajapati

esp-cjh, 2023-11-07:
@Ritesh236 @SB-HiroakiUchiyama 
Please use this controller lib and check whether the problem has been fixed.
lib version [48f8803]
[231107.zip](https://github.com/espressif/esp-idf/files/13277986/231107.zip)

SB-HiroakiUchiyama, 2023-11-10:
@esp-cjh 
Thanks for working on the problem.
So far there seems to be no problem with this lib.

I would like to spend more time to check the problem, as the problem occurs infrequently, sometimes only once a week.
so I will report back on the results.

thank you.

Ritesh236, 2023-11-10:
> @Ritesh236 @SB-HiroakiUchiyama Please use this controller lib and check whether the problem has been fixed. lib version [48f8803] [231107.zip](https://github.com/espressif/esp-idf/files/13277986/231107.zip)

Thanks @esp-cjh for sharing library. 

Actually, we are also tried to check few debug link which has been shared by Espressif Team earlier but still we are not able to reproduce issue at our end. We sometimes faced issue after 15 or 20 days one time when we were working earlier.

But, We will continue to monitor it at our end and will let you know if we will be able to reproduce issue at our end.

Regards,
Ritesh Prajapati

SB-HiroakiUchiyama, 2023-11-21:
@esp-cjh 
Tests have been performed and appear to be fine.
Thank you for fixing the problem.

When will the ESP-IDF be released incorporating this fix?
It would be helpful to know the expected release date and the version that will be released.

thank you.

Ritesh236, 2023-11-21:
> @esp-cjh Tests have been performed and appear to be fine. Thank you for fixing the problem.
> 
> When will the ESP-IDF be released incorporating this fix? It would be helpful to know the expected release date and the version that will be released.
> 
> thank you.

Great. Thanks for providing updates. 

I have same suggestions like when it will be released into ESP-IDF so that it would be helpful for people to get fixed solution with that. 

Regards, 
Ritesh Prajapati

esp-cjh, 2023-11-21:
OK, I will submit a merge request ASAP and let you know once it's merged.

Ritesh236, 2023-11-21:
> OK, I will submit a merge request ASAP and let you know once it's merged.

Thanks for providing updates. 

Do we have any identification for the patch once it will be fixed? Like can it be possible to identify patch with this bug id? Is there any mapping done for patch with this bug is? 

Just asking so that people will get to know that this bug has been fixed into this patch with some details. 

Regards, 
Ritesh Prajapati

1technophile, 2023-11-27:
@esp-cjh 
Thanks for the lib.
I confirm that it gets rid of the scan evt timeout issue on my tests.
I'm planning to integrate your lib into OpenMQTTGateway as a patch, what is the board scope?
ESP32 only, or can it be used for ESP32 S3/C3 ?

esp-cjh, 2023-11-28:
@1technophile This lib is for EPS32 only.  This issue does not currently occur on the ESP32C3.

Ritesh236, 2023-11-28:
> @1technophile This lib is for EPS32 only. This issue does not currently occur on the ESP32C3.

Hello @esp-cjh ,

I believe that there is no any dependencies about hardware for this issue as it is BLE Stack level issues. 

So, It will be reproduced into any hardware of ESP32 while using this kind of library and stack.

Let me correct if i am wrong at any place.

Regards,
Ritesh Prajapati

tomasznowik, 2024-01-16:
@esp-cjh is merge request with the fix accepted & merged?

Ritesh236, 2024-01-19:
> @esp-cjh is merge request with the fix accepted & merged?

I think they have provided workaround libraries with fix but not sure that they have included that change into main release or not. So, you can directly check with Espressif Team to get details for the same.

Regards,
Ritesh Prajapati

esp-cjh, 2024-02-23:
@tomasznowik The merge request was merged. The commit is 74f99d267939674b7e67813aa12199d14ed1aa28.
Please check if this issue can be closed.

Ritesh236, 2024-02-23:
Thanks a lot @esp-cjh for sharing commit ID with details which has been fixed before two months. 

We will also check same issue into latest commit and will let you know if any further query for the same.

Regards,
Ritesh Prajapati

ma261065, 2024-03-08:
So this fix is not in the latest v5.2.1 release? I don't see it in the list of fixes...

Ritesh236, 2024-03-10:
> So this fix is not in the latest v5.2.1 release? I don't see it in the list of fixes...

@ma261065 , Are you facing similar kind of issues at your end? Would you please check commit ID which has been mentioned into earlier comments from which you can get idea like in which release it has been fixed?

_The merge request was merged. The commit is https://github.com/espressif/esp-idf/commit/74f99d267939674b7e67813aa12199d14ed1aa28.
Please check if this issue can be closed._

Regards,
Ritesh Prajapati

ma261065, 2024-03-10:
Yes I have the same issue. 

I was hoping to see it fixed in an official IDF release - but checking the page here https://github.com/espressif/esp-idf/releases doesn't show this commit in any release, even the latest one (v5.2.1) which was only released a few days ago.

Ritesh236, 2024-03-11:
> Yes I have the same issue.
> 
> I was hoping to see it fixed in an official IDF release - but checking the page here https://github.com/espressif/esp-idf/releases doesn't show this commit in any release, even the latest one (v5.2.1) which was only released a few days ago.

Hello,

Have you checked this below image regarding this issue fix? And also can you please check by updating library which is provided into fix link?

Hello @esp-cjh,

Would you please check from your end as per queries raised by end developers as they are not seeing commit into ESP IDF?

esp-cjh, 2024-04-23:
Hi, @Ritesh236 
The fix commit on release/v5.2 branch is 56563f709229cbef07b79e1b5db48146f3b17e42.

closed 2024-04-23 by Indastri

Ritesh236, 2024-04-24:
> Hi, @Ritesh236 The fix commit on release/v5.2 branch is [56563f7](https://github.com/espressif/esp-idf/commit/56563f709229cbef07b79e1b5db48146f3b17e42).

Thanks a lot @esp-cjh for your prompt response.

Regards,
Ritesh Prajapati

thorrak, 2024-05-13:
The problem is that Arduino 2.x is based on ESP-IDF v4.4.7 which I doubt this fix was backported to. My question is if this fix was backported to v5.1, as that is what the upcoming Arduino 3.x release is based on.

esp-cjh, 2024-05-13:
Hi, @jensolsson @thorrak 
Please find this line `BTDM_INIT: BT controller compile version` in your log file. And I'll check whether the fix is included according to the lib version.

jensolsson, 2024-05-13:
@thorrak For me it will not just reboot the device but hang it so it will not do anything other than putputing this error over and over. I think it rendera the esp32 very ubreliable when used with arduino and scanning. I believe this is a fix that should be considered for backporting. After all Arduino 3 is not released and there is a lot of both companies and individuals who use the esp32 with arduino.

esp-cjh, 2024-05-14:
@thorrak The fix was backported to v5.1 on tag v5.1.3. The fixed commit on v5.1.3 is 2ec5d6b0f2ed20fb179ccd854faac8fcf6e9a441.

AstroTaka, 2024-11-11:
@esp-cjh Is the backported to 5.0?