espressif/esp-idf #9387: "ESP_DHCP_DEBUG" is undeclared", problem building with docker container esp-idf 4.1.3

madeinaus, 2022-07-19:
## Environment
docker image: "docker pull espressif/idf:v4.1.3"

- Development Kit:      [none]

- Module or chip used:  [none]
- IDF version
    "docker pull espressif/idf:v4.1.3
    docker digest: 632e629efdb
    hub.docker.com/layers/espressif/idf/v4.1.3/images/sha256-632e629edfdb0c8302dcdc266c566169557fc513979268f8742f9d586e106e95
 
- Build System:         [idf.py]
  exp-idf v4.1.3-dirty

- Compiler version: 8.4.0

- Operating System:     [Linux]

- Using an IDE?: [No]
- Power Supply:         [none]

## Problem Description

"ESP_DHCP_DEBUG" is undeclared

### Expected Behavior
build completed
### Actual Behavior
/opt/esp/idf/components/lwip/lwip/src/core/ipv4/dhcp.c:292:15: error: 'ESP_DHCP_DEBUG' undeclared (first use in this function); did you mean 'DHCP_DEBUG'?

### Steps to reproduce

1.  docker pull espressif/idf:v4.1.3
2.  docker run --rm  -it espressif/idf:v4.1.3
3. (change to docker terminal session)
4. mkdir project
5. cp -r opt/esp/idf/examples/ethernet/basic project
6. cd project
7. idf.py build

example project builds correctly in idf:v4.1.2. 
Problem is in lwip core: not limited to 'example' projects.

MaxwellAlan, 2022-07-26:
Hi @madeinaus
Test in my local environment, it works well. Can you try below command in IDF root path 
git submodule update --init --recursive

david-cermak, 2024-10-01:
closing as "not reproducible". please reopen if needed.