espressif/esp-idf #9388: Max BLE characteristics with table declaration, and crash on connection

mngr0, 2022-07-19:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32E
- IDF version : v5.0-dev-3654-gc2ccc383da
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:  Linux
- Using an IDE?: No (vscodium to edit files)
- Power Supply:   LDO regulated 3.3V

## Problem Description

Hello,

I designed a Bluetooth profile. I declared a few hundred characteristics in a service.
According to [314](https://github.com/espressif/esp-idf/issues/314#issuecomment-332445178) it should be possible to add any number of characteristics in a service. Yet, using the table declaration:
```
esp_err_t esp_ble_gatts_create_attr_tab(const esp_gatts_attr_db_t *gatts_attr_db, esp_gatt_if_t gatts_if, uint8_t max_nb_attr, uint8_t srvc_inst_id);
```
up to 255 attributes (uint8_t) are accepted. How should I proceed?

Moreover, if I declare more than 16 characteristics(yes, I mean characteristics, not attributes ) in a service the server crashes when a client connects.
I have 3 services, and only one is managing the advertisement (i.e. calling ```esp_ble_gap_config_adv_data``` ) as shown in [this example](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_server/main/gatts_demo.c#L499), in which are declared two services (A and B) and only A is calling gap functions.
The server fails on connection if I use more than 16 characteristics in any of the 3 services.

### Expected Behavior

I expect to connect flawlessly with up to 255 attributes per service

### Actual Behavior

I am not able to declare more than 255 attributes using the table declaration.
I am not able to connect to a service with more than 16 characteristics.

### Steps to reproduce

todo: making repo to clone

### Code to reproduce this issue

I am opening the repository, soon you will be able to see all the code
(I am not using gist because the code is divided in several files)

## Debug Logs

```
todo: I will soon edit and add the log
```

## Other items if possible

todo: I will attach the sdkconfig
- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

esp-cjh, 2022-07-25:
I have reproduced the problem. The reason of crash is that the index exceeds the range of the handles array (line 584 in btc_gatts.c). There are some ways to avoid this problem.
1. Increase the handles array size.
components/bt/host/bluedroid/btc/profile/std/include/btc_gatts.h
`#define ESP_GATT_ATTR_HANDLE_MAX 100`
2. Add characteristics and characteristics descriptor instend of creating attr table.
There will be no quantitative limit by using `esp_ble_gatts_add_char()` and `esp_ble_gatts_add_char_descr()`.
You can refer to the examples/bluetooth/ble/gatt_server.

Additionally, it's not recommended to update conn params in all gatts profile cb. I noticed that you update conn params in `gatts_profile_conf_event_handler()`, `gatts_profile_status_event_handler()` and `gatts_profile_main_event_handler()`. You can also refer to the gatt_server example.

mngr0, 2022-08-02:
Hello esp-cjh, thank you for helping me.
I hope that this issue can help other users in the future

Alvin1Zhang, 2022-11-21:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/e2d488cf4b7721f92d5c214bbcadd724bcf1921c, and also back ported to other release branches, feel free to reopen. Thanks.

closed 2022-11-21 by Alvin1Zhang