espressif/esp-idf #9389: i2s_ad_dac example has gaps in playback

PilnyTomas, 2022-07-19:
## Environment

- Development Kit: [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used: [ESP32-WROOM-32D]
- IDF version: Tried on `v4.4.1-581-gee2029c38a` and `v5.0-dev-4257-g508661b03e`
- Build System: [idf.py]
- Compiler version: 8.4.0
- Operating System: [Linux]
- Using an IDE?: [No]
- Power Supply: [USB, directly from computer - no hubs inbetween]

## Problem Description

Audio playback is laggy.
There seems to be approx 64.4ms period of playback followed by approx the same length (64.4ms) period of silence - this pattern repeats and the resulting audio is horribly unusable, laggy / chopped.

### Expected Behavior
Clear audio without pauses, lags, gaps,...

### Actual Behavior
Gaps in playback

### Steps to reproduce

1. Setup example https://github.com/espressif/esp-idf/tree/release/v4.4/examples/peripherals/i2s/i2s_adc_dac
2. Listen
3. Record on an oscilloscope

In the pictures, channel 2 is ADC output, and channel 3 is amplifier output

[image]
[image]

PilnyTomas, 2022-08-01:
Uploading some more pictures - audio file playback - both DAC channels

Overview
[image]

Closeup - gap period
[image]

Closeup - playback period
[image]

L-KAYA, 2022-09-23:
The issue can be fixed by changing the I2S configuration from
```c
        .dma_buf_count = 2,
        .dma_buf_len = 1024,

```
to
```
        .dma_buf_count = 6,
        .dma_buf_len = 256,
```

The fix is on the way.

PilnyTomas, 2022-09-26:
Hi @L-KAYA, I just tested the example with the new settings and can confirm that the resulting audio is clear and without gaps.

closed 2022-09-26 by PilnyTomas