espressif/esp-idf #9391: Confserver doesn't return update message when options are changed

FanhuaCloud, 2022-07-19:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S3-WROOM-1-N8R8
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-4018-g12e8f0ba8c
- Build System:         idf.py
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes VSCode
- Power Supply:         USB

## Problem Description
It seems that this update broke the feature: 
https://github.com/espressif/esp-idf/commit/12e8f0ba8c4558467549599f38a0914b789d1423

[image]
As shown in the picture, when I check the option, the option below should be changed, but it doesn't
### Steps to reproduce
1. use VSCode esp-idf plugin.
2. open menuconfig.
3. change the option.

## Debug Logs
Error case, no change notification sent
```
[SDK Configuration Editor]
{"version": 2, "set": { "LV_MEM_CUSTOM": true }}
```
Correct case, when option is set, can receive change message
```
[SDK Configuration Editor]
{"version": 2, "set": { "LV_MEM_CUSTOM": true }}

[SDK Configuration Editor]
Set LV_MEM_CUSTOM
{"version": 2, "values": {"LV_MEM_CUSTOM": true, "LV_MEM_CUSTOM_INCLUDE": "stdlib.h"}, "ranges": {}, "visible": {"LV_MEM_SIZE_KILOBYTES": false, "LV_MEM_ADDR": false, "LV_MEM_CUSTOM_INCLUDE": true, "LV_USE_MEM_MONITOR": false}}
```

semchupin, 2022-08-03:
@FanhuaCloud can you sand the project files? Because I can't reproduce this error on basic examples, `confserver` returns modify elements without problems.

FanhuaCloud, 2022-08-03:
> @FanhuaCloud can you sand the project files? Because I can't reproduce this error on basic examples, `confserver` returns modify elements without problems.

Thanks, I found that the latest commit has been fixed

closed 2022-08-03 by FanhuaCloud

dobairoland, 2022-08-03:
Thanks for the feedback @FanhuaCloud! We are happy that it is solved.