espressif/esp-idf #9392: Support SNI for http_client

WolfspiritM, 2022-07-19:
**Is your feature request related to a problem? Please describe.**

I'm trying to fetch a https website. The website hosts multiple domains and uses different SSL Certs for each domains created by LetsEncrypt. The switch is done via SNI. The certificate for the default page is self signed and therefore "invalid". When using esp_http_client I receive the following error.
```
 (4598) wifi station: Waiting for system time to be set... (2/60)
E (11198) esp-x509-crt-bundle: Failed to verify certificate
E (11198) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x3000
E (11198) esp-tls: Failed to open new connection
E (11208) TRANSPORT_BASE: Failed to open a new connection
E (11208) HTTP_CLIENT: Connection failed, sock < 0
E (11218) wifi station: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
```

I changed the verify callback and checked which certificate it actually gets and it's the root certificate instead of the domain certificate.
It seems like it's not using SNI to request a specific certificate from the server correctly.

**Describe the solution you'd like**
Successfully transmit the host name as SNI to get the correct cert back from the server

mahavirj, 2022-07-20:
@WolfspiritM 

SNI is already supported, please see relevant config at:

https://github.com/espressif/esp-idf/blob/508661b03eec6e15af66f77050e699a7658b9793/components/mbedtls/port/include/mbedtls/esp_config.h#L1303

Can you please check hostname at following location, this must match with the endpoint that you are trying to connect:

https://github.com/espressif/esp-idf/blob/508661b03eec6e15af66f77050e699a7658b9793/components/esp-tls/esp_tls_mbedtls.c#L598-L608

WolfspiritM, 2022-07-20:
> Can you please check hostname at following location, this must match with the endpoint that you are trying to connect:

@mahavirj 
Thanks! That actually pointed me in the right direction!
I think however that there is a bug with that implementation.
Due to some issues with the certificate I had this set in my code beforehand
```
        .skip_cert_common_name_check = true,
```

I did not expect this to affect SNI, however around the code you mentioned there is actually a check for that flag.
Removing that flag fixed the problem for me!

I think SNI should still work even if the common name is not checked cause it theoretically is possible to have a valid certificate for a vhost that does just not have a valid common name while the default SSL Certificate that is send without SNI is invalid (due to time or just self signed) completly. Or am I wrong?

I agree that skip_cert_common_name_check is insecure and should not be used anyway so I count this as a minor "feature request" to support SNI without common name check

WolfspiritM, 2022-07-21:
Thanks I guess it's a feature request on mbedtls side then

closed 2022-07-21 by WolfspiritM

masterxq, 2025-04-23:
This took me so much time to finger out :/ This is not the expected behavior.

If some trying to setup some new certs and there are some problems, he/she will enable the skip for avoid having some problems with the name in first iteration.
This results in that all openssl checks and everything that can be done will working but the espressif http client will reject the cert.
In my case the webserver (apache2) just delivered the default cert as the name was unknown and *not* the cert for the correct virtualhost... What else should it do...

There are two possiblities to solve this:
1. Give a hint in the logs where the user should search for solving the issue. There are many ways like printing some information about the server cert. Or directly explain something about this trap. It's fine if hidden when log level is low and shown if somebody trying to find the issue (high loglevel)
2. Change the behavior

This took me so many hours of hacking libraries and force some logs from mbedtls and I only found the issue by luck at the end. Please do something to protect other users from this. It is just super super hard to find even with acceptable knowledge about ssl...

@WolfspiritM please give this a second look with special attention on the user perspective! :)

bryghtlabs-richard, 2025-07-03:
@mahavirj , FYI, still an issue on master branch. Perhaps updating the documentation to reflect `skip_common_name` is incompatible with SNI would help?

mahavirj, 2025-07-04:
> FYI, still an issue on master branch. Perhaps updating the documentation to reflect skip_common_name is incompatible with SNI would help?

Yes, we will consider adding a note for this. CC @hrushikesh430 

> Rather than dropping the server-name, would it make sense to instead degrade the auth-mode?

We have a config for this mode:

https://github.com/espressif/esp-idf/blob/76cec451cee2f253ab2e81460dd169dd3074d4cc/components/esp-tls/Kconfig#L90