espressif/esp-idf #9394: please keep architecture specific dependency information in architecture specific locations

davidzuhn, 2022-07-20:
I have a project that I build for multiple devices -- one is ESP32 and the other is ESP32-S3.   I've got things nicely separated based on the target architecture, and I have a wrapper script that builds one and then the other as needed, saving the files from build.

But the new dependencies.lock file throws a monkey wrench into the mix.   I've got this saved in version control, and it's wrong for one or the other because the architecture is saved in the file.   Thus, my version (based on git describe --tags) will show one as 'dirty' because the build process changes this file from what I had in version control.

Does the architecture need to be in the dependencies.lock file?   If so, can the format be changed to include an architecture tag for each piece of information and then don't change it if it's not the architecture for this build.  Or something so that running a build twice, once for each architecture, does not have to needlessly change the file?

0xjakob, 2022-07-21:
@davidzuhn Please don't maintain the `dependencies.lock` file. It is generated and maintained by the component manager.

> The lock-file `dependencies.lock` and content of `managed_components` directory is not supposed to be modified by a user. When the component manager runs it always make sure they are up to date. 

From the [component manager documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html).

It's basically a build artifact. Whenever you run e.g. `idf.py build`, it should be changed or re-generated if necessary.

davidzuhn, 2022-07-21:
Wouldn't it be better then for build time artifacts to exist in the build/ directory instead of in the top level source?

igrr, 2022-07-21:
Similar to other package ecosystems, .lock file can optionally be added to version control to get exact same versions of packages (including their transitive dependencies) on every build. It is similar to Pipfile.lock, package-lock.json, etc.

I think the fact that we don't support keeping this file stable when switching between targets is a bug (or a missed design requirement), we will try to address it. Apologies if the resolution takes some time, as part of the team is on leave now.

igrr, 2022-07-26:
We have discussed the solution for this issue, it will probably look as follows:

1. In https://github.com/espressif/idf-component-manager (python package where the component manager client is implemented) we will support overriding the name of the `dependencies.lock` file.
2. In IDF, we will introduce new build system variable, `DEPENDENCIES_LOCK`. Setting it in the project CMakeLists.txt file will allow overriding the dependencies.lock file name. For example, if the build matrix includes different targets, one will be able to set it to `depenencies.lock.${IDF_TARGET}`. If another dimension (such as IDF version or product type) is required, one can add extra variables to the name, e.g. `dependencies.lock.${IDF_TARGET}.${IDF_VERSION}.${PRODUCT_TYPE)`. One will also be able to move the file to the build directory by setting the name to `build/dependencies.lock`, in case version locking is not necessary.

@davidzuhn You are developing with the `master` branch, right? (Asking to see if we need to backport this feature to release/v4.4 as well)

davidzuhn, 2022-07-30:
@igrr -- yes, I'm tracking the master branch, and no longer work with 4.4, so no back port is needed on my behalf.   

The proposed solutions looks like it would suit my needs nicely, thanks.

closed 2022-12-22 by espressif-bot