espressif/esp-idf #9395: [Bug][v4.4.1][Websockets] "httpd_accept_conn: error in accept (23)" despite lru_purge_enable = true

chipweinberger, 2022-07-20:
**ESP32-S3 w/ 2MB PSRAM**
**ESP-IDF v4.4.1**

**Problem:**
I have a single websocket endpoint "/logs"
After a few (roughly ~3) client connections using websockets, eventually I hit "httpd_accept_conn: error in accept (23)" and wifi refuses to make further connections.

I am already setting lru_purge_enable, 

<details>
    <summary>Logs:</summary>

```
I (288009) pd http api: GET /api/logs/websocket called
I (288009) pd http api: GET /api/logs/websocket called
I (288019) pd http api: new client fd 62 Websocket
I (288029) keep alive: Added client: 62
lwip_send(62, data=0x3fce9442, size=2, flags=0x0)
lwip_send(62) err=0 written=2
lwip_send(62, data=0x3fca8230, size=51, flags=0x0)
lwip_send(62) err=0 written=51
lwip_select(63, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=62 ready for reading
lwip_select: nready=1
lwip_recvfrom(62, 0x3fce96eb, 1, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=0, pbuf=0x3fcc39a0
lwip_recv_tcp: buflen=23 recv_left=1 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fcc39a0
lwip_recvfrom(62):  addr=192.168.0.2 port=56722 len=1

// removed some websocket revc packet for brevity 

I (288079) pd http api: ws: /api/logs/websocket recv pkt
lwip_send(62, data=0x3fce95a2, size=2, flags=0x0)
lwip_send(62) err=0 written=2
lwip_send(62, data=0x3fca8230, size=50, flags=0x0)
lwip_send(62) err=0 written=50
lwip_recvfrom(62, 0x3fce9703, 1, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fcc39a0
lwip_recv_tcp: buflen=22 recv_left=1 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fcc39a0
lwip_recvfrom(62):  addr=192.168.0.2 port=56722 len=1
lwip_recvfrom(62, 0x3de5529d, 4, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fcc39a0
lwip_recv_tcp: buflen=21 recv_left=4 off=0
lwip_recv_tcp: lastdata now pbuf=0x3fcc39a0
lwip_recvfrom(62):  addr=192.168.0.2 port=56722 len=4
lwip_recvfrom(62, 0x3fcc3710, 17, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x3fcc39a0
lwip_recv_tcp: buflen=17 recv_left=17 off=0
lwip_recv_tcp: deleting pbuf=0x3fcc39a0
lwip_recvfrom(62):  addr=192.168.0.2 port=56722 len=17
lwip_select(63, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_sendto(56, data=0x3fcf2d30, short_size=12, flags=0x0 to=127.0.0.1 port=32768
lwip_selscan: fd=55 ready for reading
lwip_select: nready=1
lwip_recvfrom(55, 0x3fce9830, 12, 0x0, ..)
lwip_recvfrom_udp_raw[UDP/RAW]: top sock->lastdata=0x0
lwip_recvfrom_udp_raw[UDP/RAW]: netconn_recv err=0, netbuf=0x3fcc39a0
lwip_recvfrom_udp_raw: buflen=12
lwip_recvfrom_udp_raw(55):  addr=127.0.0.1 port=55494 len=12
lwip_send(62, data=0x3fce9792, size=2, flags=0x0)
lwip_send(62) err=0 written=2
lwip_send(62, data=0x3c10c294, size=5, flags=0x0)
lwip_send(62) err=0 written=5
lwip_select(63, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=62 ready for reading
lwip_select: nready=1
lwip_recvfrom(62, 0x3fce96eb, 1, 0x0, ..)
[... 371 lines cut ...]

I (354209) pd http api: Client disconnected 63
lwip_select(56, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=54 ready for reading
lwip_select: nready=1
lwip_accept(54)...
W (354299) httpd: httpd_accept_conn: error in accept (23)
W (354299) httpd: httpd_server: error accepting new connection
lwip_select(56, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=54 ready for reading
lwip_select: nready=1
lwip_accept(54)...
W (354399) httpd: httpd_accept_conn: error in accept (23)
W (354399) httpd: httpd_server: error accepting new connection
lwip_select(56, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
lwip_selscan: fd=54 ready for reading
lwip_select: nready=1
lwip_accept(54)...
W (354499) httpd: httpd_accept_conn: error in accept (23)
W (354499) httpd: httpd_server: error accepting new connection
lwip_select(56, 0x3fce9884, 0x0, 0x0, tvsec=-1 tvusec=-1)
I (475719) pd wifi: Wifi Event: AP Mode - Client Disconnected
I (475729) pd wifi: Client disconnected: f8:4d:89:7c:19:f0 leave, AID=1
```
</details>

chipweinberger, 2022-07-20:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9147097/sdkconfig.txt)

chipweinberger, 2022-07-20:
Logs from another run:
[logs.txt](https://github.com/espressif/esp-idf/files/9147131/logs.txt)

chipweinberger, 2022-07-23:
@david-cermak  any thoughts? 

Been hitting this issue for ~6 months, and just collected more detailed logs.

ESP-YJM, 2022-07-28:
@chipweinberger 23 means allocate socket number failed.  Maybe the closed sockets are in TIME-WAIT state and not free the socket number. I think you can use `SO_LINGER` socket option in `httpd.close_fn` callback function.
Like this :
```
void close_fd_cb(httpd_handle_t hd, int sockfd)
{
   struct linger so_linger;
   so_linger.l_onoff = true;
   so_linger.l_linger = 0;
   setsocketopt(sockfd, SOL_SOCKET, SO_LINGER, &so_linger, sizeof(so_linger))
   close(sockfd);
}
```

chipweinberger, 2022-07-28:
**That fixes it !!!**🥳 🥳 🥳 

2 Questions:
1. Would it make more sense to call this code in open_fd_cb? That way, it is set as soon as the socket is established?
2. Why is SO_LINGER = 0 not default when setting lru_page_enable = true?

Small note, I called `lwip_setsockopt` instead of `setsocketopt` since `#include <sys/socket.h>` does not seem to include that one.

ESP-YJM, 2022-07-28:
@chipweinberger Glad to hear that. For question one, i think that set the option before close also make sense. This option is for socket close. No need set this option when socket created. For question two, Setting `lru_page_enable = true` is different from this `SO_LINGER` option.  Setting `lru_page_enable = true` is to close the less use session, and this option is to make socket number avalibale when call close socket.

chipweinberger, 2022-07-28:
Thanks for the response. 

I ask question 2 because `error in accept (23)` is difficult to fix without deep knowledge. 

**An idea:** Perhaps we can add `so_linger` to `httpd_config_t`? This way users will see documentation to fix `error in accept (23)` in the `esp_http_server.h`

```
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();
    config.so_linger = false;
```

Or perhaps we can add an FAQ to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/protocols/esp_http_server.html

Regardless, thank you for your assistance.

ESP-YJM, 2022-07-28:
@chipweinberger Good idea and we have already add the `errno 23`. You can refer to https://docs.espressif.com/projects/espressif-esp-faq/en/latest/software-framework/protocols/lwip.html#after-creating-and-closing-tcp-socket-several-times-an-error-is-reported-as-unable-to-create-tcp-socket-errno-23-how-to-resolve-such-issue

chipweinberger, 2022-07-28:
Ah! =)

Google search for the exact error string does not find that document:
https://www.google.com/search?q=httpd_accept_conn

But at least Google now finds this github thread =)

Perhaps the FAQ can include the string "httpd_accept_conn: error in accept (23)".

I will close this issue now. 

Thank you for your help.

closed 2022-07-28 by chipweinberger

chipweinberger, 2022-09-13:
@ESP-YJM, I often see lwip_setsockopt fail. Is this expected?

`E (103797) : e1 : [pd http api] lwip_setsockopt() failed. fd 52 rv -1 errno 109`

```
// This code is needed to solve "httpd_accept_conn: error in accept (23)"
// See: https://github.com/espressif/esp-idf/issues/9395
struct linger so_linger;
so_linger.l_onoff = true;
so_linger.l_linger = 0;
int rv = lwip_setsockopt(fd, SOL_SOCKET, SO_LINGER, &so_linger, sizeof(so_linger));
if (rv < 0){
    ESP_LOGE(TAG, "lwip_setsockopt() failed. fd %d rv %d errno %d", fd, rv, errno);
}

int rv = close(fd);
if (rv != 0){
    ESP_LOGE(TAG, "close() fd (%d) failed. rv %d errno %d", fd, rv, errno);
}
```

chipweinberger, 2022-09-27:
Bump @ESP-YJM

ESP-YJM, 2022-09-28:
@chipweinberger Sorry for late reply. 109 means ENOPROTOOPT, protocol not available. You need check whether you enable `CONFIG_LWIP_SO_LINGER` in menuconfig.

chipweinberger, 2022-09-28:
Sorry if this is a dumb question, but why does 109 = ENOPROTOOPT?

**I see this code:**
`#define  ETOOMANYREFS   109  /* Too many references: cannot splice */`

ESP-YJM, 2022-09-28:
@chipweinberger We not use `errno.h` in lwip. The actual `errno.h` we used is in toolchain(/root/.espressif/tools/xtensa-esp32-elf/esp-2021r2- 8.4.0/xtensa-esp32-elf/xtensa-esp32- elf/include/sys/errno.h).