espressif/esp-idf #9397: Uart at low baud rate doesn't read correctly during short period after initializing it

floitsch, 2022-07-20:
- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it): v5.0-dev-4257-g508661b03e
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
When setting the UART to low baud rates (below 9600) the reading doesn't always work unless I put in a delay.
That is, I need to wait after an `uart_driver_install`, `uart_flush_input`, and the first `uart_read_bytes`.

This behavior is on master, but is even worse on 4.4.

Small observations:
- it's important to disable the UART pins before running the test. If the pins stay initialized the read (usually?) works.
- increasing the baud rate fixes the issue.
- on 4.4 I frequently got garbled reads. That is, characters were not correctly read. This excludes timing issues with the read-buffers. It basically means that it's not because I'm not reading fast enough that some data is lost.

### Expected Behavior
As soon as `uart_driver_install` and `uart_flush_input` return, the UART starts reading and can receive data.

### Actual Behavior
I need to add a sleep before the UART can receive data.

### Steps to reproduce

1. Connect pin 16 to pin 26.
2. Connect pin 17 to pin 27.
3. Initialize 16, 17 as UART_NUM_2 with baud_rate 400 (or 1200, ...). Install the driver and flush the input.
4. Initialize 26, 27 as UART_NUM_1 with the same baud rate. Install the driver and flush the input
5. Send data on UART_2
6. Read data on UART_1
7. Verify that the data is correct.
8. If it is, disable all pins, and restart.

### Code to reproduce this issue

https://gist.github.com/floitsch/99340a2e7cdbd302931fb4cbbfe915e5

Relevant sections; a condensed version:
``` c
static void init_port(uart_port_t port, int tx, int rx) {
  uart_config_t uart_config = {
    .baud_rate = 400,
    .data_bits = 8 - 5,
    .parity = UART_PARITY_DISABLE,
    .stop_bits = UART_STOP_BITS_1,
    .flow_ctrl = 0,
    .rx_flow_ctrl_thresh = 122,
  };
  ESP_ERROR_CHECK(uart_param_config(port, &uart_config));
  ESP_ERROR_CHECK(uart_set_pin(port, tx, rx, -1, -1));
  int buffer_size = 2 * 1024;
  QueueHandle_t uart_queue;
  ESP_ERROR_CHECK(uart_driver_install(port, buffer_size, buffer_size, 32, &uart_queue, 0));
  ESP_ERROR_CHECK(uart_flush_input(port));
}

void app_main(void)
{
  uart_port_t port1 = UART_NUM_2;
  init_port(port1, 17, 16);
  uart_port_t port2 = UART_NUM_1;
  init_port(port2, 26, 27);

  // vTaskDelay(2000 / portTICK_PERIOD_MS);  // <=== with this delay things work.

  const char* str = "good day you all";
  uart_write_bytes(port1, str, 16);
  ESP_ERROR_CHECK(uart_wait_tx_done(port1, portMAX_DELAY));
  char buffer[20];
  int read = uart_read_bytes(port2, buffer, 20, 0);
...
```

### Output of monitor

Condensed output of the monitor.
This is the second iteration.
Note that the read bytes should be "good day you all", or a prefix thereof.

```
Restarting now
[...]
I (0) cpu_start: Starting scheduler on APP CPU.
I (329) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (339) uart: queue free spaces: 32
I (339) uart: queue free spaces: 32
Read 10 bytes: tday y
I (749) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (749) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (759) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (769) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Restarting in 1s
```

### Screenshot

Shows that the RX of the reading UART has a stable input. The blue line is from pin18 (see code in gist).

[image]

moefear85, 2022-07-23:
some small notes:
1- uart_flush_input() is unnecessary. uart_driver_install() does it automatically
2- if not using flow control, "rx_flow_ctrl_thresh" can be set to 0 (or not set at all), since it is not used.
3- ".data_bits = 8 - 5", I'm pretty sure this evaluates to ".data_bits=3", which in turn means 8 bits, so you can use "UART_DATA_8_BITS". Just want to be sure you are aware of this (see uart_word_length_t in uart_types.h).

As for the main problem: not an expert, so can't help there, but will keep an eye on it, as I'm studying the uart driver myself, as I'm facing some unrelated issues with it. However, latest master has some important fixes relating to scheduling ([36f49f3](https://github.com/espressif/esp-idf/commit/36f49f361c001b49c538364056bc5d2d04c6f321) 2 days ago)) and being a temporary problem at start, it could be . Does the problem still exist there too?

Moreover, I might be mistaken, but afaik, a function defined static simply means it is only visible in that file. But any local variables (that are not themselves created static), are still temporary variables on the stack. So uart_driver_install() is taking the address of "uart_queue" which will no longer exist once init_port() returns. That could be part of the problem. I don't see why you would create and pass it anyways since it appears you are not using the event model, because you are then immediately calling "uart_read_bytes" rather than waiting on the queue for UART_DATA events. If you pass NULL instead, and give 0 instead of 32, does the problem still exist?

floitsch, 2022-07-25:
Thanks for your feedback.
The given code is extracted from a bigger code-base, which explains most of the weirdness (and the bug with the queue).
That said, I learned that `uart_flush_input` was done by `uart_driver_install`.

I updated my initial report.
The newest master is definitely better. Interestingly, it now seems to have the inverse problem: it works fine the first time after `monitor`, but then fails with `esp_restart` or when recycling with the reset button.

Having the driver in IRAM does not seem to make any difference.
Uploaded the sdkconfig I used.
I also changed the `uart_driver_install` line to use IRAM:
```
ESP_ERROR_CHECK(uart_driver_install(port, buffer_size, buffer_size, 0, NULL, ESP_INTR_FLAG_IRAM));
```

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/9179626/sdkconfig.zip)

moefear85, 2022-07-25:
and setting the pins after install (instead of before) also has no effect? I'll check later if the clock even allows 400 baudrate, but I'm not from espressif. Either way I'm sure they've read this and if they think there is reason to, they would reply.