espressif/esp-idf #9398: BCM_STRNCPY_S 'strncpy' output may be truncated

dizcza, 2022-07-20:
I'm trying to build a project with esp-idf master branch. The last successful build was on Feb 25th, also master branch.

The compilation gives me the following error:

```
In file included from /home/dizcza/tools/esp-idf/components/bt/host/bluedroid/common/include/common/bt_target.h:41,
                 from /home/dizcza/tools/esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c:26:
/home/dizcza/tools/esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c: In function 'bta_dm_discover_device':
/home/dizcza/tools/esp-idf/components/bt/host/bluedroid/stack/include/stack/bt_types.h:35:34: error: 'strncpy' output may be truncated copying 247 bytes from a string of length 248 [-Werror=stringop-truncation]
   35 | #define BCM_STRNCPY_S(x1,x2,x3)  strncpy((x1),(x2),(x3))
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
/home/dizcza/tools/esp-idf/components/bt/host/bluedroid/bta/dm/bta_dm_act.c:2635:9: note: in expansion of macro 'BCM_STRNCPY_S'
 2635 |         BCM_STRNCPY_S((char *)p_msg->disc_result.result.disc_res.bd_name,(char *)bta_dm_search_cb.peer_name, (BD_NAME_LEN - 1));
```

Wth-Esp, 2022-07-21:
It's okay, the copies lengh is 247, smaller than 248, because the last char of the device name should be '/0'.
So, we only copy the first 247 chars.

dizcza, 2022-07-21:
> It's okay, the copies lengh is 247, smaller than 248, because the last char of the device name should be '/0'. So, we only copy the first 247 chars.

This results in the error.

Wth-Esp, 2022-07-21:
I have checkout to the master branch and test the a2dp_sink demo in IDF, this error doesn't occur.

Could you please `git pull` on master or use release/v4.4 to build again?

And could you tell me more information about your compiler?

dizcza, 2022-07-21:
@Wth-Esp please build the "bt_spp_acceptor" example (master branch) with the "Optimize for performance (-O2)" compiler flag.

And you shall see.

Wth-Esp, 2022-07-22:
[fix_compilier_error.zip](https://github.com/espressif/esp-idf/files/9164971/fix_compilier_error.zip)
Please try this patch and I have a MR on the master branch to fix this issue.

Thanks you very much!

dizcza, 2022-07-22:
The fix solves the issue.
I noticed though you're calling `osi_malloc` without freeing the memory elsewhere.

Wth-Esp, 2022-07-22:
[fix_compilier_error.zip](https://github.com/espressif/esp-idf/files/9165253/fix_compilier_error.zip)
yes， I forgot free it, but free is not so easy in `bta_sys_sendmsg()`, so I change the fix.

Please check it out.

dizcza, 2022-07-22:
It works.
I'm also not sure if these lines - `sec_event.auth_cmpl.bd_name[sizeof(sec_event.auth_cmpl.bd_name) -1] = '\0';` - are needed.

Wth-Esp, 2022-07-22:
[bug_fix.zip](https://github.com/espressif/esp-idf/files/9166288/bug_fix.zip)
it's not necessary when the last byte of `BD_NAME` is set to 0. And from the spec, the user friendly name should always be  248 long.

This patch change is all synced into the MR.

dizcza, 2022-07-22:
Well, there are too much changes to check. Let some other devs to have a closer look.
The build is fine.

closed 2022-08-04 by espressif-bot

martinberlin, 2022-09-02:
Can this fix be added also in the v5.0 release also?
I'm hitting exactly the same when building for target esp32 and using bt (BLE) component.

Wth-Esp, 2022-09-05:
@martinberlin oh sure, let me give it a `cherry-pick`. When I was arranging the fix, the release/v5.0 is not released yet.

martinberlin, 2022-09-05:
Hi @Wth-Esp 
My apologies maybe it was because I was missing that commit. Just updated idf v5 to the latest stand. Just tried again compiling this [example](https://github.com/martinberlin/epdiy-rotation/tree/lilygos3/examples/ble-image) that is a download and decompress JPG, and could not reproduce again. Before I was hitting exactly the same error.