espressif/esp-idf #9399: Is WIFI_REASON_ASSOC_EXPIRE not a reason to retry?

homedad, 2022-07-20:
## Environment

Environment: N/A
IDF Version: master and at least since 4.1

See https://github.com/espressif/esp-idf/blob/508661b03eec6e15af66f77050e699a7658b9793/components/wifi_provisioning/src/manager.c#L877

## Problem Description

While using the wifi provisioning manager to re-provision to WiFi the user received a WIFI_PROV_STA_AUTH_ERROR while the logs indicated a "Disconnect reason:" 4 (WIFI_REASON_ASSOC_EXPIRE) about 1.0 seconds after connect attempt.

In wifi_prov_mgr_event_handler_internal() the provisioner's disconnect reason is used to identify the problem class (auth or not found), or perform a retry.

According to the WiFi Driver documentation (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#association-phase), this is due to a timeout during the association phase.

Likewise WIFI_REASON_AUTH_EXPIRE is due to a timeout during the authentication phase.

Shouldn't a timeout be retried? Maybe all of the timeouts below be retried?

When we get a WIFI_REASON_AUTH_EXPIRE or WIFI_REASON_ASSOC_EXPIRE during re-provisioning, the user shouldn't need to restart provisioning.

Excerpt from wifi_prov_mgr_event_handler_internal() from master below:

See https://github.com/espressif/esp-idf/blob/508661b03eec6e15af66f77050e699a7658b9793/components/wifi_provisioning/src/manager.c#L877

    ```
    switch (disconnected->reason) {
        case WIFI_REASON_AUTH_EXPIRE:
        case WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT:
        case WIFI_REASON_AUTH_FAIL:
        case WIFI_REASON_ASSOC_EXPIRE:
        case WIFI_REASON_HANDSHAKE_TIMEOUT:
        case WIFI_REASON_MIC_FAILURE:
            ESP_LOGE(TAG, "STA Auth Error");
            prov_ctx->wifi_disconnect_reason = WIFI_PROV_STA_AUTH_ERROR;
            break;
        case WIFI_REASON_NO_AP_FOUND:
            ESP_LOGE(TAG, "STA AP Not found");
            prov_ctx->wifi_disconnect_reason = WIFI_PROV_STA_AP_NOT_FOUND;
            break;
        default:
            /* If none of the expected reasons,
             * retry connecting to host SSID */
            prov_ctx->wifi_state = WIFI_PROV_STA_CONNECTING;
            esp_wifi_connect();
        }
```

### Expected Behavior
Retry the connect with the same credentials because the error was not permanent.
 
### Actual Behavior
Timeout in connect phase is handled as a permanent error

### Steps to reproduce

This is a tough one. This error is only happening a customer site. Possibly due to a router that is slow at the auth and associate phases.

## Debug Logs

```
I (32072) wifi: new: <1,0>, old: <1,1>, ap: <255,255> , sta: <1,0>, prof: 1
I (33202) wifi:state: init -> auth (bO)
I (33212) wifi:state: auth -> assoc (0)
I (34212) wifi:state: assoc -› init (400)
I (34222) wifi:new: <1,0>, old: <1,0>, ap: <255,255>, sta: <1,0>, prof: 1
I (34222) WIFI: Disconnected
E (34222) wifi_prov_mgr: STA Disconnected
E (34232) wifi_prov_mgr: Disconnect reason :4
E (34232) wifi_prov_mgr: STA Auth Error
E (34232) PROVISION: Provisioning failed!
                  Reason (0): Wi-Fi station authentication failed
```

## Other items if possible

N/A

hmalpani, 2022-08-02:
Hello @homedad ,
It is a little difficult to reproduce this issue. I am sharing a patch with you. Can you please try it out and share the results?
Also, I think timeout can also occur when the AP is stopped during provisioning, so retrying might not be a good option.

[GH9399.txt](https://github.com/espressif/esp-idf/files/9241576/GH9399.txt)

hmalpani, 2022-08-08:
> Hello @homedad , It is a little difficult to reproduce this issue. I am sharing a patch with you. Can you please try it out and share the results?

Hello @homedad,
Any update on this?

homedad, 2022-09-02:
Looking at it now, thank you

hmalpani, 2022-09-21:
Hello @homedad ,
Any update?

homedad, 2022-09-24:
It works! Thanks for your help.

Tested locally where I have no WiFi problems, and at the site where
association takes > 1s (the association timeout). Works great.

closed 2022-10-14 by espressif-bot