espressif/esp-idf #9400: ESP32-C3: Wifi watchdog not trigerred, stuck while connecting to Wifi :ESP-IDF v4.4

MMucka, 2022-07-20:
## Environment

- Development board:      AiThinker ESP-C3-M1, AiThinker ESP-C3-12F, Custom board
- IDF versionv:    4.4.1-262-g5e6cffbb14
- Build System:   idf.py
- Power Supply:  USB, external 5V, external 3.3V

## Our use case:
We have battery powered device with AiThinker module ESP-C3-M1.

 - every 1s collect data (PM and light sleep enabled)
 - after 60s start wifi, connect to AP (STA mode), send data to cloud, disconnect and stop wifi

## Problem:
After arround 2h of work the wifi task got stack and watchdog not trigerred

```
E (2109930) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (2109930) task_wdt:  - IDLE (CPU 0)
E (2109930) task_wdt: Tasks currently running:
E (2109930) task_wdt: CPU 0: wifi
```
	
### Possible solution:
Enable Watchdog Panic and reset device when watchdog not trigerred. 
The problem is, that we loose data, which were not send. (arround 60 samples).

Is it possible to have some handler on Wifi watchdog or on Fatal error? 
It would be great to be abble at least save the data to SPIFFS or restart only Wifi.

### Expected Behavior
Work without stuck several hours.

### Actual Behavior
After ~120 mins stops working.

### Steps to reproduce

1. connect to Wifi
2. stop Wifi
3. light sleep 
4. again from 1.

## Debug Logs
[Debug log with Watchdog Panic and reset](https://gist.github.com/MMucka/b396b98608c6eb6f3e5fc04c2ebbdc19)

```
I (562698) wifi:mode : sta (a0:76:4e:4e:52:78)
I (562698) wifi:enable tsf
I (562708) wifi:Set ps type: 2
I (562718) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
I (562718) wifi:state: init -> auth (b0)
I (562728) wifi:state: auth -> init (200)
I (562728) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:6
E (567848) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (567848) task_wdt:  - IDLE (CPU 0)
E (567848) task_wdt: Tasks currently running:
E (567848) task_wdt: CPU 0: wifi
E (567848) task_wdt: Aborting.
```

## Other items if possible

- [ ] [sdkconfig](https://gist.github.com/MMucka/eb746d56a9e4a1878f4bc3aba1f86c5e)
- [ ] [log](https://gist.github.com/MMucka/b396b98608c6eb6f3e5fc04c2ebbdc19)

MaxwellAlan, 2022-07-21:
Hi @MMucka 
Does this issue always happen in 2h?
Can you try with the latest Master branch?
>Steps to reproduce
    1.connect to Wifi
    2.stop Wifi
    3.light sleep
    4.again from 1.

Can you try add esp_wifi_disconnect before stop wifi?
Any other debug infos are welcome.
Thanks.

MMucka, 2022-07-21:
Hi [MaxwellAlan](https://github.com/MaxwellAlan)
Thank you very much for your advice. 
After adding **esp_wifi_disconnect()** before **esp_wifi_stop()**, the problem dissapeared.

I'm sorry for this issue, it was my fault. 
I wanted to avoid raising WIFI_EVENT_STA_DISCONNECTED, because it calls esp_wifi_connect() on wifi retry connection.
So I have to unregister the handler, or set some flag and handle it when closing wifi connection.

Again, thank you.

closed 2022-07-21 by MMucka