espressif/esp-idf #9402: CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y causes no recovery from light or deep sleep

martaaay, 2022-07-21:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         USB and Battery

## Problem Description

I cannot recover from light or deep sleep with timer wakeup or rtc gpio wakeup. I finally figured out that if I set to
CONFIG_ESP32_RTC_CLK_SRC_INT_RC=y and unset CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256, then wake up works.

### Expected Behavior
That the device will continue where left off after light sleep or reset after waking from deep sleep. 

### Actual Behavior
No response. JTAG off as well. To recover I have to disconnect power and then I can connect over jtag to reprogram.

### Steps to reproduce

1. Enable CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256
2. Call esp_deep_sleep(5*1000*1000);
3. It never recovers.

### Code to reproduce this issue

```cpp
#include "esp_sleep.h"

void app_main()
{
    gpio_set_direction(21, GPIO_MODE_OUTPUT); // For indicating execution with an LED
    gpio_set_level(21, 1);
    vTaskDelay(pdMS_TO_TICKS(5*1000));
    gpio_set_level(21, 0);
    vTaskDelay(pdMS_TO_TICKS(1*1000));
    esp_deep_sleep(5*1000*1000);
}

```

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

AxelLin, 2022-07-21:
This is duplicated of #8007 and #9291, the v4.4 tag does not include the fix but v4.4 branch should already include the fix. Please update your esp-idf to verify.

martaaay, 2022-07-21:
Ah thank you! I did search a fair bit, but sorry that I didn't search quite enough!

closed 2022-07-21 by martaaay