espressif/esp-idf #9405: CONFIG_BT_BTC_TASK_STACK_SIZE too low after latest esp-idf v4.4 branch update

KonssnoK, 2022-07-21:
Using esp32s3 on v4.4 f0a3ccc73201c2d94e135daaa36973f2bd389e3b

```
W (23303) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
E (29323) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
W (29404) BT_SMP: Non bonding: No keys will be exchanged
W (31563) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK

***ERROR*** A stack overflow in task BTC_TASK has been detected.

Backtrace:0x40375d92:0x3fcbd1e00x4037f0b1:0x3fcbd200 0x40382aa2:0x3fcbd220 0x4038108b:0x3fcbd2a0 0x4037f168:0x3fcbd2c0 0x4037f15e:0x3c151622  |<-CORRUPTED
0x40375d92: panic_abort at C:/src/esp-idf/components/esp_system/panic.c:402

0x4037f0b1: esp_system_abort at C:/src/esp-idf/components/esp_system/esp_system.c:128

0x40382aa2: vApplicationStackOverflowHook at C:/src/esp-idf/components/freertos/port/xtensa/port.c:394

0x4038108b: vTaskSwitchContext at C:/src/esp-idf/components/freertos/tasks.c:3505

0x4037f168: _frxt_dispatch at C:/src/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4037f15e: _frxt_int_exit at C:/src/esp-idf/components/freertos/port/xtensa/portasm.S:231

```

rahult-github, 2022-07-25:
Hi @KonssnoK ,

I tried multiple times on the commit you mentioned for the wifi_prov_mgr example. Since the error prints shared in the issue has bluedroid, i switched the code to use bluedroid ( instead of Nimble).  However, for multiple phones i am not observing this issue in my setup. Can you please share some more updates like:

1. Is this a once issue, or you able to observe it 100% 
2. Is this issue, seen with specific phone OR specific AP in the vicinity ?
3. Please share the sdkconfig being used at your end.
4. Can you please share complete console logs , and elf file ? It would be helpful if you able to share debug enabled logs.

Thanks. 

[s3_bluedroid_success_log.txt](https://github.com/espressif/esp-idf/files/9179379/s3_bluedroid_success_log.txt)

KonssnoK, 2022-07-25:
- 100% of the times prior to updating the task size
- i'm using the provisioning app, no other phone or ap involved
-attached
[sdconfig_stackoverflow.txt](https://github.com/espressif/esp-idf/files/9181439/sdconfig_stackoverflow.txt)
- that i can't do because of the amount of custom code we have. If you still have the issue after aligning the sdkconfig i'll find some time to make a dedicated example!

Thanks!

rahult-github, 2022-07-26:
Hi @KonssnoK 

 The default provisioning example app available in IDF works ok with the stack size as 3072.  Since i infer from above comment, that there is some more application side changes, please confirm if any of below is being done:
 
 1. Is the application making any function calls in the stack context itself ? E.g. in a event posted from stack to application, a new function call (say mbedtls connection functionality) is getting invoked ? 
 2. Are there any new function calls being done by application but not in a separate thread , but using stack's thread context only?
 3. Are there extra logging information added  ( say printf ) which is getting executed , while it is still stack's context ?   Such functionalities can lead to increase in size.  

> If you still have the issue after aligning the sdkconfig i'll find some time to make a dedicated example! 

This would be very helpful. I tried the sdkconfig you provided, and it works ok at my end. 

Thanks,

QuentinFarizon, 2022-08-19:
I have the issue with the same scenario (during provisioning, after giving POP and before selecting the Wifi), but with a stock overflow on "ble" task :

```
I (2720) NimBLE: GAP procedure initiated: stop advertising.

I (2730) NimBLE: GAP procedure initiated: advertise; 
I (2730) NimBLE: disc_mode=2
I (2730) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256
I (2740) NimBLE: 

D (2750) protocomm_nimble: Minimum free heap size = 200576, free Heap size = 200736
D (4570) protocomm_nimble: Inside BLE connect w/ conn_id - 1
D (5510) protocomm_nimble: Write attempt for uuid = 1775ff53-6b43-439b-877c-060f2d9bed07, attr_handle = 12, data_len = 3
D (5510) protocomm_nimble: UUID did not match... 53
D (5520) protocomm_nimble: UUID did not match... 53
D (5520) protocomm_nimble: UUID did not match... 53
D (5530) protocomm_nimble: UUID (0x53) matched with proto-name = proto-ver
D (5530) protocomm: VER_EP Req handler returned 0
D (5590) protocomm_nimble: Read attempted for characteristic UUID = 1775ff53-6b43-439b-877c-060f2d9bed07, attr_handle = 12
D (5680) protocomm_nimble: Read attempted for characteristic UUID = 1775ff53-6b43-439b-877c-060f2d9bed07, attr_handle = 12
D (5770) protocomm_nimble: Read attempted for characteristic UUID = 1775ff53-6b43-439b-877c-060f2d9bed07, attr_handle = 12
D (11480) protocomm_nimble: Write attempt for uuid = 1775ff51-6b43-439b-877c-060f2d9bed07, attr_handle = 6, data_len = 41
D (11480) protocomm_nimble: UUID did not match... 51
D (11490) protocomm_nimble: UUID (0x51) matched with proto-name = prov-session
D (11490) security1: Request to handle setup0_command
D (11500) gdma: new group (0) at 0x3fcb0ed4
D (11500) gdma: new pair (0,0) at 0x3fcb0f10
D (11510) gdma: new tx chan
***ERROR*** A stack overflow in task ble has been detected.

Backtrace:0x40375d1e:0x3fcaf2c00x4037e70d:0x3fcaf2e0 0x40382162:0x3fcaf300 0x403806f2:0x3fcaf380 0x4037e7c4:0x3fcaf3a0 0x4037e7ba:0x00000000  |<-CORRUPTED
0x40375d1e: panic_abort at /home/quentin/esp/esp-idf/components/esp_system/panic.c:402

0x4037e70d: esp_system_abort at /home/quentin/esp/esp-idf/components/esp_system/esp_system.c:128

0x40382162: vApplicationStackOverflowHook at /home/quentin/esp/esp-idf/components/freertos/port/xtensa/port.c:394

0x403806f2: vTaskSwitchContext at /home/quentin/esp/esp-idf/components/freertos/tasks.c:3504

0x4037e7c4: _frxt_dispatch at /home/quentin/esp/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4037e7ba: _frxt_int_exit at /home/quentin/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231
```

rahult-github, 2023-08-10:
Hi, since the task size is configurable, users can increase it as per their use case / usage. So closing this issue. Feel free to reopen in case of any further issue.

closed 2023-08-10 by rahult-github

KonssnoK, 2023-08-10:
sorry we couldn't move on with this, we are too busy with other aspects of the firmware at the moment. We'll reopen when needed.
Thanks