espressif/esp-idf #9406: Unknown CMake command "idf_component_optional_requires".

5abina, 2022-07-21:
IDF 4.3
IDE: Visual Studio Code

Using latest IDF on Visual Studio Code

When building pre_encrypted_ota from examples folder:
URL: https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota

### Expected Behavior
Build example

### Actual Behavior
CMake Error from esp/esp-idf/examples/common_components/protocol_examples_common/CMakeLists.txt:20 (idf_component_optional_requires):
  Unknown CMake command "idf_component_optional_requires".

### Steps to reproduce

1. git clone repository: https://github.com/espressif/esp-idf/tree/master/examples/system/ota/pre_encrypted_ota
2. Build project

igrr, 2022-07-21:
Hi @5abina, it seems that you are trying to compile an example from the `master` branch of this repository with IDF v4.3. In most cases this combination won't work, since examples are generally not backwards compatible. You have to use examples from the same version of ESP-IDF as the one you have installed.

5abina, 2022-07-22:
Hello @igrr. Thank you for your help. 
I have just built a binary using tools from the master branch. The problem has been solved.

Kind regards
Daniel

Alvin1Zhang, 2022-07-25:
Thanks for reporting, and glad to hear the issue has been solved, feel free to reopen.

closed 2022-07-25 by Alvin1Zhang