espressif/esp-idf #9409: Two pairing request during bluetooth secure pairing

Steven35700, 2022-07-22:
## Environment

- Module or chip used:  ESP32-S3
- IDF version : v4.4.1

## Problem Description

During the bluetooth pairing of the "ble50_security_server" example provided by ESP-IDF, there are two pairing requests made which can lead to confusion for the user. The user must therefore click twice on "Allow and associate the connection...". I don't think this is normal. 

[image]

### Expected Behavior

Normally we should only have one matching request.

### Actual Behavior

Currently, we have two pairing requests.

### Steps to reproduce

1. Build and flash "ble50_security_server" example.
2. Try to connect with an app like nrfconnect.

esp-cjh, 2022-07-25:
I follow you steps but can't reproduce the problem. What phone you are using? Have you modified this example?

esp-cjh, 2022-07-26:
Could you please provide the source code of your example?

Steven35700, 2022-07-26:
Hi,

I'm using a Redmi Note 9S from Xiaomi. I could carry out in the next few days a test with another phone.
I did not make any changes to the "ble50_security_server" example of ESP-IDF v4.4.1.
Moreover, since I upgraded to v4.4.1 (before I was in v4.4), I have a BLE error and warnings when connecting as you can see below.

[image]

Steven35700, 2022-07-26:
Here is the source code, but no changes have been made!

```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_bt.h"

#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
#include "ble50_sec_gatts_demo.h"

#define GATTS_TABLE_TAG "SEC_GATTS_DEMO"

#define HEART_PROFILE_NUM                         1
#define HEART_PROFILE_APP_IDX                     0
#define ESP_HEART_RATE_APP_ID                     0x55
#define HEART_RATE_SVC_INST_ID                    0
#define EXT_ADV_HANDLE                            0
#define NUM_EXT_ADV_SET                           1
#define EXT_ADV_DURATION                          0
#define EXT_ADV_MAX_EVENTS                        0

#define GATTS_DEMO_CHAR_VAL_LEN_MAX               0x40

static uint16_t profile_handle_table[HRS_IDX_NB];

static uint8_t ext_adv_raw_data[] = {
        0x02, 0x01, 0x06,
        0x02, 0x0a, 0xeb, 0x03, 0x03, 0xab, 0xcd,
        0x11, 0X09, 'E', 'S', 'P', '_', 'B', 'L', 'E', '5', '0', '_', 'S', 'E', 'R', 'V', 'E', 'R',
};

static esp_ble_gap_ext_adv_t ext_adv[1] = {
    [0] = {EXT_ADV_HANDLE, EXT_ADV_DURATION, EXT_ADV_MAX_EVENTS},
};

esp_ble_gap_ext_adv_params_t ext_adv_params_2M = {
    .type = ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE,
    .interval_min = 0x20,
    .interval_max = 0x20,
    .channel_map = ADV_CHNL_ALL,
    .filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy = ESP_BLE_GAP_PHY_1M,
    .max_skip = 0,
    .secondary_phy = ESP_BLE_GAP_PHY_2M,
    .sid = 0,
    .scan_req_notif = false,
    .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
};

struct gatts_profile_inst {
    esp_gatts_cb_t gatts_cb;
    uint16_t gatts_if;
    uint16_t app_id;
    uint16_t conn_id;
    uint16_t service_handle;
[... 444 lines cut ...]
    //set static passkey
    uint32_t passkey = 123456;
    uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
    uint8_t oob_support = ESP_BLE_OOB_DISABLE;
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
    /* If your BLE device acts as a Slave, the init_key means you hope which types of key of the master should distribute to you,
    and the response key means which key you can distribute to the master;
    If your BLE device acts as a master, the response key means you hope which types of key of the slave should distribute to you,
    and the init key means which key you can distribute to the slave. */
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));

    /* Just show how to clear all the bonded devices
     * Delay 30s, clear all the bonded devices
     *
     * vTaskDelay(30000 / portTICK_PERIOD_MS);
     * remove_all_bonded_devices();
     */
}
```

esp-cjh, 2022-07-27:
Thanks! Could you please provide the complete log file?

Steven35700, 2022-07-27:
Hi, here are the [ESP32-S3-logs.txt](https://github.com/espressif/esp-idf/files/9197115/ESP32-S3-logs.txt) logs and the [nrfconnect-logs.txt](https://github.com/espressif/esp-idf/files/9197116/nrfconnect-logs.txt) logs

esp-cjh, 2022-07-28:
I checked the log and found no errors. Could you please capture the air interface packets and provide the capture file? You can use ellisys , nrf sniffer or other capture tools.

esp-cjh, 2023-05-24:
@Steven35700
There is no bug on ESP32-S3. I capture the air packet and found that the ESP32-S3 sent security request once.
I suspect the reason is that the phone or nRFConnect app processed security request twice.

jboirazian, 2023-07-13:
Any news on this issue ? I have also been experiencing the same thing on multiple Android devices. I have also been experiencing it on the  v5.0 and v5.1 versions , on multiple devices and brands (Motorola one macro , Motorola G72 , Samsung Galaxy S11 , Samsung a23 ). Steps are the same as the ones described on the espressiff forums above:

1. [ESP32] Run GATT security server
2. [nRF App] scan and found the ESP32's ESP_BLE_SECURITY GATT server
3. [nRF App] press connect to the GATT server
4. [ESP32] event ESP_GATTS_CONNECT_EVT is seen, i.e. connected
5. [Android] pops up "Bluetooth Pairing Request" as expected
6. [Android] User presses "OK" to confirm pairing
7. [Android] Step 6 has no effect
8. [Android] pops up "Bluetooth Pairing Request" the second time
9. [Android] User presses "OK" to confirm pairing the second time
10. [ESP32] ESP_GAP_BLE_KEY_EVT and ESP_GAP_BLE_AUTH_CMPL_EVT are seen. It shows bonding success
11. [Android] bonding is finished on the Android side as well

Linux PCs requiere and IOS devices requieres pair request.

esp-cjh, 2024-03-04:
@Steven35700 @jboirazian 
I have reproduce the issue and capture the air interface packets. And found that the ESP32 sent security request once. So there is no problem with the ESP32, maybe you can find out if there is a problem with the nRFConnect application.

JanekGraff, 2024-04-10:
Same issue here (with a Fairphone 5 running android 13 and esp-idf v5.0.4), i found (from the forum post about this issue) that removing the call to `esp_ble_set_encryption()` on the `ESP_GATTS_CONNECT_EVT` prevents the pairing notifications (both of them). However in that case the device is not bonded (as in there is no key exchange between esp and android) and the android phone will not show up when calling `show_bonded_devices()` and thus there is no way to distinguish between already bonded devices.