espressif/esp-idf #9410: Expose NimBLE config: MYNEWT_VAL_BLE_GATT_MAX_PROCS

tcbennun, 2022-07-22:
I have an application where an ESP32-C3 acts as a BLE GATT client, connecting to multiple GATT servers and performing various tasks.

I often find that I'm encountering the upper limit of `MYNEWT_VAL_BLE_GATT_MAX_PROCS`, which is set to 4: [esp_nimble_cfg.h:543](https://github.com/espressif/esp-idf/blob/075e0729de31133dead0081d62969858c20b1cf4/components/bt/host/nimble/port/include/esp_nimble_cfg.h#L543). For example, I may want to initiate reads for 5 servers simultaneously.

Could this be something made configurable by adding it to [Kconfig](https://github.com/espressif/esp-idf/blob/075e0729de31133dead0081d62969858c20b1cf4/components/bt/host/nimble/Kconfig.in)? Is there a controller limitation affecting this?

The only workaround I can see for this is, short of keeping a fork of esp-idf, is to override `MYNEWT_VAL_BLE_GATT_MAX_PROCS` in my project `CMakeLists.txt`, which is messy:

```cmake
target_compile_definitions(__idf_bt PRIVATE "MYNEWT_VAL_BLE_GATT_MAX_PROCS=16")
```

(This workaround doesn't even work if I'm using ESP-IDF in a non-conventional way, e.g. calling CMake manually, as in a cargo-first Rust project.)

rahult-github, 2022-07-22:
Hi @tcbennun , 
 Thanks for raising this request. Will analyze the requirement and add needed implementation for this.

rahult-github, 2022-08-16:
https://github.com/espressif/esp-idf/commit/2be66a6151c4fa4dc0da57f10950f544615fd00e commit done on master to handle this . Closing issue.

closed 2022-08-16 by rahult-github