espressif/esp-idf #9411: ESP32S3 _DoubleExceptionVector on esp_flash_erase_region

KonssnoK, 2022-07-22:
running on latest v4.4 f0a3ccc73201c2d94e135daaa36973f2bd389e3b

When running esp_https_ota the following issue is triggered sometimes
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
0x403743c0: _DoubleExceptionVector at C:/src/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:461

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2fe8
entry 0x403c9954
```

since it was not clear what was triggering it, i added prints around to discover that the issue is being triggered only when the function 
`esp_flash_erase_region` performs a yield

```
I (11:48:34.032) esp_https_ota: Starting OTA...
I (11:48:34.033) esp_https_ota: Writing to partition subtype 16 at offset 0x220000
I (11:48:34.044) esp_https_ota: 1 3FCDAEEC 121
I (11:48:34.045) esp_ota_ops: need erase 0 0 0
I (11:48:34.045) spi_flash: erase region
I (11:48:34.050) spi_flash: yield
I (11:48:34.053) spi_flash: done 0
I (11:48:34.058) spi_flash: start
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
0x403743c0: _DoubleExceptionVector at C:/src/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:461

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2fe8
entry 0x403c9954
```

The print "start" is before
```
        ESP_LOGI(TAG, "start");
        err = rom_spiflash_api_funcs->start(chip);
        if (err != ESP_OK) {
            break;
        }
```

I'm still investigating,
If you have a better way to debug the trap in trap please let me know.

KonssnoK, 2022-07-22:
More logs where it doesn't fail immediately

```
W (12:09:59.229) esp_https_ota: Continuing with insecure option because CONFIG_OTA_ALLOW_HTTP is set.
I (12:10:00.283) HTTP_CLIENT: Body received in fetch header state, 0x3fcd1c8b, 469
I (12:10:00.284) esp_https_ota: Starting OTA...
I (12:10:00.286) esp_https_ota: Writing to partition subtype 16 at offset 0x220000
I (12:10:00.294) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.299) esp_ota_ops: need erase 0 0 0
I (12:10:00.304) spi_flash: erase region
I (12:10:00.308) spi_flash: yield
I (12:10:00.312) spi_flash: done 0
I (12:10:00.316) spi_flash: start
I (12:10:00.374) spi_flash: end 0 0
I (12:10:00.375) esp_ota_ops: erased
I (12:10:00.375) esp_ota_ops: write 3FCE0C78 0 3FCD7C64 121
I (12:10:00.379) esp_ota_ops: written
W (12:10:00.383) safety: NOISE, Safety relay moved 3-up 2-down times (total 4-up 4-down)
I (12:10:00.383) esp_https_ota: Written image length 578
I (12:10:00.395) mqtt_app: sent publish returned msg_id=61526
I (12:10:00.399) esp_https_ota: 1 3FCD7C64 121
W (12:10:00.405) safety: NOISE, Safety relay moved 0-up 1-down times (total 4-up 5-down)
I (12:10:00.409) esp_ota_ops: need erase 0 0 121
I (12:10:00.420) mqtt_app: sent publish returned msg_id=41466
I (12:10:00.423) esp_ota_ops: erased
I (12:10:00.433) esp_ota_ops: write 3FCE0C78 121 3FCD7C64 121
I (12:10:00.440) esp_ota_ops: written
I (12:10:00.443) esp_https_ota: Written image length 867
I (12:10:00.450) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.454) esp_ota_ops: need erase 0 0 242
I (12:10:00.458) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=61526
I (12:10:00.461) esp_ota_ops: erased
I (12:10:00.470) esp_ota_ops: write 3FCE0C78 242 3FCD7C64 121
I (12:10:00.477) esp_ota_ops: written
I (12:10:00.481) esp_https_ota: Written image length 1156
I (12:10:00.487) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.492) esp_ota_ops: need erase 0 0 363
I (12:10:00.497) esp_ota_ops: erased
I (12:10:00.501) esp_ota_ops: write 3FCE0C78 363 3FCD7C64 121
I (12:10:00.508) esp_ota_ops: written
I (12:10:00.512) esp_https_ota: Written image length 1445
I (12:10:00.522) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.523) esp_ota_ops: need erase 0 0 484
I (12:10:00.524) mqtt_app: MQTT_EVENT_PUBLISHED, msg_id=41466
I (12:10:00.529) esp_ota_ops: erased
I (12:10:00.538) esp_ota_ops: write 3FCE0C78 484 3FCD7C64 121
I (12:10:00.546) esp_ota_ops: written
I (12:10:00.549) esp_https_ota: Written image length 1734
I (12:10:00.556) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.560) esp_ota_ops: need erase 0 0 5A5
I (12:10:00.565) esp_ota_ops: erased
I (12:10:00.569) esp_ota_ops: write 3FCE0C78 5A5 3FCD7C64 121
I (12:10:00.577) esp_ota_ops: written
I (12:10:00.580) esp_https_ota: Written image length 2023
I (12:10:00.587) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.591) esp_ota_ops: need erase 0 0 6C6
I (12:10:00.596) esp_ota_ops: erased
I (12:10:00.600) esp_ota_ops: write 3FCE0C78 6C6 3FCD7C64 121
I (12:10:00.608) esp_ota_ops: written
I (12:10:00.611) esp_https_ota: Written image length 2312
I (12:10:00.618) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.622) esp_ota_ops: need erase 0 0 7E7
I (12:10:00.627) esp_ota_ops: erased
I (12:10:00.631) esp_ota_ops: write 3FCE0C78 7E7 3FCD7C64 121
I (12:10:00.639) esp_ota_ops: written
I (12:10:00.642) esp_https_ota: Written image length 2601
I (12:10:00.649) esp_https_ota: 1 3FCD7C64 121
I (12:10:00.653) esp_ota_ops: need erase 0 0 908
I (12:10:00.658) esp_ota_ops: erased
I (12:10:00.662) esp_ota_ops: write 3FCE0C78 908 3FCD7C64 121
[... 105 lines cut ...]
I (12:10:01.248) esp_ota_ops: erased
I (12:10:01.252) esp_ota_ops: write 3FCE0C78 1C39 3FCD7C64 121
I (12:10:01.261) esp_ota_ops: written
I (12:10:01.263) esp_https_ota: Written image length 7803
I (12:10:01.270) esp_https_ota: 1 3FCD7C64 121
I (12:10:01.274) esp_ota_ops: need erase 1 1 1D5A
I (12:10:01.279) esp_ota_ops: erased
I (12:10:01.283) esp_ota_ops: write 3FCE0C78 1D5A 3FCD7C64 121
I (12:10:01.291) esp_ota_ops: written
I (12:10:01.294) esp_https_ota: Written image length 8092
I (12:10:01.301) esp_https_ota: 1 3FCD7C64 121
I (12:10:01.305) esp_ota_ops: need erase 1 1 1E7B
I (12:10:01.311) esp_ota_ops: erased
I (12:10:01.315) esp_ota_ops: write 3FCE0C78 1E7B 3FCD7C64 121
I (12:10:01.322) esp_ota_ops: written
I (12:10:01.325) esp_https_ota: Written image length 8381
I (12:10:01.332) esp_https_ota: 1 3FCD7C64 121
I (12:10:01.337) esp_ota_ops: need erase 1 2 1F9C
I (12:10:01.342) spi_flash: erase region
I (12:10:01.346) spi_flash: yield
I (12:10:01.350) spi_flash: done 0
I (12:10:01.354) spi_flash: start
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
```

KonssnoK, 2022-07-22:
Apparently the issue is related to high power consumption when switching a GPIO during a FW update.
Still investigating

KonssnoK, 2022-07-22:
Update:
the issue is related on a GPIO that is switched every 10ms.
Removing the gpio_set_level solves the issue (leaving the complete 10ms logic)
The same happens also if the gpio is switched at 50ms.

I tried to put a scope on both PIN3 and GPIO45 to monitor for voltage drops but they do not present any when crashing

chipweinberger, 2022-07-22:
Sounds similar to https://github.com/espressif/esp-idf/issues/9332

KonssnoK, 2022-07-24:
Tomorrow i'll make another test and update.
Currently the suspicion is that having the GPIO pin as input/output is making the ESP go crazy when the capacitors start discharging-
I'll move them to be output only, so that the GPIO45 is detached from feeling what goes on on that line.
Let's see

KonssnoK, 2022-07-25:
The issue seems to be related to the handling an interrupt attached to GPIO1 which triggers when the OTA library is actually erasing the memory. 
Probably a wrong handling of interrupts from Espressif.

KonssnoK, 2022-07-25:
i confirm this being a duplicate of 
https://github.com/espressif/esp-idf/issues/9129

mahavirj, 2022-07-25:
> The issue seems to be related to the handling an interrupt attached to GPIO1 which triggers a trap in trap when the OTA library is actually erasing the memory.

@KonssnoK Can you please share your code that you added for handling GPIO1 interrupt? Please refer to documentation at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/spi_flash_concurrency.html#iram-safe-interrupt-handlers that might help in this context.

KonssnoK, 2022-07-25:
@mahavirj 
the code for the interrupt:
```
    /* Initialize the normal close relay feedback as GPIO input with interrupt */
    gpio_reset_pin(RELAY_FB_GPIO);
    gpio_set_direction(RELAY_FB_GPIO, GPIO_MODE_INPUT);
    
    /* Get interrupt on both rising and falling edge. */
    err = gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
    if (err == ESP_OK) {
        err = gpio_isr_handler_add(RELAY_FB_GPIO, relay_callback, NULL);
        if (err == ESP_OK) {
            gpio_set_intr_type(RELAY_FB_GPIO, GPIO_INTR_ANYEDGE);
            err = gpio_intr_enable(RELAY_FB_GPIO);
            if (err == ESP_OK) {
                ctx.fb.isr_enabled = true;
            }
        }
    }
```

```
static void IRAM_ATTR relay_callback(void *arg)
{
    int level = gpio_get_level(RELAY_FB_GPIO);
    if(level) {
        ctx.fb.pos_edge_cnt++; //Moved to 1
    } else {
        ctx.fb.neg_edge_cnt++; //Moved to 0
    }
}
```

KonssnoK, 2022-07-25:
Attaching an example of what happens to the PWM without having the interrupt installed:
[image]

This should be considered as a secondary problem tho.

mahavirj, 2022-07-25:
@KonssnoK 

If you replace `ESP_INTR_FLAG_IRAM` with 0, do you still see the problem? This will ensure that interrupt is kept disabled when flash cache is disabled. Alternatively, if you want to keep this interrupt enabled whilst flash cache is disabled then please ensure that its interrupt handlers uses all routines which are IRAM resident (please check map file carefully for `gpio_get_level` call once) and data that's DRAM resident. 

Relevant issue pointer: https://github.com/espressif/esp-idf/issues/8801#issuecomment-1106410081

KonssnoK, 2022-07-25:
@mahavirj 
replacing `ESP_INTR_FLAG_IRAM` with zero solves the crash.

How can i make `gpio_get_level` IRAM resident considering it's provided by espressif?

KonssnoK, 2022-07-25:
@mahavirj question: 
we generate the PWM from a high priority task.
Is there any reason why this task shouldn't run (every 10 ms) while there is a flash erase ongoing? Is disabling the cache stopping the scheduler somehow?

KonssnoK, 2022-07-25:
It seems this long thread was based on our ignorance on the ESP32 architecture.
At least I hope it will help somebody else :)
Thanks @mahavirj

closed 2022-07-25 by KonssnoK