espressif/esp-idf #9412: MQTT esp_mqtt_client_publish() resets ESP32-S3

keithco123, 2022-07-22:
## Environment

- Development Kit:   ESP32-S3-DevKit M-1-N8 Development Board
- Kit version 
- Module or chip used:  ESP32-S3-MINI
- IDF version Version: 2.4.2
- Build System: 20220308-1406
- Compiler version:  xtensa-esp32-elf-gcc
- Operating System:     Windows 10 Pro
- (Windows only) environment type: 64-bit x64-based processor
- Using an IDE?:  Yes, Espressif IDE
- Power Supply:         USB

## Problem Description

Every time a publish is sent outside the mqtt_event_handler() it resets the ESP32-S3. I have to wait for WiFi to reconnect.

example code:
`				if( payload_is( "/pool/valve/5/on" ) )
				{
					valve_5_NO();
					**esp_mqtt_client_publish(client, "/pool/topic", "/pool/valve/state/5/on", 0, 0, 1);**
					vTaskDelay(5000 / portTICK_PERIOD_MS);
				}
				else if(payload_is( "/pool/valve/5/off" ) )`

david-cermak, 2022-07-25:
Hi @keithco123 

Could you please give us more details about the problem? Could you please share the logs, when the reset happens? This would help us to understand more about the reason, as from your code it seems like a very common use-case.

> IDF version Version: 2.4.2

Could you please check the exact version, you're using? Please navigate to the IDF tree and type
```bash
git describe
```

The only thing I could think off why the publishing won't work from one thread and works okay from the handler is the stack size. Could you please check that was the task you're calling the publishing from has enough stack size, and/or the reset reason is not stack overflow (see the [docs on fatal errors](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html?highlight=stack%20overflow))

keithco123, 2022-07-28:
Hi David,
Sorry for the delayed response.
Problem solved!
`void payload_publish( const char * topic, const char * payload)
{
	_**_esp_mqtt_client_enqueue_**_(client, topic, payload, 0, 1, 0, true);
}`
"Enqueue" was the ticket.
Thanks,
Keith

closed 2022-07-28 by keithco123