espressif/esp-idf #9413: Size mismatch wifi_ap_record_t  ssid[33] VS wifi_sta_config_t ssid[32] and wifi_ap_config_t  ssid[32]

ivnny, 2022-07-22:
Hi, any reason to have  wifi_ap_record_t  ssid[33] instead of 32 bytes?

MaxwellAlan, 2022-07-26:
Hi @ivnny
Since wifi_sta_config_t ssid[32] is 32byte, someone may config 32 charactor in this without considering '\0' in the last, so in wifi_ap record_t ssid is 33 byte adding '\0'.
Thanks.

ivnny, 2022-07-27:
Is it possible to make it equal next release?

MaxwellAlan, 2022-07-29:
Hi @ivnny,

Since these fields are in NVS now, so it's a breaking changes for release version.

Does this really affect your products? Could you provide more detail about this issue.
Actually we'd not want to change this.

Emill, 2022-08-17:
There is also missing a "length" field for the ssid in some structs (some have it). Importantly, ssid can contain embedded NUL characters since the ssid is just a byte array of arbitrary octets. Using strlen on the ssid, treated as a string is not compliant. It is perfectly valid for an ssid to consist of 32 bytes all containing 0.

MaxwellAlan, 2022-10-19:
> There is also missing a "length" field for the ssid in some structs (some have it). Importantly, ssid can contain embedded NUL characters since the ssid is just a byte array of arbitrary octets. [...]

Yes, you are right. 

Considering that in most cases, the SSID of the router is input through the graphical interface, it is difficult to input NUL characters. So we treate this as a string is a simplified implementation, we will consider to optimize this in the future IDF v6.0.

Thanks.