espressif/esp-idf #9414: failed CmakeFiles/ flash

LuizFernandoDeveloper, 2022-07-22:
# ALREADY TRIED
- [ x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x ] I have searched the issue tracker for a similar issue and not found a similar issue.
##
## Details
-  windows 
- vscode 
- esp idf : 4.4
- 
## Can't flash to model esp : ESP32-WROOM-32D
## log Erro

PS C:\engenharia_de_software\learningespidf\blink> idf.py -p COM3 flash
Executing action: flash
Running ninja in directory c:\engenharia_de_software\learningespidf\blink\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\engenharia_de_software\learningespidf\blink\build\esp-i...artition-table.bin C:/engenharia_de_software/learningespidf/blink/build/blink.bin"
blink.bin binary size 0x2b9f0 bytes. Smallest app partition is 0x100000 bytes. 0xd4610 bytes (83%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\engenharia_de_software\learningespidf\blink\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/engenharia_de_software/learningespidf/blink/build/bootloader/bootloader.bin"
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.1-2\components\esptoo...Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 blink.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3-dev
Serial port COM3
Traceback (most recent call last):
  File "C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py", line 5347, in <module>
    _main()
  File "C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py", line 5340, in _main
    main()
  File "C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py", line 4647, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py", line 114, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py", line 320, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'O sistema nÒo pode encontrar o arquivo especificado.', None, 2)
CMake Error at run_serial_tool.cmake:56 (message):
  C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
  C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash
cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v4.4.1-2\components\esptool_py && C:\Espressif\tools\cmake\3.20.3\bin\cmake.exe -D IDF_PATH="C:/Espressif/frameworks/esp-idf-v4.4.1-2" -D SERIAL_TOOL="C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="C:/engenharia_de_software/learningespidf/blink/build" -P C:/Espressif/frameworks/esp-idf-v4.4.1-2/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
PS C:\engenharia_de_software\learningespidf\blink>
##

dobairoland, 2022-07-22:
The COM3 port which you've specified doesn't exist. Open the Windows device manager before plugging-in the ESP board and see which port is created for it.

LuizFernandoDeveloper, 2022-07-22:
[image]

LuizFernandoDeveloper, 2022-07-22:
so that another "COM" port does not appear

LuizFernandoDeveloper, 2022-07-23:
help

dobairoland, 2022-07-25:
I'm sorry but this isn't an ESP-IDF issue but a Windows one:
```
serial.serialutil.SerialException: could not open port 'COM3': FileNotFoundError(2, 'O sistema nÒo pode encontrar o arquivo especificado.', None, 2)
```
There is nothing we could do about it.

You will find many discussion on the Internet if you search for  "Could not open port 'COM3': FileNotFoundError The system cannot find the specified file". 
 
- You could not have access to the port. Check that your user has access to serial ports.
- You could update the driver for COM3.
- I found one post suggesting that COM3 was disabled and enabling it helped.
- You could try to remove the port in device manager. It will reinstall the driver after you re-attach it.

closed 2022-07-25 by dobairoland

esterlinpol, 2023-04-28:
In case anybody is still having this problem:

Running the ExpressLRS configurator as admin just worked for me!

PWrzesinski, 2023-05-05:
In case this helps anyone: I've had the same issue and it turned out to be a faulty cable

Metdez, 2023-07-02:
Hey I had the same issue and I tried all the recommended advice and that didn't work so I just did it over wifi and that seemed to do the trick. Here's the youtube video I followed and learned from

https://youtu.be/J3Hg2f7RL1A

JD120802, 2023-10-19:
En el Administrador de dispositivos solo le di en "Deshabilitar dispositivo", espere unos minutos y lo volví a habilitar

Recov25, 2023-11-30:
I experienced this too, but my problem was the uart adapter, it was a fake one, and the manufacturer disabled the driver for fake chips (the name of the cable on device manager was "PL2303HXA phased out since 2012. please contact your supplier"), so i had to uninstall the default driver and install a generic one

DanHuang25, 2024-02-23:
可以尝试这个帖子的方法，我就是同样的报错，在这里解决了 https://ask.csdn.net/questions/7907449

lucca-rsilva, 2024-12-16:
Unplugging and plugging back the adapter fixed this for me