espressif/esp-idf #9415: mqtt `select` calls cluttering debug level logs

KaeLL, 2022-07-22:
- IDF version: 36f49f3

## Problem Description

Do these really need to be DEBUG rather than VERBOSE?
```
D (16:26:43.753) TRANSPORT_BASE: poll_read: select - Timeout before any socket was ready!
D (16:26:43.754) TRANSPORT_BASE: poll_read: select - Timeout before any socket was ready!
D (16:26:43.754) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready!
```
Setting these components' log level to INFO would hide other relevant DEBUG level logs.

moefear85, 2022-07-23:
the timeouts are logged using the "TRANSPORT_BASE" tag, not mqtt, so just use:

> esp_log_level_set("TRANSPORT_BASE",ESP_LOG_NONE");

but place it after all setup so it doesn't get changed back.

KaeLL, 2024-03-15:
@moefear85 🤔
[image]
Again I ask: do these really need to be DEBUG rather than VERBOSE?

moefear85, 2024-03-15:
I'm neither a contributor nor maintainer... but you can search for the error message in vscode (or use grep) , then simply replace ESP_LOGD with ESP_LOGV. most likely it would be a one-liner edit.

KaeLL, 2024-03-16:
@euripedesrocha

euripedesrocha, 2024-03-20:
@KaeLL this was already changed in IDF 5.2.

AxelLin, 2024-04-22:
@euripedesrocha 
IDF 5.0 and 5.1 still need fix.

AxelLin, 2024-05-18:
> @euripedesrocha IDF 5.0 and 5.1 still need fix.

@euripedesrocha
There is no update for IDF 5.0 and 5.1.
Now I'm wondering if you stop maintaining esp-mqtt for older release branches?

euripedesrocha, 2024-05-21:
@AxelLin we didn't stop maintaining 5.0 and 5.1 branches. We just decided to not backport this at the moment. I'll reassess and maybe start the backport process.  Sorry for let you request without an answer.

closed 2024-10-01 by euripedesrocha