espressif/esp-idf #9416: Move MCPWM group and timer prescale settings into menuconfig

MarsTechHAN, 2022-07-23:
MCPWM is a very useful component in ESP32. But because the prescale is fixed to some value in IDF, it makes the whole unit much less useful. There is no reason to have a fixed value here, and based on the application, it has to be able to vary. 

For example, I want to make a driver for a motor that I want the driver to run at an ultrasonic frequency, such as 20khz. At this frequency, with a preset value, I only got about 5.5bit of the resolution, which is not enough, but if I change the group prescales to 1, I can get around 10bit of the resolution, which would make it a very good driver.

**Describe the solution you'd like**
https://github.com/espressif/esp-idf/blob/16a4ee7c36a848ca155791677ce011f3ca75c519/components/driver/mcpwm.c#L49-L51
Just moving the following settings to menuconfig would be good enough.  (It would be even nicer if we can set it in the software during the setup, but it will be pretty painful to implement.)

I can even do the PR if you guys think it is a good idea.

suda-morris, 2022-07-25:
Hi @MarsTechHAN 
Can `mcpwm_group_set_resolution` meet your requirement?

MarsTechHAN, 2022-07-25:
Hi. I took a try, but it does not work as expected. For example, my goal is
```C
#define MCPWM_GROUP_CLK_PRESCALE (1)
#define MCPWM_GROUP_CLK_HZ (SOC_MCPWM_BASE_CLK_HZ / MCPWM_GROUP_CLK_PRESCALE)
#define MCPWM_TIMER_CLK_HZ (MCPWM_GROUP_CLK_HZ / 10)
```
So instead of changing mcpwm.c, I add this after mcpwm_init
```C
mcpwm_group_set_resolution(MCPWM_UNIT_0, SOC_MCPWM_BASE_CLK_HZ);
// mcpwm_timer_set_resolution(MCPWM_UNIT_0, MCPWM_TIMER_0, SOC_MCPWM_BASE_CLK_HZ / 10);
// mcpwm_timer_set_resolution(MCPWM_UNIT_0, MCPWM_TIMER_1, SOC_MCPWM_BASE_CLK_HZ / 10);
```
and use this snap of code to read the result.
```C
#ifdef DEBUG
    uint32_t current_duty = mcpwm_ll_timer_get_peak(mcpwm_unit0_hal.dev, 0, false);
    ets_printf("MCPWM Group Prescaler Setting: %d, Timer Prescaler Setting: %d, Current Duty: %d\r\n", mcpwm_ll_group_get_clock_prescale(mcpwm_unit0_hal.dev), mcpwm_ll_timer_get_clock_prescale(mcpwm_unit0_hal.dev, MCPWM_TIMER_0), current_duty);
    if(current_duty != 800) {
        ets_printf("!!!!!! MCPWM Group Prescaler Setting is not correct !!!!!!\n");
        ets_printf("You need to add #define MCPWM_GROUP_CLK_PRESCALE (1) to the esp-idf/components/driver/mcpwm.c\n");
    }
#endif
```

However, the motor does not turn nor does the program read the correct result.
```
MCPWM Group Prescaler Setting: 1, Timer Prescaler Setting: 10, Current Duty: 50
!!!!!! MCPWM Group Prescaler Setting is not correct !!!!!!
You need to add #define MCPWM_GROUP_CLK_PRESCALE (1) to the esp-idf/components/driver/mcpwm.c
```

suda-morris, 2022-07-25:
Override the group resolution and timer resolution should be applied before calling `mcpwm_init`:

[image]

suda-morris, 2022-08-01:
ESP-IDF provides a new MCPWM driver in 5.0, where the new APIs are more flexible, it exposes all hardware resources to users, more configurations, and fewer assumptions.

With regard to the timer resolution, it can be set in the code during driver initialization. https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/mcpwm_timer.h#L34

suda-morris, 2022-08-01:
We will close this issue, as it seems esp-idf has provided the required features already, in both the new driver and legacy driver.

closed 2022-08-01 by suda-morris