espressif/esp-idf #9417: when checkout to v4.4.1, esp-idf compiler error

zhangoneone, 2022-07-23:
idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/one/work/esp32/car_screen/esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/one/work/esp32/car_screen/esp32"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32 is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/one/work/esp32/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/one/work/esp32/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:7 (include)

CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32c3_family is out of date.
  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/one/work/esp32/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/one/work/esp32/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:7 (include)

CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/host/nimble/nimble is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/one/work/esp32/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/one/work/esp32/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:7 (include)

CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_phy/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/one/work/esp32/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/one/work/esp32/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:7 (include)

CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_wifi/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/one/work/esp32/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/one/work/esp32/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:7 (include)

CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/ieee802154/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:174 (__build_get_idf_git_revision)
  /home/one/work/esp32/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/one/work/esp32/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:7 (include)

CMake Warning at /home/one/work/esp32/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/lwip/lwip is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
[... 108 lines cut ...]
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "main" version: 821fda1-dirty
-- Adding linker script /home/one/work/esp32/car_screen/esp32/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/one/work/esp32/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
CMake Error at /home/one/work/esp32/esp-idf/tools/cmake/component.cmake:301 (message):
  Include directory
  '/home/one/work/esp32/esp-idf/components/asio/asio/asio/include' is not a
  directory.
Call Stack (most recent call first):
  /home/one/work/esp32/esp-idf/tools/cmake/component.cmake:472 (__component_add_include_dirs)
  /home/one/work/esp32/esp-idf/components/asio/CMakeLists.txt:24 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/one/work/esp32/car_screen/esp32/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

moefear85, 2022-07-23:
everytime you checkout a different branch, you have to then rerun the install & export scripts, even if switching back/forth between 2 branches. Even then, you have to reconfigure the esp-idf extension, & modify the .vscode folder in your project.

the simple solution to this dilemma is to use docker/containers & the remote containers extension for vscode. there is a section in the idf documentation on using docker.

Dazza0, 2022-08-15:
@zhangoneone When checking out a different branch/tag, please check that you have done the following

- Update submodules using `git submodule update --recursive`
- Run `./install.sh` to make sure all the required tools for the current IDF version are installed
- Run `. ./export.,sh` to export the required paths for the current IDF version
- If you are building an old project, run `idf.py fullclean` to clean the build directory

@zhangoneone  Based on the log, it seems like you need to update the IDF submodules before building. I'm closing this issue as it doesn't seem to be a bug in IDF. Please feel free to reopen if it is a bug.

closed 2022-08-15 by Dazza0