espressif/esp-idf #9418: Error using Winbond memory(SPI) and FatFS, esp-idf 4.4.1

simeondmr, 2022-07-23:
Hello,

I'm using a Winbond memory, using the SPI and FatFS.

I have no problem during the mount of the file system, and I can create correctly the file, but sometimes during file or directory creation I get this error:

```
wl_partition: erase_range - start_address=0x00002000, size=0x00001000, result=0x00000105
wl_flash: erase_sector(557): result = 0x00000105
wl_flash: erase_range(571): result = 0x00000105
ff_diskio_spiflash: wl_erase_range failed (261)
```

Please could you help me?
What could be the problem?

Thanks a lot!

simeondmr, 2022-08-08:
News about this issue?

adokitkat, 2022-08-31:
Hello @simeondmr,

It seems you're getting an error number 261, 0x105 in hex, which is `ESP_ERR_NOT_FOUND`. Could you please share more info? 

- Could you share a minimal piece of code where you can reproduce this issue? 
- Have you tried switching ESP-IDF to master branch and compiling to see if the issue is still present there? 
- What ESP32 board and what SPI Flash memory model do you use?

Thank you.

adokitkat, 2022-09-21:
I could not reproduce the issue because of the lack of detail information. Closing while waiting for the user's response. Feel free to reopen if the problem is still there.

closed 2022-09-21 by adokitkat