espressif/esp-idf #9419: Problem with I2S_PDM.H master

remyhx, 2022-07-23:
Yesterday I updated to the new version of I2S_PDM.H(.C), because of the right channel mono issue. It seems now that this is solved, but the recording seems like a 2x speed. I had one time before, changing DSR was enough to change. But now it seems to not.

ESP-IDF v5.0-dev-4364-g36f49f361c

Microphone used is Vesper VM3011, PDM mode.

```
i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER);
chan_cfg.dma_desc_num   = 3;
chan_cfg.dma_frame_num  = 512; 
```

(I read 1024 bytes a a time)

```    
i2s_new_channel(&chan_cfg, NULL, &rx_handle);

 /* Init the channel into PDM RX mode */
 i2s_pdm_rx_config_t pdm_rx_cfg = {
     .clk_cfg    = I2S_PDM_RX_CLK_DEFAULT_CONFIG(SAMPLE_RATE),
     .slot_cfg   = I2S_PDM_RX_SLOT_DEFAULT_CONFIG(BITS_PER_SAMPLE,  NUM_CHANNELS),   //bits per sample is 16, num chan 1
     .gpio_cfg   = {
         .clk    = WS_PIN,
         .din    = DATA_IN_PIN,
         .invert_flags   = {
             .clk_inv    = false,
         },
     },
 };

pdm_rx_cfg.clk_cfg.dn_sample_mode = I2S_PDM_DSR_16S;  //(8 or 16 no effect
pdm_rx_cfg.slot_cfg.slot_mask = I2S_PDM_SLOT_RIGHT;
    
i2s_channel_init_pdm_rx_mode(rx_handle, &pdm_rx_cfg);
```

Channel is enabled before reading:

```
i2s_channel_enable(rx_handle);`
```

And per reading:

```
i2s_channel_read(rx_handle, (char *)buf, AUDIO_BUF, &bytes_read, portMAX_DELAY);  //AUDIO_BUF = 1024
```

Any idea?

The sound quality before this update was very good. Now I hear a lot of noise, hardware is not changed. Only problem was the MONO slot selection.

L-KAYA, 2022-07-24:
May I ask which target you're using? ESP32 or ESP32S3?

And yes, the latest update has some breaking changes on PDM mode. We did an all-round test and fixed some bugs of the slot sequence, the PDM RX result can be find in the API Reference ([ESP32](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#pdm-rx-usage) or [ESP32S3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/i2s.html#pdm-rx-usage)).

BTW, have you checked if the [i2s_record example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/i2s_recorder) works well or has same issue? I'll look into this issue next week

remyhx, 2022-07-24:
Hi @L-KAYA , thanx for the quick response!

I wasn't aware of the example, but just took a look in it and it was very helpful. I build for the ESP32.

In the previous version with the mono issue for the right channel I read chunks of 1024 each, in the example that's 16*1024. When I change this value to 16*1024 it's ok. But because this record task is inside a task pinned to one core, it also takes a huge toll on stack size.

NB I disabled this part:

```
chan_cfg.dma_desc_num   = 3;
chan_cfg.dma_frame_num  = 512;
```

First impression gain looks better, but so is the background noise now also. I can get rid of it if I extend this 16*1024 sized buffer * 5 and out of the task (in favor of task stack). But reading in little chunks seemed to also do the trick.

L-KAYA, 2022-07-28:
Oh, forgot to ask for the sample rate, according to your description, the noise might be produced by the incomplete data, which mean data might lost, you can refer to the  [how-to-prevent-data-lost](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#how-to-prevent-data-lost) chapter to see how to avoid it.

closed 2025-01-13 by Alvin1Zhang