espressif/esp-idf #9420: Problem with ESP_NIMBLE_HCI.C master branch

remyhx, 2022-07-23:
There seems to be a problem with ESP_NIMBLE_HCI.C after last update ESP-IDF v5.0-dev-4364-g36f49f361c.

When I compiled, `esp_nimble_hci_and_controller_init()` seemed to be non-existing anymore. In the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html?highlight=esp_nimble_hci_init) of the master branch it is mentioned, but after the function overview at the end, only `esp_err_t esp_nimble_hci_init(void)` and `esp_err_t esp_nimble_hci_deinit(void)` seems to be present and declared. Initially I thought there was a cleanup, like ADC, I2S, etc (migration), but those functions were already there before and don't look changed.

Is it possible they are by accident deleted?

GYC-Espressif, 2022-07-25:
Hi, we removed the `esp_nimble_hci_and_controller_init` API, the related documentation is still in the works. 
You can now directly remove the call to the `esp_nimble_hci_and_controller_init` API. Its functionality is currently implemented in `nimble_port_init` which already exists in the nimble. Please note that the nimble submodule needs to be updated synchronously.

AxelLin, 2022-07-25:
@GYC-Espressif 

The esp_nimble_hci_and_controller_init exist in v5.0 branch but does not exist in master now.
This is a problem if I want to complile my application with both branches because checking ESP_IDF_VERSION
on both branch shows the same, 5.0.0.
I usually use #if guard by checking ESP_IDF_VERSION for such API changes, but it does not work for v5.0 and master.
If I change my application code to build with master, it will fail to build with v5.0 branch.
It's not clear to me if such breaking changes will be applied to v5.0 branch or not.

PS. Similar API breaking changes in esp-mqtt submodule as well.

remyhx, 2022-07-25:
@GYC-Espressif that did the trick. So I think it's the same for deinit?

rahult-github, 2023-04-25:
This is now fixed on latest branches. Closing this. Please reopen in case issue is still observed.

closed 2023-04-25 by rahult-github