espressif/esp-idf #9421: Build error if using littlefs

AxelLin, 2022-07-24:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v5.0-dev-4379-g36f49f361c
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

After pulling the latest esp-idf master update, I got below build error:
In file included from /home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/src/esp_littlefs.c:16:
/home/axel/esp/esp-idf-dev/apps/myproj/managed_components/joltwallet__littlefs/src/littlefs_api.h:9:10: fatal error: esp_vfs.h: No such file or directory
    9 | #include "esp_vfs.h"
      |          ^~~~~~~~~~~
compilation terminated.

suda-morris, 2022-07-24:
This is a [breaking change](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/build-system.html#specify-component-requirements-explicitly) brought by esp-idf 5.0. `vfs` component is not a common requirement now, the littlefs needs to specify the dependency explicitly. https://github.com/joltwallet/esp_littlefs/blob/master/CMakeLists.txt#L8

AxelLin, 2022-07-24:
@suda-morris 

If the esp-idf components are maintained by espressif, I would say this is a *BUG* rather than *breaking change*.
You need to make sure the components can be compiled. Probably need to review this issue for all components in IDF component registry.

AxelLin, 2022-07-24:
The littlefs component needs fix.
Is here the correct place to report ESP-IDF managed component build issue?

igrr, 2022-07-24:
Probably https://github.com/joltwallet/esp_littlefs/issues is the right place in this case.

I'd like to clarify that while most of the components currently available in the component registry are maintained by Espressif, there are several exceptions. Littlefs is one of them. You can determine the maintainer of the component by looking at the namespace: e.g. espressif/expat is maintained by Espressif, joltwallet/littlefs is maintained by joltwallet org. Additionally, 'homepage' link on the component's page (https://components.espressif.com/component/joltwallet/littlefs) typically leads to the repository where the component is hosted.

You are right though that in this case we should have been proactive, for example by sending a PR to the littlefs maintainer to prepare for this breaking change.

moefear85, 2022-07-24:
the effort to minimize dependencies and slim down core esp-idf is appreciated alot, atleast by me

pacucha42, 2022-07-25:
Fixed in https://github.com/joltwallet/esp_littlefs/issues/86. @AxelLin thank you for your effort!

AxelLin, 2022-07-25:
Hi @pacucha42 
I still get the same build error.
I'm wondering if I need to wait the update in IDF Component Registry?
(In https://components.espressif.com/component/joltwallet/littlefs , the version: "1.3.1~1" does not exist.)

pacucha42, 2022-07-25:
Right, it's not yet merged - sorry for confusion. I'm reopening this issue until https://github.com/mahavirj/esp_littlefs/commit/fc04bb6153f494c381d24a2632994750c85a6d89 gets merged. Could you use the code patch for the meantime, please?

BrianPugh, 2022-07-25:
merged the PR, and CI pushed `v1.3.1~1` to the idf component registry. @AxelLin please confirm that this resolves the problem.

closed 2022-07-25 by AxelLin

robbawebba, 2022-07-28:
Hi @BrianPugh, I think this version on the component registry is breaking builds for my project due to invalid semantic version format. When performing a clean build with idf.py v4.4, I receive the following traceback:
```
Solving dependencies requirements
CMake Error at /home/rob/esp/esp-idf/tools/cmake/build.cmake:455 (message):
  Traceback (most recent call last):

    File "/usr/lib64/python3.10/runpy.py", line 196, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib64/python3.10/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/__main__.py", line 3, in <module>
      main()
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py", line 110, in main
      args.func(args)
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/prepare_components/prepare.py", line 37, in prepare_dep_dirs
      ComponentManager(args.project_dir).prepare_dep_dirs(
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/core.py", line 325, in prepare_dep_dirs
      downloaded_component_paths = download_project_dependencies(
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/dependencies.py", line 36, in download_project_dependencies
      solution = solver.solve()
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/version_solver/version_solver.py", line 28, in solve
      self.solve_manifest(manifest)
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/version_solver/version_solver.py", line 47, in solve_manifest
      self.solve_component(requirement)
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_manager/version_solver/version_solver.py", line 50, in solve_component
      cmp_with_versions = requirement.source.versions(
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/sources/web_service.py", line 96, in versions
      cmp_with_versions = self.api_client.versions(name, spec, target)
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/idf_component_tools/api_client.py", line 193, in versions
      if semantic_spec.match(Version(version['version'])):
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/semantic_version/base.py", line 105, in __init__
      major, minor, patch, prerelease, build = self.parse(version_string, partial)
    File "/home/rob/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/semantic_version/base.py", line 311, in parse
      raise ValueError('Invalid version string: %r' % version_string)

  ValueError: Invalid version string: '1.3.1~1'

Call Stack (most recent call first):
  /home/rob/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:20 (project)
```
Here is the manifest for my main component, notice that the selected component version is "==1.3.1":
```
## IDF Component Manager Manifest File
dependencies:
  joltwallet/littlefs: "==1.3.1"
  ## Required IDF version
  idf:
    version: ">=4.1.0"
```
I have not yet tried any newer IDF version besides v4.4. Is this version string with a tilde character supported in newer IDF versions? If this version is no longer needed after this bug fix was tested, can the version be removed from the component registry to prevent future builds with IDF v4.4 from breaking? Thanks for taking a look at this issue.

igrr, 2022-07-28:
@robbawebba i think you will either have to lock the version to littlefs==4.1.0, or run `pip install --upgrade idf-component-manager` to get a version of the client which supports the extended component version format.

BrianPugh, 2022-07-28:
@igrr to be clear, does `1.3.1~1` properly parse with up-to-date versions of idf-component-manager? Or, should I just release a `1.3.2`? I don't really get the point of the `~1` when incrementing the patch version would have been sufficient.

robbawebba, 2022-07-28:
@igrr @BrianPugh  Locking the version string does not solve the issue because it seems like the component manager still attempts to parse all versions from the registry (judging by the traceback I shared). I confirmed that updating idf-component-manager, or a fresh install/upgrade of IDF, are the only solutions that work. Thanks for the info and the recommendation :slightly_smiling_face: 

However, this only solves the problem for my installation, and any existing installations are still affected. For example, the v4.4 and v4.4.1 espressif/idf docker images are still affected, but could probably be fixed with a rebuild of the image. IMO increasing the patch version seems to be the safest route since it does not break the Docker images and other existing IDF installations.

BrianPugh, 2022-07-28:
@igrr does me tagging `1.3.2` and deleting `1.3.1~1` sound like a reasonable solution? I'm actually unsure on how to delete a release on the idf component registry.

igrr, 2022-07-28:
@BrianPugh does the joltwallet/littlefs release number (1.3.1) somehow relate to https://github.com/littlefs-project/littlefs release number? If yes, then the best solution would be to keep using the x.y.z part of the release number for the upstream version, and ~w part for the "port" version.
If you aren't tracking the upstream versions, then of course you can bump the version to 1.3.2.

Regarding deleting the existing versions, I'm not sure; @kumekay could you please take a look?

BrianPugh, 2022-07-28:
`joltwallet/esp_littlefs` versions are independent of upstream releases; that said I should probably update my reference soon as it seems like there's been new releases with upstream littlefs.

If @kumekay can confirm that we can delete `1.3.1~1` (I see there's a `delete-version` action in the `idf-component-manager` cli tool), then I'll cut a release for `1.3.2`. I've never explicitly used the manager tool, and the help string doesn't indicate the usage for that command.

kumekay, 2022-07-29:
@BrianPugh Hi Brian, yes, you can use `delete-version` to delete a particular component version. Just keep in mind that after deletion, the same version cannot be uploaded again (for security reasons, once uploaded, a version hash sum should never change)

So uploading `1.3.2` and deleting 1.3.1~1 should work.

We are working on a solution to clarify when the component manager needs an update and print the instructions, but it's not deployed yet. Also, coming versions of the component manager will be better in checking features format.

BrianPugh, 2022-07-29:
@robbawebba @AxelLin 

Version `1.3.2` is released and should resolve these issues. I deleted `1.3.1~1` from the registry.

I couldn't find it anywhere in the documentation, but here was the command I had to run to delete it:

```
export IDF_COMPONENT_API_TOKEN=${YOUR_API_TOKEN_HERE}
python -m idf_component_manager delete-version --name=LittleFS --version=1.3.1~1 --namespace=joltwallet
```

kumekay, 2022-07-29:
@BrianPugh This command is also exposed through idf.py, and it's a bit documented there ` idf.py delete-version --help`

We are working on a separate CLI for component developers to keep `idf.py` focused on project-specific tasks.

BrianPugh, 2022-10-06:
assuming those versions coincide with esp-idf versions, v3.X.X is not supported (both by esp_littlefs, as well as by espressif). v3.3.x (let alone v3.1.X) was dropped in March 2022.