espressif/esp-idf #9424: Problem about esp-lcd flags.swap_color_bytes causing extra line

kaidegit, 2022-07-25:
# My Env:
* idf version : v4.4-263-g000d3823bb-dirty, it should be a clone of release/v4.4 branch
* module : esp32s3-wroom-1u
* screen : RM68120, 8bit parallel mode, 10M speed.

# Problem:
As a 8 bit parallel screen, I should swap the color bit. However, when I use` .flags.swap_color_bytes = 1` to swap it, it may cause some extra lines on the screen. Using `Swap the 2 bytes of RGB565 color` in `menuconfig > Component config > LVGL configuration > Color settings` don't cause these lines .

# Init codes:
```c
    ESP_LOGI(TAG, "Initialize LCD Intel 8080 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
            .dc_gpio_num = LCD_RS_Pin,
            .wr_gpio_num = LCD_WR_Pin,
            .data_gpio_nums = {
                    LCD_D0_Pin,
                    LCD_D1_Pin,
                    LCD_D2_Pin,
                    LCD_D3_Pin,
                    LCD_D4_Pin,
                    LCD_D5_Pin,
                    LCD_D6_Pin,
                    LCD_D7_Pin,
            },
            .bus_width = 8,
            .max_transfer_bytes = LCD_WIDTH * 40 * sizeof(uint16_t)    //最大32640 *2, 受DMA限制
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));

    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i80_config_t io_config = {
            .cs_gpio_num = LCD_CS_Pin,
            .pclk_hz = 10 * 1000 * 1000, // 测试14MHz以上花屏 10MHZ=26.18fps   13MHZ=34.04fps
            .trans_queue_depth = 10,
            .dc_levels = {
                    .dc_idle_level = 1,
                    .dc_cmd_level = 0,
                    .dc_dummy_level = 1,
                    .dc_data_level = 1,
            },
            // .on_color_trans_done = example_notify_lvgl_flush_ready,  //Callback invoked when color data transfer has finished
            // .user_ctx = &disp_drv,                                   //User private data, passed directly to on_color_trans_done’s user_ctx
            .lcd_cmd_bits = 16,
            .lcd_param_bits = 16,
//            .flags.swap_color_bytes = 1,
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));
```

driver codes can be found in [W00ng's repo](https://github.com/W00ng/ESP32-S3-HMI-DevKit)， I just modify the bud width ,io ,and port on lvgl.

using init to swap
[image]
[image]
using LVGL to swap
[image]

suda-morris, 2022-07-25:
I don't have an RM68120 LCD around, but I do have an NT35510, I suppose they two share the same **basic** register layout. I tried the example [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller), with or without enabling the `LV_COLOR_16_SWAP`, and I got the same result, didn't see the strange afterimage,
Could you reproduce the issue with the same example? (you can try the master branch for now, to make sure this is not an issue leftover in the release/v4.4 branch only).

kaidegit, 2022-07-25:
It seems not to happen with esp-idf v5.0....

closed 2022-07-25 by kaidegit

kaidegit, 2022-07-25:
Actually, [this commit](https://github.com/espressif/esp-idf/commit/a019db6880cf0519f310c0aa1b45159f82310931) already solved my problem... release version also have a commit solved this. My idf version is a little bit outdated.

Thanks.

kaidegit, 2022-07-25:
Newest release/v4.4 branch also don't have this problem...Sorry for my outdated idf causing this.

Thanks