espressif/esp-idf #9425: nimble_port_freertos_deinit does not delete the Task

alberk8, 2022-07-25:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/tools/idf-monitor.html), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     ESP32 DeviKit
- Kit version v1
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
     v4.4.1-releasse
- Build System:         CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
The function nimble_port_freertos_deinit does not release the Task.   The problem could be that the function blehr_host_task(void *param) is the task and trying to release the Task while inside of a running task.  

//Detailed problem description goes here.

### Expected Behavior
Task released.

### Actual Behavior
The function infact never returns.

### Steps to reproduce
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/blehr/main/main.c

## Other items if possible

rahult-github, 2022-07-25:
Hi @alberk8 ,   

So, in general,  when nimble_port_freertos_init creates a task  -->  
  `xTaskCreatePinnedToCore(host_task_fn, "ble", NIMBLE_STACK_SIZE,
                NULL, (configMAX_PRIORITIES - 4), &host_task_h, NIMBLE_CORE);`

And in nimble_port_freertos_deinit , it checks if task is present and attempts to delete it -- >

`    if (host_task_h) {
        vTaskDelete(host_task_h);
    }
`

This is upstream mynewt-nimble default behaviour.  As the code is taken as is from upstream nimble, i also suggest opening a ticket at https://github.com/apache/mynewt-nimble for better tracking. 

Thanks

alberk8, 2022-07-25:
Any knowledge why I cannot recreate the ble again after the deinit.

rahult-github, 2022-07-25:
Hi @alberk8  , 

May i know the use case you are attempting ?  Does init / deinit of BLE stack / controller not sufficient , instead of trying to completely stop the port which leads to deleting the task ? 

Thanks,

alberk8, 2022-07-26:
Hi @rahult-github,

This is what I am trying to do.   
1) Enable Bluetooth BLE Server/Peripheral mode when I receive data from i2c.  This enable any other Central to connect to it and retrieve it plus sent a command to shutdown the BLE.
2) if no Central connects to it within 2 mins it, the ESP32 will also shutdown the BLE.
3) Repeat step 1 and 2 for as long as the ESP32 is powered on.

rahult-github, 2022-07-26:
Hi @alberk8 ,

 This is a common usecase in Nimble and current examples too have handling for such scenario. You can have below two functions implemented to get this functionality.  Please note, disabling / enabling BLE by below method is an async process. So please ensure you give sufficient time for one operation (either enable or disable ) to be executed before the next operation is started. 

 Start / enable BLE: 

    ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());
    nimble_port_init();
   
    //Initialize the NimBLE Host configuration

    nimble_port_freertos_init(nimble_host_task);
     

  Stop/ disable BLE:
   
 
       int ret = nimble_port_stop();
       if(ret == 0) {
            nimble_port_deinit();
            ret = esp_nimble_hci_and_controller_deinit();
            if (ret != ESP_OK) {
                 ESP_LOGE(TAG, "esp_nimble_hci_and_controller_deinit() failed with error: %d", ret);
           }
       }

  `
   
Thanks.

alberk8, 2022-07-27:
Hi @rahult-github ,

I have been using the example that you have posted above.   The issue I am having is that after stopping the BLE and waited for 5 seconds then calling the Enable BLE the ESP will crash `nimble_port_freertos_init(nimble_host_task);`

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40097cde  PS      : 0x00060b33  A0      : 0x80098221  A1      : 0x3ffd3540
A2      : 0x3ffe4364  A3      : 0x00010010  A4      : 0x3ffd37ec  A5      : 0x3ffd4830
A6      : 0x3ffb6a50  A7      : 0x3ffd34d4  A8      : 0x00000017  A9      : 0x3fff89b4
A10     : 0x0000001f  A11     : 0x3ffe4f60  A12     : 0x3fff8b10  A13     : 0x3fff89bc
A14     : 0xfffffffc  A15     : 0x3fff8abc  SAR     : 0x00000018  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0001001c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x40097cdb:0x3ffd35400x4009821e:0x3ffd3560 0x40084596:0x3ffd3580 0x400845a9:0x3ffd35a0 0x4009457d:0x3ffd35c0 0x400e8ec1:0x3ffd3600 0x400e872c:0x3ffd3630 0x400da157:0x3ffd3660 0x400da225:0x3ffd3680 0x400d8288:0x3ffd36a0 0x400d8f21:0x3ffd36c0 0x400e3479:0x3ffd3700 0x400f91c2:0x3ffd3730 0x400f7929:0x3ffd3750

ELF file SHA256: 301db3574a9fe3a4
```

KaeLL, 2022-07-27:
@rahult-github you should probably add that to an example.

IshaESP, 2022-08-03:
Hi @alberk8,

Can you try out this patch? It's for esp-idf release/v4.4
I have made some changes in blecent/main/main.c file where host goes for initialization and de-initialization in an infinite loop, also added a delay in between.

[patch_4.4.zip](https://github.com/espressif/esp-idf/files/9250174/patch_4.4.zip)

Thanks,
Isha

alberk8, 2022-08-03:
Hi @IshaESP ,

Thanks for the sample file and I did apply the delay in the "code" but it did not fix the problem.   The code I am running is from another project [nanoFramework here ](https://github.com/nanoframework/nf-interpreter/blob/cbee7b0a78bfbf8b5c8cd0d36fb09927bdf31662/targets/ESP32/_nanoCLR/nanoFramework.Device.Bluetooth/esp32_nimble.cpp#L359)

Maybe this could help.  The error only happens on the second call to 'device_ble_start'  after the 'device_ble_dispose' was called previously.  When I was tracing it looks like vTaskDelete of 'nimble_port_freertos_deinit` is deferring the disposal to idle task.  Could be because the main_app is running in Core 1 and the  'nimble_port_freertos_init' is running in Core 0.

```
0x40098037: insert_free_block at /sources/esp-idf/components/heap/heap_tlsf.c:242
0x40098037: block_insert at /sources/esp-idf/components/heap/heap_tlsf.c:269
0x40098037: block_trim_free at /sources/esp-idf/components/heap/heap_tlsf.c:383
0x40098037: block_prepare_used at /sources/esp-idf/components/heap/heap_tlsf.c:460
0x40098037: tlsf_malloc at /sources/esp-idf/components/heap/heap_tlsf.c:758
0x4009857a: multi_heap_malloc_impl at /sources/esp-idf/components/heap/multi_heap.c:197
0x400846ae: heap_caps_malloc_base at /sources/esp-idf/components/heap/heap_caps.c:147
0x400846c1: heap_caps_malloc at /sources/esp-idf/components/heap/heap_caps.c:167
0x400947b5: xTaskCreatePinnedToCore at /sources/esp-idf/components/freertos/tasks.c:871
0x40127255: nimble_port_freertos_init at /sources/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/nimble_port_freertos.c:65
0x400fe52a: device_ble_start(ble_services_context&) at /workspaces/nf-interpreter/targets/ESP32/_nanoCLR/nanoFramework.Device.Bluetooth/esp32_nimble.cpp:579
0x400fdce2: Library_sys_dev_ble_native_nanoFramework_Device_Bluetooth_GenericAttributeProfile_GattServiceProvider::NativeStartAdvertising___BOOLEAN(CLR_RT_StackFrame&) at /workspaces/nf-interpreter/targets/ESP32/_nanoCLR/nanoFramework.Device.Bluetooth/sys_dev_ble_native_nanoFramework_Device_Bluetooth_GenericAttributeProfile_GattServiceProvider.cpp:776
0x400e239e: CLR_RT_Thread::Execute_Inner() at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:840
0x400e20f9: CLR_RT_Thread::Execute() at /workspaces/nf-interpreter/src/CLR/Core/Interpreter.cpp:642
0x400dd08d: CLR_RT_ExecutionEngine::ScheduleThreads(int) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:1205
0x400dc815: CLR_RT_ExecutionEngine::Execute(wchar_t*, int) at /workspaces/nf-interpreter/src/CLR/Core/Execution.cpp:632
0x400f4cbd: ClrStartup at /workspaces/nf-interpreter/src/CLR/Startup/CLRStartup.cpp:405
0x40116e66: CLRStartupThread at /workspaces/nf-interpreter/targets/ESP32/_nanoCLR/CLR_Startup_Thread.c:17
0x401150d9: main_task at /workspaces/nf-interpreter/targets/ESP32/_IDF/esp32/app_main.c:37
```

alberk8, 2022-08-24:
I am closing and leaving this comment here for future reference.   The issue surface due to double free/release of a [pointer ](https://github.com/nanoframework/nf-interpreter/blob/4174225fb7f707870f00dca71a2ad8b5c7bc36f7/targets/ESP32/_nanoCLR/nanoFramework.Device.Bluetooth/sys_dev_ble_native_nanoFramework_Device_Bluetooth_GenericAttributeProfile_GattServiceProvider.cpp#L36) that was not in direct reference in `nimble_port_freertos_init`.    In total the `FreeContext` function has a double free and a memory leak, taking care of those fix the odd issue.

closed 2022-08-24 by alberk8