espressif/esp-idf #9428: wifi connection failed with message "Association refused temporarily, comeback time 205824 mSec"

FlynnMa, 2022-07-25:
## Environment

- Development Kit:      esp-idf + esp_rainmaker
- Kit version : esp-idf the main branch after v5.0-dev
- Module or chip used:  ESP32C3
- IDF version (run ``git describe --tags`` to find it):    v5.0-dev-4257-g508661b03e
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
after I flash and reboot, I some time get reported wifi failure:
```
Association refused temporarily, comeback time 205824 mSec
```
20 seconds seem too long, how can I avoid it?

Full log looks like:
```
I (689) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (719) wifi:mode : sta (60:55:f9:79:7e:84)
I (719) wifi:enable tsf
I (729) wifi:new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:1
I (729) wifi:state: init -> auth (b0)
I (759) wifi:state: auth -> assoc (0)
E (759) wifi:Association refused temporarily, comeback time 205824 mSec
```

### Expected Behavior

### Actual Behavior

### Steps to reproduce
download esp-idf and build it, source the export.sh like:
```
source path/to/esp-idf/export.sh
```
download esp-rainmaker
```
cd examples/temperature_sensor
idf.py set-target esp32c3
idf.py build
idf.py /dev/ttyACM0 flash monitor
```

FlynnMa, 2022-07-25:
I also submited this issue under esp_rainmaker, here is the link
https://github.com/espressif/esp-rainmaker/issues/143#issue-1316478843

nishanth-radja, 2022-07-26:
@FlynnMa can you pls share a wireless capture during the association of the STA.Come back time is send from the AP in the assoc response.

MaxwellAlan, 2022-07-26:
Hi @FlynnMa 
Before your flash new bin, suggest ESP STA disconnect from the AP.
Thanks.

FlynnMa, 2022-07-27:
@nishanth-radja Could you please explain what is wireless capture do you mean? How to capture it?

FlynnMa, 2022-07-27:
Hi @MaxwellAlan what should I do in order to `disconnect from the AP`? I already configured reset before flash and after, should I send any additional command to perform the disconnect?

nishanth-radja, 2022-07-27:
> @nishanth-radja Could you please explain what is wireless capture do you mean? How to capture it?

@FlynnMa wireless capture is for capturing the packets in the air.If you have a linux laptop,Pls do the following.
1.install wireshark
2.ifconfig <wlan-interface> down
3.iwconfig wlp0s20f3 mode monitor
4.ifconfig wlp0s20f3 up
5.iwconfig wlp0s20f3 channel <ap-channel>
6.wireshark

MaxwellAlan, 2022-07-29:
> Hi @MaxwellAlan what should I do in order to `disconnect from the AP`? I already configured reset before flash and after, should I send any additional command to perform the disconnect?

Call esp_wifi_disconnect(); this api before reset.

FlynnMa, 2022-08-07:
@MaxwellAlan I found to manual call esp_wifi_disconnect() might not easy, can you please tell me how to make this function call as a part of download pre-process automatically?

MaxwellAlan, 2022-08-08:
@FlynnMa 

Could you try calling this API in your GOT_IP event?

AxelLin, 2022-08-08:
> > Hi @MaxwellAlan what should I do in order to `disconnect from the AP`? I already configured reset before flash and after, should I send any additional command to perform the disconnect?
> 
> Call esp_wifi_disconnect(); this api before reset.

@MaxwellAlan 
Does it matter to call esp_wifi_disconnect() before reset?

People usually just unplug usb cable (power-off) and replug usb cable (power-on).
i.e. esp_wifi_disconnect() is not called in a very common use case.
If reset without calling esp_wifi_disconnect() can cause issue, this is a big problem.

FlynnMa, 2022-11-12:
is there any patch could fix this ?  205 seconds are just not acceptable

sarveshb14, 2022-11-14:
Yes. we are working on allowing maximum value of assoc comeback time of 5 seconds. Will update shortly

sarveshb14, 2022-12-15:
Hi @FlynnMa ,
 
Thank you for waiting.
Fix is merged internally on master. It will be available on GitHub 5.0 soon. 
Since it includes a Wi-Fi libraries change, fix can not be included in a simple patch-file.

FlynnMa, 2022-12-17:
@sarveshb14 

Could you please share me the git commit id?

AxelLin, 2022-12-17:
> Hi @FlynnMa ,
> 
> Thank you for waiting. Fix is merged internally on master. It will be available on GitHub 5.0 soon. 

@sarveshb14 
Just to clarify:
1. Does this issue only happen on esp32c3? or also happens on other SoCs?
2. Does this issue happen on v4.x branches? If yes, please fix the impacted branches.

sarveshb14, 2022-12-18:
> @sarveshb14
> 
> Could you please share me the git commit id?

Please find [esp32-wifi-lib](https://github.com/espressif/esp32-wifi-lib/commit/39bcc426a6abc6dd7b1f1eb45f2c1bd8f6e0e778).
Also make sure to apply following change: [long_assoc_comeback_time.txt](https://github.com/espressif/esp-idf/files/10253330/long_assoc_comeback_time.txt)

sarveshb14, 2022-12-18:
>     1. Does this issue only happen on esp32c3? or also happens on other SoCs?
Yes. Issue is present on all SoCs and v4.x branches as well.
>     2. Does this issue happen on v4.x branches? If yes, please fix the impacted branches.
Yes. Will do

closed 2022-12-22 by espressif-bot

sforconi, 2022-12-28:
Hi @sarveshb14 can you please provide more details on how to update the esp_wifi-lib submodule?
I have tried this procedure on Linux, using esp-idf-v4.4.3.

Update the submodule from linux console:
```
$ cd esp-idf-v4.4.3 
$ git submodule update --remote components/esp_wifi/lib
Submodule path 'components/esp_wifi/lib': checked out 'c0deded2742ef619403eb3bda0c61e87fe5bcdf7'
```
Then clean and rebuild the project:

```
-- Found Git: /usr/bin/git (found version "2.34.1") 
CMake Warning at /home/samuele/data/esp/esp-idf-v4.4.3/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_wifi/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/samuele/data/esp/esp-idf-v4.4.3/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/samuele/data/esp/esp-idf-v4.4.3/tools/cmake/build.cmake:175 (__build_get_idf_git_revision)
  /home/samuele/data/esp/esp-idf-v4.4.3/tools/cmake/idf.cmake:50 (__build_init)
  /home/samuele/data/esp/esp-idf-v4.4.3/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)
...
...
[100/102] Linking C executable bootloader.elf
[101/102] Generating binary image from built executable
esptool.py v3.3.2
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated /home/samuele/data/VSCodeProjects/my-proj/build/bootloader/bootloader.bin
[102/102] cd /home/samuele/data/VSCodeProjects/my-proj/build/bootloader/esp-idf/esptool_py && /data/.espressif-v4.4.3/python_env/idf4.4_py3.10_env/bin/python /home/samuele/data/esp/esp-idf-v4.4.3/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /home/samuele/data/VSCodeProjects/my-proj/build/bootloader/bootloader.bin
Bootloader binary size 0x5160 bytes. 0x2ea0 bytes (36%) free.
[1043/1075] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/my_service/my_data.c.obj
/home/samuele/data/VSCodeProjects/my-proj/main/my_service/my_data.c: In function 'onDataReceived':
/home/samuele/data/VSCodeProjects/my-proj/main/my_service/my_data.c:55:40: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     packet_received_event.payload.data = (num_bytes > 0) ? bytes : NULL;    // discards const qualifier
                                        ^
[1073/1075] Linking CXX executable my-proj.elf
FAILED: my-proj.elf 
: && /home/samuele/data/.espressif-v4.4.3/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -mlongcalls  CMakeFiles/my-proj.elf.dir/project_elf_src_esp32s3.c.obj -o my-proj.elf  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/console/libconsole.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/usb/libusb.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  -Wl,--cref  -Wl,--Map="/home/samuele/data/VSCodeProjects/my-proj/build/my-proj.map"  -Wl,--gc-sections  -fno-rtti  -fno-lto  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp- [... cut ...] nfig_profile':
/home/samuele/data/esp/esp-idf-v4.4.3/components/wpa_supplicant/esp_supplicant/src/esp_wpa_main.c:81: undefined reference to `esp_wifi_sta_prof_is_wpa3_internal'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

The build process fails with:
`esp-idf-v4.4.3/components/wpa_supplicant/esp_supplicant/src/esp_wpa_main.c:81: undefined reference to esp_wifi_sta_prof_is_wpa3_internal'
`

Do we need to update also other submodules to make this works?

AxelLin, 2022-12-29:
> Hi @sarveshb14 can you please provide more details on how to update the esp_wifi-lib submodule? I have tried this procedure on Linux, using esp-idf-v4.4.3.

The esp_wifi fix is not yet available for v4.4 branch. (I'm also waitting for the fix for v4.3 branch)

sarveshb14, 2022-12-29:
Hi @sforconi @AxelLin , work is in progress to make this wifi fix availbale for v4.4 and v4.3 branches. Will be available soon.

AxelLin, 2023-01-10:
@sarveshb14 
Just remind that v4.4 branch still does not include this fix.

Energiz3r, 2023-01-19:
yeah can we get this one shipped please guys? this is a pain in the ass, every single flash requires a manual reboot to follow

AxelLin, 2023-01-28:
@sarveshb14 Please make sure v4.4.4 release include this fix. Thanks.

AxelLin, 2023-01-31:
@sarveshb14 @jack0c 

How is the status of this fix for v4.4 branch?
I cannot find this fix in release/v4.4 branch of esp32-wifi-lib, which means it's unlikely to be fixed in esp-idf-v4.4.4.
Can someone help to clarify the status?

walidamriou, 2023-01-31:
Hi @AxelLin  

- Is this fix included in esp-idf v5.0 (it happened with esp32-c3)?

- if not, when is it included?

heikkiw, 2023-01-31:
Hi !
I have ESP32C3 and this problem too. However I added WiFi.disconnect() just before trying to start WiFi,
that message disappeared from the serial monitor, and now it takes about 10 sec to start Wifi !

FlynnMa, 2023-02-02:
> Hi ! I have ESP32C3 and this problem too. However I added WiFi.disconnect() just before trying to start WiFi, that message disappeared from the serial monitor, and now it takes about 10 sec to start Wifi !

sounds something missing at the startup

sforconi, 2023-02-17:
Just tried with ESP-IDF-v4.4.4 and the problem still remains.
Invoking `esp_wifi_disconnect()` before `esp_wifi_start()` or `esp_wifi_connect()` has no effect.

sarveshb14, 2023-02-18:
Hi All, fix for this issue [0c8ec2a0](https://github.com/espressif/esp-idf/commit/0c8ec2a0668d5374bfedd2c9a926c93fed67541c) was merged after v4.4.4 release.

sarveshb14, 2023-02-18:
@sforconi 
> Invoking esp_wifi_disconnect() before esp_wifi_start() or esp_wifi_connect

There is not need/use of invoking `esp_wifi_disconnect()` before `esp_wifi_start()` or `esp_wifi_connect()`.  Issuing esp_wifi_disconnect() before connection is established will have no effect, as station is not connected to  AP. 

This issue is due to invalid association comeback value sent by an Access Point in association response frame. Provided fix is to reject AP when assoc comeback time sent by AP is > 5 seconds. (and try connecting again depending on connection retries configuration)

Maybe you can try with rebooting the AP or if Router configuration permits, set Assoc Comeback to some valid value.  

Please let me know if this helped.

sarveshb14, 2023-02-19:
@walidamriou 
> Is this fix included in esp-idf v5.0 (it happened with esp32-c3)?

Yes. This issue happened with esp32-c3 as well.
This fix is included in latest [v5.0.1](https://github.com/espressif/esp-idf/releases/tag/v5.0.1) tag (which was released recently)

sforconi, 2023-02-20:
@sarveshb14 
Hi Sarvesh, 
I've just done another try, but I'm still getting that problem with ESP-IDF v4.4.4, this is the monitor log that I get:

```I (6797) pp: pp rom version: e7ae62f
I (6797) net80211: net80211 rom version: e7ae62f
I (6807) wifi:wifi driver task: 3fca9fc8, prio:23, stack:6656, core=0
I (6807) system_api: Base MAC address is not set
I (6807) system_api: read default base MAC address from EFUSE
I (6807) wifi:wifi firmware version: 6567a16
I (6807) wifi:wifi certification version: v7.0
I (6807) wifi:config NVS flash: disabled
I (6807) wifi:config nano formating: disabled
I (6807) wifi:Init data frame dynamic rx buffer num: 32
I (6807) wifi:Init management frame dynamic rx buffer num: 32
I (6807) wifi:Init management short buffer num: 32
I (6807) wifi:Init static tx buffer num: 16
I (6807) wifi:Init tx cache buffer num: 32
I (6807) wifi:Init static tx FG buffer num: 2
I (6807) wifi:Init static rx buffer size: 1600
I (6807) wifi:Init static rx buffer num: 10
I (6807) wifi:Init dynamic rx buffer num: 32
I (6807) wifi_init: rx ba win: 6
I (6807) wifi_init: tcpip mbox: 32
I (6807) wifi_init: udp mbox: 6
I (6807) wifi_init: tcp mbox: 6
I (6807) wifi_init: tcp tx win: 5744
I (6807) wifi_init: tcp rx win: 5744
I (6807) wifi_init: tcp mss: 1440
I (6807) wifi_init: WiFi IRAM OP enabled
I (6807) wifi_init: WiFi RX IRAM OP enabled
I (6807) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (6847) wifi:mode : sta (f4:12:fa:d7:a3:9c)
I (6847) wifi:enable tsf
I (6847) wifi:Set ps type: 0

I (6847) MAIN: wifi started
I (6867) MAIN: wait for network up (00000002)...
I (9257) MAIN: wifi scan done: status 0, number 15
I (9257) MAIN: Total APs scanned = 15
I (9257) MAIN: 0 - Terrecablate_2.4Ghz_8941 [10] (-45) DE 
I (9257) MAIN: 1 - blueup-guest [10] (-46) DE 
I (9257) MAIN: 2 - B910_1990531478 [12] (-64) CN 
I (9257) MAIN: 3 - Terrecablate_2.4Ghz_91E1 [4] (-67) DE 
I (9257) MAIN: 4 - DIRECT-F6-HP DesignJet T630 [6] (-68) 
I (9257) MAIN: 5 - AGRISOLVING [2] (-71) 
I (9267) MAIN: 6 - GL-AR150-344 [11] (-71) 
I (9267) MAIN: 7 - TIM-93898011 [1] (-73) 
I (9267) MAIN: 8 - BOCCONBACCO [1] (-73) 
I (9267) MAIN: 9 - staff@dinamo [11] (-84) 
I (9267) MAIN: 10 - 1e6180b0 [1] (-86) CN
I (9267) MAIN: 11 - 1e6180df [1] (-87) CN
I (9267) MAIN: 12 - Zyxel_1401 [7] (-88) DE 
I (9267) MAIN: 13 - DIRECT-F5-HP DeskJet 2600 series [6] (-89) 
I (9267) MAIN: 14 - 1e617fc8 [1] (-90) CN
I (9267) MAIN: WiFi try connecting to Terrecablate_2.4Ghz_8941
I (11677) wifi:new:<10,0>, old:<1,0>, ap:<255,255>, sta:<10,0>, prof:1
I (12857) wifi:state: init -> auth (b0)
I (12867) wifi:state: auth -> assoc (0)
E (12877) wifi:Association refused temporarily, comeback time 268435 mSec
```

sarveshb14, 2023-02-20:
Hi @sforconi ,
Can you please checkout to this commit [0c8ec2a06](https://github.com/espressif/esp-idf/commit/0c8ec2a0668d5374bfedd2c9a926c93fed67541c) and try. 

This commit was merged into [release/v4.4](https://github.com/espressif/esp-idf/tree/release/v4.4) after [v4.4.4](https://github.com/espressif/esp-idf/tree/v4.4.4) was forked out. Hence v4.4.4 does not have this fix.

sforconi, 2023-02-20:
Thanks @sarveshb14, updating the to the latest commit has solved the problem, I can now see that the maximum allowed reconnection time is 5000ms.

```
I (6797) pp: pp rom version: e7ae62f
I (6797) net80211: net80211 rom version: e7ae62f
I (6807) wifi:wifi driver task: 3fca9540, prio:23, stack:6656, core=0
I (6807) system_api: Base MAC address is not set
I (6807) system_api: read default base MAC address from EFUSE
I (6807) wifi:wifi firmware version: 13c0b6c
I (6807) wifi:wifi certification version: v7.0
I (6807) wifi:config NVS flash: disabled
I (6807) wifi:config nano formating: disabled
I (6807) wifi:Init data frame dynamic rx buffer num: 32
I (6807) wifi:Init management frame dynamic rx buffer num: 32
I (6807) wifi:Init management short buffer num: 32
I (6807) wifi:Init static tx buffer num: 16
I (6807) wifi:Init tx cache buffer num: 32
I (6807) wifi:Init static tx FG buffer num: 2
I (6807) wifi:Init static rx buffer size: 1600
I (6807) wifi:Init static rx buffer num: 10
I (6807) wifi:Init dynamic rx buffer num: 32
I (6807) wifi_init: rx ba win: 6
I (6807) wifi_init: tcpip mbox: 32
I (6807) wifi_init: udp mbox: 6
I (6807) wifi_init: tcp mbox: 6
I (6807) wifi_init: tcp tx win: 5744
I (6807) wifi_init: tcp rx win: 5744
I (6807) wifi_init: tcp mss: 1440
I (6807) wifi_init: WiFi IRAM OP enabled
I (6807) wifi_init: WiFi RX IRAM OP enabled
I (6807) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (6847) wifi:mode : sta (f4:12:fa:d7:a3:9c)
I (6847) wifi:enable tsf
I (6847) wifi:Set ps type: 0

I (6857) MAIN: wifi started
I (6867) MAIN: wait for network up (00000002)...
I (9267) MAIN: wifi scan done: status 0, number 14
I (9267) MAIN: Total APs scanned = 14
I (9267) MAIN: 0 - blueup-guest [10] (-54) DE 
I (9267) MAIN: 1 - Terrecablate_2.4Ghz_8941 [10] (-55) DE 
I (9267) MAIN: 2 - NAVEL_93C564 [1] (-65) CN
I (9267) MAIN: 3 - BOCCONBACCO [1] (-66) 
I (9267) MAIN: 4 - TIM-93898011 [1] (-68) 
I (9267) MAIN: 5 - Terrecablate_2.4Ghz_91E1 [4] (-68) DE 
I (9267) MAIN: 6 - GL-AR150-344 [11] (-69) 
I (9267) MAIN: 7 - AGRISOLVING [2] (-70) 
I (9267) MAIN: 8 - DIRECT-F6-HP DesignJet T630 [6] (-73) 
I (9267) MAIN: 9 - B910_1990531478 [12] (-74) CN 
I (9267) MAIN: 10 - Loggini WiFi [7] (-83) 
I (9267) MAIN: 11 - Zyxel_1401 [7] (-85) DE 
I (9267) MAIN: 12 - 1e6180df [1] (-88) CN
I (9267) MAIN: 13 - 1e617fd1 [1] (-89) CN
I (9267) MAIN: WiFi try connecting to Terrecablate_2.4Ghz_8941
I (11677) wifi:new:<10,0>, old:<1,0>, ap:<255,255>, sta:<10,0>, prof:1
I (12967) wifi:state: init -> auth (b0)
I (12977) wifi:state: auth -> assoc (0)
E (12977) wifi:Association refused temporarily, comeback time 268435 mSec
E (12987) wifi:Association comeback time (268435) mSec too long. max allowed 5000 mSec
I (12987) wifi:state: assoc -> init (d000)
I (12987) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
I (12987) MAIN: wifi disconnected: Terrecablate_2.4Ghz_8941, reason 208
I (15387) MAIN: wifi scan done: status 0, number 14
I (15397) MAIN: Total APs scanned = 14
I (15397) MAIN: 0 - Terrecablate_2.4Ghz_8941 [10] (-55) DE 
I (15397) MAIN: 1 - blueup-guest [10] (-55) DE 
I (15397) MAIN: 2 - NAVEL_93C564 [1] (-64) CN
I (15397) MAIN: 3 - TIM-93898011 [1] (-67) 
I (15397) MAIN: 4 - BOCCONBACCO [1] (-67) 
I (15397) MAIN: 5 - Terrecablate_2.4Ghz_91E1 [4] (-67) DE 
I (15397) MAIN: 6 - GL-AR150-344 [11] (-68) 
I (15397) MAIN: 7 - DIRECT-F6-HP DesignJet T630 [6] (-72) 
I (15397) MAIN: 8 - B910_1990531478 [12] (-74) CN 
I (15397) MAIN: 9 - 1e617fc8 [1] (-80) CN
I (15397) MAIN: 10 - Loggini WiFi [7] (-83) 
I (15397) MAIN: 11 - 1e6180b0 [1] (-86) CN
I (15397) MAIN: 12 - Zyxel_1401 [7] (-86) DE 
I (15397) MAIN: 13 - DIRECT-F5-HP DeskJet 2600 series [6] (-88) 
I (15397) MAIN: WiFi try connecting to Terrecablate_2.4Ghz_8941
I (17807) MAIN: wifi disconnected: Terrecablate_2.4Ghz_8941, reason 205
I (20217) MAIN: wifi scan done: status 0, number 11
I (20217) MAIN: Total APs scanned = 11
I (20217) MAIN: 0 - blueup-guest [10] (-52) DE 
I (20217) MAIN: 1 - Terrecablate_2.4Ghz_8941 [10] (-54) DE 
I (20217) MAIN: 2 - BOCCONBACCO [1] (-65) 
I (20217) MAIN: 3 - Terrecablate_2.4Ghz_91E1 [4] (-67) DE 
I (20217) MAIN: 4 - TIM-93898011 [1] (-68) 
I (20217) MAIN: 5 - AGRISOLVING [2] (-69) 
I (20217) MAIN: 6 - GL-AR150-344 [11] (-71) 
I (20217) MAIN: 7 - DIRECT-F6-HP DesignJet T630 [6] (-72) 
I (20217) MAIN: 8 - B910_1990531478 [12] (-74) CN 
I (20217) MAIN: 9 - Zyxel_1401 [7] (-85) DE 
I (20217) MAIN: 10 - 1e6180df [1] (-88) CN
I (20217) MAIN: WiFi try connecting to Terrecablate_2.4Ghz_8941
I (22627) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
I (22627) wifi:state: init -> auth (b0)
I (22627) wifi:state: auth -> assoc (0)
I (22637) wifi:state: assoc -> run (10)
I (22747) wifi:connected with Terrecablate_2.4Ghz_8941, aid = 11, channel 10, BW20, bssid = 88:ac:c0:7e:89:41
I (22747) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (22747) wifi:pm start, type: 0

I (22747) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (22757) MAIN: wifi connected: Terrecablate_2.4Ghz_8941, channel 10, authmode = 3
I (22807) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (24257) esp_netif_handlers: sta ip: 192.168.1.149, mask: 255.255.255.0, gw: 192.168.1.1
```