espressif/esp-idf #9429: BLE cannot discovery services before connect timeout.

Zmmfly, 2022-07-25:
## Environment

- Module or chip used:  
 - ESP32-WROVER-E
- IDF version
 - v4.4

## Problem Description

Cannot discovery services, when connecting to BLE 4.1 device by NimBLE library.

### Expected Behavior

Services discovered, connect success.

### Actual Behavior

Cannot discovery services, and disconnect.

rahult-github, 2022-07-26:
Hi @Zmmfly , 

Can you please share more information 

1. Any sample application or existing IDF example that can be used to reproduce this issue ?
2. Do you have any log of your run ? Debug enabled log would be helpful.
3. Is this specific to devices with BLE 4.1 only ?
4. Is the remote 4.1 device a real word device or a proprietary solution ? 

Thanks

Zmmfly, 2022-07-26:
Hi @rahult-github, sorry for the late reply.
1. I wrote a demo to reproduce the problem: [IDFGH-7910_demo.zip](https://yun.mxtech.ltd/s/XgNjtpXwLPgPPzp)
2. Yes, the logger.log file in the demo folder records the entire process log;
3. Yes, this problem is only found on BLE4.1 devices at present, I don't know if this is a special case, or all BLE4.1 devices will have this problem;
4. It is a real device, and you can see the Bluetooth version by observing the broadcast in nRF connect.

Thanks

Zmmfly, 2022-07-27:
Hi @rahult-github, I don't know how to capture HCI on IDF, I used my phone to capture the HCI of the connection process, which may be of some help.
[btsnoop_hci_IDFGH7910.log](https://yun.mxtech.ltd/s/raEAzkDnkCjLgaJ)

Zmmfly, 2022-07-27:
Great, I successfully exported [btsnoop.log](https://github.com/espressif/esp-idf/files/9199287/btsnoop_ESP32_IDFGH7910.log) on ESP32.
It looks like:
1. The first connection is disconnected by the controller before exchanging MTU;
2. The second connection is also disconnected by the controller after reading the group type request;
3. The third connection is disconnected by the controller after exchanging MTU;

First connection:
[image]

Second connection:
[image]

Third connection:
[image]

Zmmfly, 2022-07-27:
The problem has been solved, and the connection process on the device side is not standardized.

rahult-github, 2022-07-28:
Hi @Zmmfly ,

> The problem has been solved 

Can you please help share more information about how this was solved ?.  The btsnoop log shared previously is HCI level log. So a disconnect from controller to host can be either from remote or event on the nrfapp side .

Zmmfly, 2022-07-29:
Hi @rahult-github, according to btsnoop.log on successful connection on phone:
I found that after the connection is created (OPCode 0x2043), the MTU exchange (ATT OPCode 0x02) is not performed immediately, but the following process:
1. First initiate a request to read the remote version information (OPCode 0x041d);
2. Initiate the read group type request again (ATT OPCode 0x10);
3. After completing the service discovery, perform the MTU exchange (ATT OPCode 0x02).

According to this sequence, I modified the local NimBLE code and added an API to read the remote version information (OPCode 0x041d) request; and modified the local esp-nimble-cpp, when connecting, it is optional to perform MTU exchange or read the remote version information (OPCode 0x041d), or do nothing, and the user program completes the subsequent request.

rahult-github, 2022-08-01:
Hi @Zmmfly  ,  Thanks .  You can open a PR with those changes. We can review it and include it as part of our codebase.  Closing the issue.

closed 2022-08-01 by rahult-github