espressif/esp-idf #9431: BLE maximum throughput

francis2tm, 2022-07-25:
Hello,
I was wondering which BLE example allows me to achieve maximum throughput. I'd prefer if it's the [BLE SPP Server example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_spp_server). Btw, why is the UART necessary in this example?

Thanks!

esp-cjh, 2022-07-26:
You can refer to [ble throughput](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_throughput) example.

francis2tm, 2022-07-30:
I've seen that example. However, I want a "serial like" connection. As similar as possible as the bluetooth classic spp example.

esp-cjh, 2022-08-01:
I think the UART is not necessary. It's just a way to input data. You can use other ways to replace it.

francis2tm, 2022-08-02:
Thanks for the swift reply.
I'm interested in setting up a server that sends a lot of data to a client, without the client making any request for it. In the [BLE Throughput example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_throughput) I think I can achieve this via the CONFIG_EXAMPLE_GATTS_NOTIFY_THROUGHPUT=1, instead of the CONFIG_EXAMPLE_GATTC_WRITE_THROUGHPUT=1, correct?

Additionally, can you what's happening in these lines https://github.com/espressif/esp-idf/blob/fde4afc67a2035a1f461dc511cd85dde1ef90368/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server/main/example_ble_server_throughput.c#L393-L433 ?

Why is the function `esp_ble_gatts_send_indicate` invoked again, given that [there's already a task invoking this function ](https://github.com/espressif/esp-idf/blob/fde4afc67a2035a1f461dc511cd85dde1ef90368/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server/main/example_ble_server_throughput.c#L629)

francis2tm, 2022-08-08:
Hey @igrr do you think you can help? I have a close deadline and I really need a answer about this soon

esp-cjh, 2022-08-08:
Sorry, I forget to anser you question.

These lines use to handle GATT write client characteristic configuration descriptor event(see Core Spec 5.2 Vol3 PartG 3.3.3.3). `value == 0x1` means notify enable and use notification to send data. `value == 0x2` means indicate enable and use indication to send data.
The last param of `esp_ble_gatts_send_indicate(..., bool need_confirm)` means whether to use notification(false) or indication(ture). This function invoked twice but the last param is different.

francis2tm, 2022-08-08:
Hello @esp-cjh,
Thanks for the reply. However, I don't understand... The `ESP_GATTS_WRITE_EVT` is in case the client sends data to the server, correct? But in my application, I want the communication to be unidirectional: only the server sends data do the client.
Am I thinking correctly? I'm a BLE beginner.

esp-cjh, 2022-08-08:
The client write client characteristic configuration to enable notify or indicate(only need to write once). And then server can use notification to send data to client unidirectionally. Client don't need to request the data or send confirm.

francis2tm, 2022-08-08:
@esp-cjh 
So in this case, I can delete the lines https://github.com/espressif/esp-idf/blob/fde4afc67a2035a1f461dc511cd85dde1ef90368/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server/main/example_ble_server_throughput.c#L393-L433 in my example?

esp-cjh, 2022-08-08:
Why you want to delete these lines? Server don't know wether the client enable notify or indicate without these lines. We cannot send notifications when the client haven't enabled notifications.

francis2tm, 2022-08-09:
Ok I understand, thanks!

francis2tm, 2022-08-09:
@esp-cjh Another thing, what is the purpose of this? https://github.com/espressif/esp-idf/blob/fde4afc67a2035a1f461dc511cd85dde1ef90368/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server/main/example_ble_server_throughput.c#L435

esp-cjh, 2022-08-09:
1. This function is used to handle write requeset of client. Because server should send write response when receive write request.
2. No. But we can get current sendable packets number, if number > 0, we can send notification. Refer to Line 626 of the example.

closed 2022-08-15 by francis2tm