espressif/esp-idf #9432: system console output not displaying prompt or user input when using 'idf.py monitor'

davidzuhn, 2022-07-25:
## Environment

- Development Kit:     Adafruit Huzzah-32 Feather, Espressif ESP32-S3 DevKit-C
- Module or chip used:  ESP32-WROOM-32, ESP32-S3-WROOM
- IDF version: v5.0-dev-4379-g36f49f361c
- Build System:  CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0, xtensa-esp32s3-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:    macOS 12.5
- Using an IDE?:  No
- Power Supply:   USB

## Problem Description

The system console doesn't work properly when run using 'idf.py monitor'.  The same program works perfectly when using minimum instead.   When running under 'idf.py monitor', the prompt is not being displayed, and the input itself is not printed either.   The output from the command is displayed.   

When I close the 'idf.py monitor', and connect using minicom (minicom -D /dev/cu.SLAB_USBtoUART -b 115200), the prompt displays as expected, and the input text is also displayed.   

Reconnecting with 'idf.py monitor --no-reset', the input text is not displayed and the subsequent prompts are again not being displayed. 

I have tested this on multiple architectures (ESP32 & ESP32-S3) using the same ESP-IDF version.   

### Expected Behavior

Earlier versions of esp-idf have worked identically no matter whether 'idf.py monitor' or minicom are used.  This is what I see when I run using minicom, and is what I expect when using 'idf.py monitor' as well.

```
I (350) cpu_start: Pro cpu up.
I (350) cpu_start: Starting app cpu, entry point is 0x40081528
I (0) cpu_start: App cpu up.
I (364) cpu_start: Pro cpu start user code
I (364) cpu_start: cpu freq: 160000000 Hz
I (365) cpu_start: Application information:
I (369) cpu_start: Project name:     console
I (374) cpu_start: App version:      v5.0-dev-4379-g36f49f361c-dirty
I (381) cpu_start: Compile time:     Jul 25 2022 14:31:21
I (387) cpu_start: ELF file SHA256:  d025ba5dc2b0ccbb...
I (393) cpu_start: ESP-IDF:          v5.0-dev-4379-g36f49f361c-dirty
I (401) heap_init: Initializing. RAM available for dynamic allocation:
I (408) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (414) heap_init: At 3FFB7D80 len 00028280 (160 KiB): DRAM
I (420) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (426) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (433) heap_init: At 40096D2C len 000092D4 (36 KiB): IRAM
I (440) spi_flash: detected chip: gd
I (443) spi_flash: flash io: dio
I (448) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (553) example: Command history enabled

This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Press Enter or Ctrl+C will terminate the console environment.
esp32> tasks  <- I typed 'tasks'
Task Name       Status  Prio    HWM     Task#
main            X       1       5884    4
IDLE            R       0       1116    6
IDLE            R       0       1108    5
esp_timer       S       22      3652    3
Tmr Svc         B       1       1588    7
ipc1            B       24      544     2
ipc0            B       24      568     1
esp32> version  <-- I typed 'version'
IDF Version:v5.0-dev-4379-g36f49f361c-dirty
Chip info:
        model:ESP32
        cores:2
        feature:/802.11bgn/BLE/BT/External-Flash:4 MB
        revision number:100
esp32> 
```

### Actual Behavior

This is when I run 'idf.py monitor':
```
I (350) cpu_start: Pro cpu up.
I (350) cpu_start: Starting app cpu, entry point is 0x40081528
0x40081528: call_start_cpu1 at /Users/zoo/esp32/esp-idf-latest-20220724/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (364) cpu_start: Pro cpu start user code
I (364) cpu_start: cpu freq: 160000000 Hz
I (365) cpu_start: Application information:
I (369) cpu_start: Project name:     console
I (374) cpu_start: App version:      v5.0-dev-4379-g36f49f361c-dirty
I (381) cpu_start: Compile time:     Jul 25 2022 14:31:21
I (387) cpu_start: ELF file SHA256:  d025ba5dc2b0ccbb...
I (393) cpu_start: ESP-IDF:          v5.0-dev-4379-g36f49f361c-dirty
I (401) heap_init: Initializing. RAM available for dynamic allocation:
I (408) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (414) heap_init: At 3FFB7D80 len 00028280 (160 KiB): DRAM
I (420) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (426) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (433) heap_init: At 40096D2C len 000092D4 (36 KiB): IRAM
I (440) spi_flash: detected chip: gd
I (443) spi_flash: flash io: dio
I (448) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (553) example: Command history enabled

This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
Press Enter or Ctrl+C will terminate the console environment.
Your terminal application does not support escape sequences.
Line editing and history features are disabled.
On Windows, try using Putty instead.
<- I type 'tasks'
Unrecognized command
<- I type 'tasks' again (I'm certain of spelling correctly the first time as well)
Task Namesks    Status	Prio	HWM	Task#
main           	X	1	5884	4
IDLE           	R	0	1116	6
IDLE           	R	0	1108	5
esp_timer      	S	22	3652	3
Tmr Svc        	B	1	1588	7
ipc1           	B	24	544	2
ipc0           	B	24	568	1
<- I type 'version'
IDF Version:v5.0-dev-4379-g36f49f361c-dirty
Chip info:
	model:ESP32
	cores:2
	feature:/802.11bgn/BLE/BT/External-Flash:4 MB
	revision number:100
```

Note the lack of the 'esp32>' prompt, and the lack of echo on the text that was input.   

### Steps to reproduce

1. build the system/console/advanced example from the specified version of esp-idf. (no changes to the sdkconfig
```
idf.py set-target esp32
idf.py build
```
2.  flash to the device (I first discovered this on my own device, but can reproduce on the Adafruit Huzzah-32)
```
idf.py flash
```
3. Start up the monitor
```
idf.py monitor
```
See the output as shown under 'Actual Behavior'
4. Disconnect the 'idf.py monitor' using Ctrl-] and then reconnect using minicom
```
minicom -D /dev/cu.SLAB_USBtoUART -b 115200
```
and witness the behavior as seen under 'Expected Behavior'

I've also done the comparable actions for esp32s3 using the ESP32-S3-DevKit-C, and received the same results (modulo the expected differences in output due to the architecture/board changes).   

### Code to reproduce this issue

The system/console/advanced example in ESP-IDF, unmodified.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder)
- [X] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [N/A] coredump -- None produced

[console-bug.zip](https://github.com/espressif/esp-idf/files/9184507/console-bug.zip) for ESP32
[console-bug-esp32s3.zip](https://github.com/espressif/esp-idf/files/9184589/console-bug-esp32s3.zip) for ESP32-S3

igrr, 2022-07-25:
Apologies for the issue. Indeed `idf.py monitor` is currently broken for console applications on `master` branch. The fix is under review and should be merged soon.

In the meantime you can revert commit https://github.com/espressif/esp-idf/commit/43c69f0910f1de9ba744ebc635a628cfdfece789, as a temporary workaround.

leeebo, 2022-09-22:
Modify UART `source_clk` in `esp_console_new_repl_uart` from `UART_SCLK_REF_TICK` to `UART_SCLK_APB` can fix ESP32S2 bug.