espressif/esp-idf #9434: OTA Web Update for ESP-IDF

lawrenceko-intecular, 2022-07-26:
Hi,

We see there's local OTA Web Update for Arduino package, is there such same package based on ESP-IDF?

What we are looking for:
We are seeking the ESP will be hosting the web server, user can login with `192.168.0.11` and upload the bin file to perform the update.

Thank you

AxelLin, 2022-07-26:
https://github.com/espressif/esp-idf/tree/master/examples/system/ota

lawrenceko-intecular, 2022-07-26:
> https://github.com/espressif/esp-idf/tree/master/examples/system/ota

Please read the procedure I was looking for:
```
We are seeking the ESP will be hosting the web server, user can login with 192.168.0.11 and upload the bin file to perform the update.
```

FanhuaCloud, 2022-07-27:
[http_server_ota-master.zip](https://github.com/espressif/esp-idf/files/9194558/http_server_ota-master.zip)
You can refer to this code, after the burning is completed, just submit the file to the endpoint to complete the update.Note that I did not implement error handling and firmware verification, you may need to implement it.

mahavirj, 2022-07-27:
@lawrenceko-intecular 

>  We are seeking the ESP will be hosting the web server, user can login with 192.168.0.11 and upload the bin file to perform the update.

ESP-IDF examples have guidelines on hosting firmware image in secure manner as mentioned in some of the above comments. You should be able to either create local secure server instance or use endpoints like AWS, GitHub etc.

In addition, please have a look at [OTA Firmware Upgrades](https://rainmaker.espressif.com/docs/ota.html) service from our ESP-Rainmaker offering that can help in this regard. 

Please prefer posting questions on https://esp32.com forum. I will go ahead and close as this is not an ESP-IDF issue.

closed 2022-07-27 by mahavirj

lawrenceko-intecular, 2022-07-27:
> @lawrenceko-intecular
>
>
> ESP-IDF examples have guidelines on hosting firmware image in secure manner as mentioned in some of the above comments. You should be able to either create local secure [...]

@mahavirj , you have quoted the comment that I made. We would want to have it the ESP32 to host a webserver and allow user to upload firmware (bin) files to perform local updates like routers can.

There's question on this on the forum and no beneficial support was provided. Please provide which example to use for local secure server and which to use local OTA updates.

Thank you

mahavirj, 2022-07-27:
>  Please provide which example to use for local secure server and which to use local OTA updates.

Please refer to either [http_server_simple](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/simple) or [file_serving](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving) examples for hosting HTTP server on the device.  If you want server instance with SSL then please refer to [https_server_simple](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server/simple).

You may combine above example with [native_ota](https://github.com/espressif/esp-idf/blob/master/examples/system/ota/native_ota_example/main/native_ota_example.c) to have local OTA updates.

Hope this helps!

lawrenceko-intecular, 2022-07-27:
> [http_server_ota-master.zip](https://github.com/espressif/esp-idf/files/9194558/http_server_ota-master.zip) You can refer to this code, after the burning is completed, just submit the file to the [...]

@FanhuaCloud , this helps a lot. May you elaborate on how can we submit the file to the endpoint?

I went to `192.168.0.123:8080` but it seems like the uri does not exist. I do see `/upload` in the code you provided, but unsure how to upload the firmware file to ESP.

Thank you

FanhuaCloud, 2022-07-27:
You can use python to upload,If web upload is required it seems to be possible to do it using axios.js.
```
import requests

def main():
    with open("ota-test.bin", "rb") as f:
        file_data = f.read()
    r = requests.post("http://172.16.1.21:8080/upload", data=file_data)
    print(r.status_code)

if __name__ == '__main__':
    main()
```

FanhuaCloud, 2022-07-27:
Of course, you can also use postman test if you don't have python

lawrenceko-intecular, 2022-07-27:
@FanhuaCloud , I have combined with `file_serving` example so I am uploading the bin file directly on the browser (i.e. `192.168.1.123`). With merging your code to perform local OTA update, I am seeing this error which stops it from updating.

```
I (39429) file_server: Starting OTA example
I (39429) file_server: Running partition type 0 subtype 16 (offset 0x00020000)
I (39429) file_server: Writing to partition subtype 17 at offset 0x120000
I (39809) file_server: Receiving file : /hello_world.bin...
I (39809) file_server: Remaining size : 170096
I (40129) file_server: New firmware version: 
I (40129) file_server: Running firmware version: v4.4-dev-3459-gc20223c55d-dirty
I (40139) file_server: esp_ota_begin succeeded
**E (40219) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x00)**
I (40219) file_server: Written image length 4320
I (40229) file_server: Remaining size : 165776
E (40439) esp_ota_ops: not found the handle
I (40439) file_server: Written image length 4320
I (40449) file_server: Remaining size : 161456
E (40679) esp_ota_ops: not found the handle
I (40679) file_server: Written image length 4320
I (40679) file_server: Remaining size : 157136
E (40909) esp_ota_ops: not found the handle
I (40909) file_server: Written image length 4320
I (40909) file_server: Remaining size : 152816
E (41079) esp_ota_ops: not found the handle
I (41079) file_server: Written image length 1440
I (41079) file_server: Remaining size : 151376
E (41369) esp_ota_ops: not found the handle
I (41369) file_server: Written image length 4320
I (41369) file_server: Remaining size : 147056
E (41579) esp_ota_ops: not found the handle
I (41579) file_server: Written image length 4320
I (41579) file_server: Remaining size : 142736
E (42029) esp_ota_ops: not found the handle
I (42029) file_server: Written image length 4320
I (42029) file_server: Remaining size : 138416
E (42259) esp_ota_ops: not found the handle
I (42259) file_server: Written image length 4320
I (42259) file_server: Remaining size : 134096
E (42449) esp_ota_ops: not found the handle
I (42449) file_server: Written image length 4320
I (42449) file_server: Remaining size : 129776
E (42659) esp_ota_ops: not found the handle
I (42659) file_server: Written image length 4320
I (42659) file_server: Remaining size : 125456
E (42959) esp_ota_ops: not found the handle
I (42959) file_server: Written image length 4320
I (42959) file_server: Remaining size : 121136
E (43189) esp_ota_ops: not found the handle
I (43189) file_server: Written image length 4320
I (43189) file_server: Remaining size : 116816
E (43409) esp_ota_ops: not found the handle
I (43409) file_server: Written image length 2880
I (43409) file_server: Remaining size : 113936
E (43579) esp_ota_ops: not found the handle
I (43579) file_server: Written image length 4320
I (43579) file_server: Remaining size : 109616
E (43979) esp_ota_ops: not found the handle
I (43979) file_server: Written image length 4320
I (43979) file_server: Remaining size : 105296
E (44309) esp_ota_ops: not found the handle
I (44309) file_server: Written image length 4320
I (44309) file_server: Remaining size : 100976
E (44599) esp_ota_ops: not found the handle
I (44599) file_server: Written image length 4320
I (44599) file_server: Remaining size : 96656
E (44879) esp_ota_ops: not found the handle
I (44879) file_server: Written image length 4320
I (44879) file_server: Remaining size : 92336
E (45279) esp_ota_ops: not found the handle
I (45279) file_server: Written image length 4320
[... 49 lines cut ...]
E (50059) esp_ota_ops: not found the handle
I (50059) file_server: Written image length 4320
I (50059) file_server: Remaining size : 17456
E (50299) esp_ota_ops: not found the handle
I (50299) file_server: Written image length 4320
I (50299) file_server: Remaining size : 13136
E (50509) esp_ota_ops: not found the handle
I (50509) file_server: Written image length 4320
I (50509) file_server: Remaining size : 8816
E (50719) esp_ota_ops: not found the handle
I (50719) file_server: Written image length 4320
I (50719) file_server: Remaining size : 4496
E (51149) esp_ota_ops: not found the handle
I (51149) file_server: Written image length 4496
**E (51149) file_server: esp_ota_end failed (ESP_ERR_NOT_FOUND)!
E (51159) esp_image: image at 0x120000 has invalid magic byte (nothing flashed here?)
E (51169) file_server: esp_ota_set_boot_partition failed (ESP_ERR_OTA_VALIDATE_FAILED)!**
I (51179) file_server: File reception complete
I (51309) file_server: File reception complete
I (51369) file_server: Found file : hello_world.bin (170096 bytes)
```

Did you get similar logs as I do?

Thank you

FanhuaCloud, 2022-07-27:
It might be a problem with your files, you might only be able to update and program this OTA program bin file.

lawrenceko-intecular, 2022-07-27:
@FanhuaCloud , I tried that as well. Still giving me the error `E (40219) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x00)`

lawrenceko-intecular, 2022-07-27:
@FanhuaCloud , did you get `E (49699) esp_ota_ops: not found the handle` error when uploading the bin files by chunk to ESP?

FanhuaCloud, 2022-07-27:
I don't have this problem, you can try uploading via postmap

dizcza, 2023-03-07:
@FanhuaCloud how to use your code with the HTML form submission?
```html
 <form method="POST" action="/upload_firmware" enctype="multipart/form-data">
     Firmware:<br>
     <input type="file" accept=".bin,.bin.gz" name="firmware">
     <input type="submit" value="Update Firmware">
 </form>
```

The char data buffer `ota_write_data`
```c
httpd_req_recv(req, ota_write_data, MIN(OTA_BUFFSIZE, remaining));
```
starts with
```
------WebKitFormBoundaryTvCGNmvhrRH9vE0A
Content-Disposition: form-data; name="firmware"; filename="esp32-sdpsensor.bin"
Content-Type: application/octet-stream

�/@��� @?P�2Tͫ54218bf-dirtyesp32-sdpsensor10:37:37Mar  7 2023v5.0-159-g75c3a69feb-dirty�)}9����9xZ��`�vd���w�ׁU��pthread
```
and therefore esp_ota_ops complains about the wring magic byte (0x2d is ASCII `-`):
```
I (00:00:06.347) ota: Remaining size : 1014563
I (00:00:06.354) ota: OTA downloaded 0 / 1208 Kb
I (00:00:06.373) ota: New firmware version: t-Disposition: form-data; name="firmware"; filename="esp32-sdpsensor.bin"
Content-Type: application/octet-stream

�/@�
I (00:00:06.395) ota: Running firmware version: 54218bf-dirty
I (00:00:06.412) ota: esp_ota_begin succeeded
E (00:00:06.427) esp_ota_ops: OTA image has invalid magic byte (expected 0xE9, saw 0x2d)
```
The new version name is also malformed.

Backdraft007, 2023-05-04:
Hi @dizcza,
look at the code from this issue https://github.com/espressif/esp-idf/issues/10996
It cut of the header information from HTTP request.

dizcza, 2023-05-04:
Thanks @Backdraft007, I'll give it a try.

The solution I've come up with, for the reference:
```html
<script>
function upload_firmware() {
    var fileInput = document.getElementById("choose_firmware").files;

    if (fileInput.length == 0) {
        alert("No file selected!");
    } else {
        document.getElementById("choose_firmware").disabled = true;
        document.getElementById("upload_firmware").disabled = true;

        var file = fileInput[0];
        var xhttp = new XMLHttpRequest();
        xhttp.onreadystatechange = function() {
            if (xhttp.readyState == 4) {
                if (xhttp.status == 200) {
                    document.open();
                    document.write(xhttp.responseText);
                    document.close();
                } else if (xhttp.status == 0) {
                    alert("Server closed the connection abruptly!");
                    location.reload()
                } else {
                    alert(xhttp.status + " Error!\n" + xhttp.responseText);
                    location.reload()
                }
            }
        };
        xhttp.open("POST", "/upload_firmware", true);
        xhttp.send(file);
    }
}
</script>
</head>

<body>
<h2>Firmware upgrade</h2>

<input id="choose_firmware" type="file" accept=".bin,.bin.gz">
<button id="upload_firmware" type="button" onclick="upload_firmware()">Upload</button>

</body>
```

It does not require stripping the header on the server side but does some js hacks on the client side. Both approaches have pros and cons.

SurajSonawane2415, 2023-11-28:
> @FanhuaCloud , did you get `E (49699) esp_ota_ops: not found the handle` error when uploading the bin files by chunk to ESP?

Hello I get this error. Can you please help me?