espressif/esp-idf #9436: [help] How do I know when a websocket is closed?

hyansuper, 2022-07-26:
I am using ESP32S3.
In the [websocket server example](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c) I simplified the echo_handler function to the below code just to check websocket status.
so in the last but 3 line I print the ws_pkt.type.

```cpp
static esp_err_t echo_handler(httpd_req_t *req)
{
    ESP_LOGI(TAG, "in handler");
    if (req->method == HTTP_GET) {
        ESP_LOGI(TAG, "Handshake done, the new connection was opened");        
        return ESP_OK;  
    }
    httpd_ws_frame_t ws_pkt;

    uint8_t *buf = NULL;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    buf = calloc(1, 1024);
    if (buf == NULL) {
        ESP_LOGE(TAG, "Failed to calloc memory for buf");
        return ESP_ERR_NO_MEM;
    }
    ws_pkt.payload = buf;
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
    ESP_LOGI(TAG, "TYPE=%d", ws_pkt.type);
    free(buf);
    return ret;
```
output
[image]

I can get NO other types except HTTPD_WS_TYPE_TEXT and HTTPD_WS_TYPE_BINARY of [httpd_ws_type_t](https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L1573), especially I am interested in HTTPD_WS_TYPE_CLOSE as I thought it indicates close of websocket, but I don't get it.

below is my client code in python to send a text message and a binary message to server
```python
import websockets
import asyncio
ip='192.168.137.165'
async def main():
    async with websockets.connect(f'ws://{ip}:80/ws') as ws:
        await ws.send('a')
        await ws.send(b'0')
        
asyncio.run(main())
```

chipweinberger, 2022-07-26:
```
    httpd_config_t config = HTTPD_DEFAULT_CONFIG();
    config.uri_match_fn = httpd_uri_match_wildcard;
    config.max_uri_handlers = 100;
    config.stack_size = 8192;
    config.recv_wait_timeout = 10; // Timeout for recv function (in seconds)
    config.send_wait_timeout = 10; // Timeout for send function (in seconds)
    config.open_fn = ws_open_fd;
    config.close_fn = ws_close_fd;
    config.lru_purge_enable = true;
```

hyansuper, 2022-07-27:
Thank you @chipweinberger 

Adding `open_fn` and `close_fn` callbacks is a possible solution, but they are invoked for every request and not specific for websockets. of course I can work around it by storing websocket's sockfd and later identify it in `close_fn`.

But a more proper way to do it is to catch `HTTPD_WS_TYPE_CLOSE` typed messages in websocket handler function, do anyone know how to do that? just like the [arduinoWebSockets](https://github.com/Links2004/arduinoWebSockets/blob/master/examples/esp32/WebSocketServer/WebSocketServer.ino#L34) lib does (isn't it implemented by esp-idf?)

ESP-YJM, 2022-07-28:
@hyansuper I think you can follow this example https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/wss_server/main/wss_server_example.c. When you enable https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/wss_server/main/wss_server_example.c#L124, the control frame like `HTTPD_WS_TYPE_CLOSE ` can receive in handler callback.

hyansuper, 2022-07-28:
@ESP-YJM Thank you.
I run the  wss_server_example, the https server outputs errors "mbedtls_ssl_handshake returned -0x7780", and I don't need https anyway.
So I still use the [ws_echo_server](https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c) example, 
I just added `.handle_ws_control_frames = true`, but the result is the same. I don't get  `HTTPD_WS_TYPE_CLOSE` and other control frame  messages
https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L405

ESP-YJM, 2022-07-28:
@hyansuper Which IDF version you used. You need copy the `ws_handler` in wss_server_example to ws_echo_server.

hyansuper, 2022-07-28:
Haha, it's this problem @ESP-YJM 
https://github.com/espressif/esp-idf/issues/7493#issuecomment-1027633541. haven't been fixed in the current main branch v4.4

ESP-YJM, 2022-07-28:
@hyansuper It has beed fixed in master branch and release/v4.4 branch. You can test it.

hyansuper, 2022-07-28:
I installed esp-idf with the [windows installer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/windows-setup.html) do you mean it's out dated? @ESP-YJM

ESP-YJM, 2022-07-28:
@hyansuper Not toolchain, i mean the code is outdated.

hyansuper, 2022-07-28:
I have a websocket related question, Does the wifi hardware and low level software support **concurrent** sending and receving? 
Suppose I want to make a IP phone (websockets bidirectional streaming), If I create another task(even pinned to the other core than the event loop) dedicated for sending while leaving the ws_handler callback for receving, would I get better performance than doing sending and receving all in ws_handler?

ESP-YJM, 2022-07-28:
Yes, we support.

closed 2022-07-28 by hyansuper