espressif/esp-idf #9437: 使用wear_levelling挂载文件系统错误错误

Zhengqinze05, 2022-07-26:
你好，我使用了idf的wear_levelling例程，它在第一次挂载文件系统时会报错。
esp_err_t err = esp_vfs_fat_spiflash_mount(base_path, "storage", &mount_config, &s_wl_handle);
当我 .format_if_mount_failed = true 时，它挂载失败会格式化文件系统，第二次挂载就能正常使用
但是当我.format_if_mount_failed = false,它始终无法挂载成功文件系统
日志显示：

[0;32mI (664) example: Mounting FAT filesystem[0m
[0;33mW (1394) vfs_fat_spiflash: f_mount failed (13)[0m
[0;31mE (1394) example: Failed to mount FATFS (ESP_FAIL)[0m

另外，我通过mkfatfs工具生成的bin烧录到对应的partitions分区时，文件系统始终挂载不成功。报错日志和上面的一样的
分区配置如下
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
storage,  data, fat,     0x110000,0x500000,
我想请问一下，如何将文件或文件夹烧录到对应的分区，并且可以进行正常的读写。

 通过esp_vfs_fat_spiflash_mount()进行挂载并设置.format_if_mount_failed = true时。 
第一次挂载会失败并格式化我的文件系统，第二次使用就正常，但是读不到我烧录进去的文件或文件夹，第二次挂载成功后，我读取系统下的文件时，仅显示刚刚写入的文件，烧录进去的文件并没有，我不确定是读不到还是被格式化了

[0;32mI (664) example: Mounting FAT filesystem[0m
------------- test list -------------
[0;32mI (674) example: Opening dir /spiflash[0m
DIR: /spiflash
	HELLO2.TXT
	HELLO1.TXT
------------- test list -------------
[0;32mI (684) example: Opening file[0m
[0;32mI (1124) example: File written[0m
[0;32mI (1124) example: Reading file[0m
[0;32mI (1124) example: Read from file: 'written using ESP-IDF v4.4.1-405-g6c5fb29c2c-dirty'[0m
[0;32mI (1124) example: Unmounting FAT filesystem[0m
[0;32mI (1234) example: Done[0m
test list的日志我加的一个打印路径下的文件。但是文件仅有第二次挂载成功之后，通过代码创建写入的文件，并没有烧录进去的bin文件下的文件
请问有什么方法解决吗，我想要将文件直接通过bin文件烧录到对应的partirions分区，并可以正常读写。
期待得到回复，谢谢！

Zhengqinze05, 2022-07-26:
我使用的是esp32_s3_korvo2_v3的板子，idf版本是4.4的

igrr, 2022-07-26:
@Zhengqinze05 Do I understand you correctly that you are generating the filesystem image using `mkfatfs` tool and then burning the image to the partition?

This is unlikely to work. Wear levelling partition format is not the same as that of a raw FAT filesystem. So this failure is expected.

In IDF v5.0 we are introducing a tool to generate FAT and wear_levelling partitions, wl_fatfsgen.py ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/fatfs.html#fatfs-partition-generator), [example](https://github.com/espressif/esp-idf/tree/master/examples/storage/fatfsgen)). This tool should address the use case you have. This tool isn't available in v4.4.1 release, though. You can either try to use the `master` branch for development, or copy the tool from the `master` branch and call it manually to generate the filesystem image.

Zmmfly, 2022-07-27:
mk 出来的 fs 和程序配置的 fat 扇区大小得一致，不然挂载不了的

Zhengqinze05, 2022-07-27:
@igrr @Zmmfly 非常感谢两位的回答
首先我的fat扇区大小和地址都确定是正确的
我用新的idf版本工具生成的磨损平衡bin文件可以正常读写，
但是由于一些模型文件是被.a文件读取的。它的要求好像是标准fat文件，我尝试将这些文件放到磨损平衡的分区里，在自己手动读取的操作是可行的，但是被.a文件读取是不可行的。
因此，我想问一下，是否有一种直接挂载烧录partition分区表bin文件的分区，也就是标准fat格式的，我尝试使用esp_vfs_fat_rawflash_mount()挂载文件系统，但是它似乎只能读取挂载目录下的文件，
然而挂载目录下的文件夹里面的文件内容读取不了，也就是只能读取挂载目录那一级的文件，下一级的文件读取不了
[0;31mE (870) wwe_example: Failed to open file for reading[0m
并且这个接口在idf版本里并没有太多的例子。
是否有一种标准格式fat文件系统挂载在parition分区的使用方法。

igrr, 2022-07-27:
@Zhengqinze05 Please post the command which you used to generate the standard FAT partition binary, we will take a look.

Zhengqinze05, 2022-07-28:
@igrr This is my command to generate standard fat
make_fatfs:
./tools/mkfatfs/mkfatfs_linux_x86 -c ./tools/fatdisk/vfsroot/ -s 0x500000 -t fatfs ./tools/fatdisk/vfs.bin

igrr, 2022-07-28:
Could you please give a few more details:
* what is the `mkfatfs_linux_x86` tool? I found something similar at https://github.com/labplus-cn/mkfatfs, is it this one?
* please attach your project's sdkconfig file (or at least the part with `CONFIG_FATFS_*` options) and the partition table

Another thing, in the `master` branch there is also a FAT partition generator at https://github.com/espressif/esp-idf/blob/master/components/fatfs/fatfsgen.py. If the `mkfatfs` project doesn't work for some reason, you could try this tool from IDF.

Zhengqinze05, 2022-07-29:
Thank you for your reply
yes, I use mkfatfs_ linux_ X86 tools.
this is my sdkconfig and the partition table

sdkconfig

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0009

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32s3-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x0
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_REGION_PROTECTION_ENABLE=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
[... 1617 lines cut ...]
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_USB_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

partition

# Name,       Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild,,,,
nvs,          data, nvs,     0x9000,    0x4000,
otadata,      data, ota,     0xd000,    0x2000,
phy_init,     data, phy,     0xf000,    0x1000,
ota_0,        app,  ota_0,         ,    2624K,
#model,        data, spiffs,  0xb00000,  4152K,
vfs,         data,  fat,      0x760000, 0x500000,
flash_tone,   data, 0x27,          ,    200K,

igrr, 2023-01-10:
@Zhengqinze05 Sorry for not updating this issue in time. I think the problem was solved by commit https://github.com/espressif/esp-idf/commit/41742c23698f5c0b09d920c51ed439be32d7fa1b, we are now reading the sector size (4kB or 512 bytes) from the image header, and configuring FATFS accordingly.

closed 2023-01-10 by igrr