espressif/esp-idf #9439: How to change the length of key name in NVS library?

aboyuanna, 2022-07-26:
I have a lot of parameters to save.However,the length of key name is more than 16 bytes.like this:

key,value
148EA6C1263A48362149,17:20:22
28E1C354A19571554633,17:20:45

so,can the length of key name be changed?

franz-ms-muc, 2022-07-26:
NVS_KEY_NAME_MAX_SIZE
no idea if you can just increase it, i would guess no.

igrr, 2022-07-26:
Hi @aboyuanna,
No, unfortunately the key length can't be easily changed, it is limited by NVS partition layout. The partition layout is explained [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#structure-of-entry).

One solution is, instead of storing `long_key:data` key-value pair, you can store two pairs: `short_key:long_key` and `short_key:data`. How to determine `short_key` depends on your use case. You can generate short keys as sequential numbers, or as 16-byte hashes of `long_key`. If the short keys cannot be computed from long keys, you can use `nvs_entry_find`, `nvs_entry_next`, `nvs_entry_info` functions to find the matching `short_key`, and then find get the matching `data` value for the given `short_key`.

closed 2022-07-27 by aboyuanna