espressif/esp-idf #9440: ILI9488 16bit Parallel interface is not working

rmshub, 2022-07-26:
## Environment

- Development Kit:  https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-ili9488.html 
- Module or chip used:  ESP32-S3-WROOM-1-N16R2
- IDF version : v4.4.1
- Build System:  idf.py
- Compiler version : (crosstool-NG esp-2021r2-patch3) 8.4.0
- Operating System:  Linux
- Using an IDE?: VS Code
- Power Supply:  USB
## Problem Description

LCD is not displaying any characters or color fill while interface with ESP32-S3 using i8080 interface. The kit is working properly with https://github.com/lovyan03/LovyanGFX library.

### Expected Behavior
LCD should display the content 
### Actual Behavior
LCD is not displaying any content.
### Code to reproduce this issue
[display.zip](https://github.com/espressif/esp-idf/files/9191538/display.zip)

Could you please review above code & let me know if changes required.

suda-morris, 2022-07-27:
I don't have a ILI9488 LCD around, I had a quick look at the datasheet, seems like the `lcd_cmd_bits` and `lcd_param_bits` are not `16` bits in lenght (while you set them both to 16 in your code).

FanhuaCloud, 2022-07-27:
```
        .lcd_cmd_bits = 16,
        .lcd_param_bits = 16,
```
should be changed to 8.
You can upgrade to IDF5.0 to refer to this code:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/i80_controller/main/i80_controller_example_main.c

rmshub, 2022-07-27:
Initially I tried with 8 bit command & parameter size using example from master branch. It is not working. Due to this, I changed it to 16bit. Used following parameter for writing commands with 8 bit length. 
`esp_lcd_panel_io_tx_param(io, ILI9488_CMD_POWER_CONTROL_1, (uint8_t[]){0x17, 0x15,}, 2);`

How many bytes will be transferred from cmd_val register if cmd_cycles count is 0. What is byte order on bus ? For example, 0x1122, will it be { 0x22 0x11 } or { 0x11 0x22}

suda-morris, 2022-07-28:
> How many bytes will be transferred from cmd_val register if cmd_cycles count is 0

The hardware will skip the CMD phase, which means, no data saved in cmd_val will be transferred.

> What is byte order on bus ? For example, 0x1122, will it be { 0x22 0x11 } or { 0x11 0x22}

LCD is big-endian, e.g. to send command 0x1234, byte 0x12 should appear on the data line first

rmshub, 2022-07-28:
Thanks. I'm successfully configured display. Setting RD pin as high resolved the issue.
By default ILI9488 is Portrait Mode 320x480. How to configure the ILI9488 & LVGL as 480x320 using ESP32-S3.

closed 2022-07-29 by rmshub

ldharmon, 2023-11-19:
I am also wanting to have a lvgl - esp-idf driver for the ili9488 using a parallel interface.
It appears that you have gotten one working.  Are you able to share the code?
I am using an elecrow esp32 RGB terminal (S3).  It works well with the Arduino/LoyvanGFX, but I don't like the confusion of having two frameworks.

espzav, 2023-11-20:
Hi @ldharmon, you can use our LCD table: https://github.com/espressif/esp-bsp/blob/master/LCD.md there are links to all tested drivers for esp-idf and LVGL. I recommend to use 'esp_lvgl_port' component for use LVGL.