espressif/esp-idf #9442: How do I increase wifi transfer rate and packet success rate based on ETH  以太网透传情况下提高wifi速率&减少丢包率

FPOOOOOOO, 2022-07-27:
## Environment

- Development Kit:      ESP32-Ethernet-Kit_A_V1.2
- IDF version:               v4.3
- Build System:            idf.py
- Power Supply:           USB
- issue language: English Chinese

## Problem Description
I wang to do a PC-PC communication and the transfer rate to be 16MBit or higner.
我想要做一个以太网透传，两个PC之间能够达到16 MBit 的速率。
[image]
- First 
I find the example 
`esp-idf/examples/ethernet/eth2ap`
and adapt it with 
`esp-idf/examples/wifi/getting_started/station`
 into eth2sta. 
Then I use 2 Ethernet-Kit and 2 WINDOWS PC to form a LAN as the pic above. The blue line is ETH,red line is wifi.
I set the 2 WINDOWS PC's static ip as 192.168.4.66 and 192.168.4.77.
Now the two device can PING each other. 
**BUT when AP ping STA, average time is 120ms, when STA ping AP, average time is 12ms.[don't know if it's due to the code]**

- 我把idf例子里的eth2ap和station结合改编出了eth2sta，跑在两块板子上，设置电脑的静态ip以后，这时候板子两端的电脑可以互相PING通，但是AP ping STA的的时候非常慢，网络延时有120ms左右，反过来就很快12ms（可能是代码的问题）

- Second
I run [JPerf](https://code.google.com/archive/p/xjperf/) on 2 WINDOWS PC to test the network performance, Now here comes the problem. 
I use 192.168.4.66 as JPerf client and 192.168.4.77 as JPerf server. The results are below:
**client:**
[image]
**server:**
[image]
The packet loss are very high! **And the rate was far less then the [official doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html) described.** 

**How do I increase wifi transfer rate and packet success rate!!** 

在两个电脑上跑jperf但是丢包率很高，和官网写得速率也差得很多，怎么才能提高wifi速率同时减少丢包率？

## Something I have done

- I noticed that in `eth2ap` and `eth2sta`, the ETH rate was balance to wifi rate. So I want the wifi rate to be 16Mbit and **I think the core to solve the problem is to increase the wifi transfer rate to 16Mbit based on**`esp_wifi_internal_tx`
- `eth2ap` and `eth2sta` code use `esp_wifi_internal_tx` to transfer data, I found `esp_wifi_internal_set_fix_rate` in `esp-idf/components/esp_wifi/include/esp_private/wifi.h` , I adjust the rate using the function but nothing changes, the JPerf test results are the same.
- Change the TX/RX AMPDU buffer but nothing works.
- Change the JPerf UDP send rate from 1Mbit to 10Mbit, the rate was like 1.2Mbit, but the packet loss goes to 90%.

- 由于ETH和wifi之间有平衡速率的任务，所以应该只需要把wifi速率上调到16Mbit就行，而且是基于esp_wifi_internal_tx的.
- 通过调整`esp_wifi_internal_set_fix_rate`来增加wifi的速率，没什么变化
- 改变TX RX AMPDU的buffer，没什么变化
- JPerf UDP灌包速率增加10M，测试速率虽然到1.2M，但是丢包率上升到90%

## Code to reproduce this issue
- Again **I think the core to solve the problem is to increase the wifi transfer rate to 16Mbit based on**`esp_wifi_internal_tx`
So the code doesn't need to be download onto Ethernet-Kit[if you dont't have a board],only to test and solve the Wifi rate part is enough. But I will still give my whole code here. Since there is `eth2ap` in `esp-idf/examples/ethernet/eth2ap` 

- 我认为解决问题的关键在于提高esp_wifi_internal_tx的速率同时减少丢包率，eth2ap和eth2sta代码在这里，eth2sta是我改编的。但是其实不需要用到ETH的部分，WiFi的部分应该足够测试了。

https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap

- **here is my `eth2sta`**

https://github.com/FPOOOOOOO/zhonghang/blob/ZHONGHANGonMAC/eth2sta_0707.c

Zmmfly, 2022-07-27:
如果IRAM足够，可以打开WiFi和Lwip的IRAM优化

FPOOOOOOO, 2022-07-27:
> 如果IRAM足够，可以打开WiFi和Lwip的IRAM优化
这样的丢包率是正常的吗？官网说默认WiFI速率是1Mbit，那灌包1M的情况下应该接近于0啊

Zmmfly, 2022-07-27:
> 这样的丢包率是正常的吗？官网说默认WiFI速率是1Mbit，那灌包1M的情况下应该接近于0啊

没做过ETH相关应用，这方面暂时没有发言权；
按以前做WiFi to PPPoS NAT的相关经验，IRAM优化打开，buffer加高，空间换时间，速率是能提高的。

FPOOOOOOO, 2022-07-28:
> 

OK, 我去试一下

FPOOOOOOO, 2022-07-28:
> > 这样的丢包率是正常的吗？官网说默认WiFI速率是1Mbit，那灌包1M的情况下应该接近于0啊
> 
> 没做过ETH相关应用，这方面暂时没有发言权； 按以前做WiFi to PPPoS NAT的相关经验，IRAM优化打开，buffer加高，空间换时间，速率是能提高的。

OK，我去试一下，会来汇报一下结果

FPOOOOOOO, 2022-07-28:
@david-cermak  Hi, david, I have noticed you in eth-ap-nat which allows packet copy from L2 to L3, I think you are a expert in packet transfer. Do you have any idea about this ETH2wifi transfer packet loss? Or something that could improve the wifi-transfer rate?

FPOOOOOOO, 2022-07-30:
Oh esp/esp-idf/examples/ethernet/eth2ap 如果只用这个例程进行jperf的话，PC1-espkit(ap)——PC2(sta) 是可以达到20Mbit/s。但是PC1-espkit(ap)——espkit(sta)-PC2，这个路径是丢包率极其高的，而eth2sta是两个官方例程的结合，等待检查中。

FPOOOOOOO, 2022-07-31:
看起来 Wifi STA的接口收发速率都只有13M，所以测试一个系统的时候应当分开来测试。在我看来这个Issue被我自己解决了，接下来就是Mesh速率的部分。
[image]
口

kostaond, 2022-08-01:
@FPOOOOOOO good to see that you were probably able to resolve the issue. Could you please share your solution in English? Thank you/

FPOOOOOOO, 2023-01-03:
我认为STA的速率应该不止这么一点，所以重新开了这个issue，欢迎大家讨论。我的邮箱是
email：
1284946208@qq.com

Xiehanxin, 2023-05-09:
可以在WiFi start 之前调用esp_wifi_set_mode(WIFI_MODE_NONE)); 关闭modem sleep 看一下

closed 2024-02-27 by Alvin1Zhang